arendst/Tasmota #2716: Wemos D1 Lite rebooting when Tasmota set as Generic Module

RJC078, 2018-05-13:
Hi,

Apologies in advance, im new at this!!

I have programmed a Wemos D1 Lite with Tasmota using both Arduino IDE and PlatformIO and am having problems! After programming the Wemos and it is configured as a Sonoff module, all seems to work fine, but after I change the module type to Generic, the Wemos continually reboots.

The output of "Status 0" is

> 00:00:13 RSL: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
> 00:00:13 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"Garage1","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:12","StartupUTC":"","Sleep":0,"BootCount":141,"SaveCount":145,"SaveAddress":"FB000"}}
> 00:00:13 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.13.1","BuildDateTime":"2018-05-13T19:51:34","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
> 00:00:13 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Telstra737F",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
> 00:00:13 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":533,"Free":468,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
> 00:00:13 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-0697","IPAddress":"192.168.0.67","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"61.9.194.49","Mac":"DC:4F:22:1D:A2:B9","Webserver":2,"WifiConfig":3}}
> 00:00:13 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.46","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1DA2B9","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
> 00:00:13 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:00:13 1970","Local":"Thu Jan 01 00:00:13 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":1,"Sunrise":"07:43","Sunset":"16:03"}}
> 00:00:13 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:00:13"}}
> 00:00:13 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:00:13","Uptime":"0T00:00:12","Vcc":3.032,"Wifi":{"AP":1,"SSId":"Telstra737F","RSSI":80,"APMac":"10:DA:43:4F:0C:37"}}}_

The Log shows the following

> _```
> 00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_1DA2B9, GroupTopic Garage1) Version 5.13.1-2_4_1
> 00:00:00 WIF: Connecting to AP1 Telstra737F in mode 11N as sonoff-0697...
> 00:00:08 WIF: Connected
> 00:00:08 DNS: Initialized
> 00:00:08 HTP: Web server active on sonoff-0697.local with IP address 192.168.0.67
> 00:00:09 MQT: Attempting connection...
> 00:00:09 MQT: Connected
> 00:00:09 MQT: tele/s
> 00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_1DA2B9, GroupTopic Garage1) Version 5.13.1-2_4_1
> 00:00:00 WIF: Connecting to AP1 Telstra737F in mode 11N as sonoff-0697...
> 00:00:08 WIF: Connected
> 00:00:08 DNS: Initialized
> 00:00:08 HTP: Web server active on sonoff-0697.local with IP address 192.168.0.67
> 00:00:09 MQT: Attempting connection...
> 00:00:09 MQT: Connected
> 00:00:09 MQT: tele/sonoff/LWT = Online (retained)
> 00:00:09 MQT: cmnd/sonoff/POWER = 
> 00:00:09 MQT: tele/sonoff/INFO1 = {"Module":"Generic","Version":"5.13.1","FallbackTopic":"DVES_1DA2B9","GroupTopic":"Garage1"}
> 00:00:09 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-0697","IPAddress":"192.168.0.67"}
> 00:00:09 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
> 00:00:09 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
> 00:00:09 MQT: stat/sonoff/RESULT = {"GPIO0":"0 (None)","GPIO1":"0 (None)","GPIO2":"0 (None)","GPIO3":"0 (None)","GPIO4":"0 (None)","GPIO5":"1 (DHT11)","GPIO12":"0 (None)","GPIO13":"0 (None)","GPIO14":"0 (None)","GPIO15":"0 (None)","GPIO16":"0 (None)"}
> 11:25:27 APP: Restarting
> 
>  ets Jan  8 2013,rst cause:2, boot mode:(3,0)
> 
> load 0x4010f000, len 1384, room 16 
> tail 8
> chksum 0x2d
> csum 0x2d
> v614f7c32
> ~ld
> ```_

I have tried erasing all the memory and re-programming and the fault still occurs. If i change the Module back to the Sonoff module it settles down and maintains a connection to the router.

Thanks in advance for your help!

reloxx13, 2018-05-13:
mqtt command is wrong and repeatling the whole time.

changing the type is causing a reboot which is normal.

their was another ticket some days before with same issue.

reloxx13, 2018-05-13:
found it, here you go: https://github.com/arendst/Sonoff-Tasmota/issues/2658#issuecomment-387112217

RJC078, 2018-05-13:
Thanks reloxx

Is it the "MQT: cmnd/sonoff/POWER =" message that is causing the reboot?? Im just not sure which message is causing it and trying to understand!

Thanks

reloxx13, 2018-05-13:
noo, this is causing it:

```
00:00:09 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}   

00:00:09 MQT: stat/sonoff/RESULT = {"GPIO0":"0 (None)","GPIO1":"0 (None)","GPIO2":"0 (None)","GPIO3":"0 (None)","GPIO4":"0 (None)","GPIO5":"1 (DHT11)","GPIO12":"0 (None)","GPIO13":"0 (None)","GPIO14":"0 (None)","GPIO15":"0 (None)","GPIO16":"0 (None)"}
```

2nd one is the output of changing something at the device configuration and this is repeatly sent, causing the (normal) restart.

RJC078, 2018-05-13:
O.K., so i have no idea what message was being sent from my Home Assistant Mosqitto Broker, but I have Uninstalled then re-installed the Broker and it has fixed the fault! 

Thankyou for your guidance reloxx!

closed 2018-05-13 by RJC078

reloxx13, 2018-05-13:
Isnt their a button like "Clear Retain Messages" ?