arendst/Tasmota #2718: #undef USE_IR_RECEIVE in user_config_override.h does not disable IRrecv

TRIROG, 2018-05-13:
Playing around with wemos, minimizing the code footprint i disabled the IR sending with #undef USE_IR_REMOTE and receiving with #undef USE_IR_RECEIVE since i dont need these functions.

However i noticed that there is still option 47 IRrecv available in module configuration, but if understand correctly it shouldnt be, due to #undef USE_IR_RECEIVE in user_config_override.h.

arendst, 2018-05-13:
Although the option 47 is still there the code has been disabled. 

Removing option 47 also would change the sequence of options available resulting in wrong options selected once USE_IR_RECEIVE is enabled again.

Just ignore it is still there.

TRIROG, 2018-05-13:
can you please confirm 
do i need to use #undef USE_IR_RECEIVE if i also use  #undef USE_IR_REMOTE in order to reduce the code size for ~9k or is there no need to use  #undef USE_IR_RECEIVE becausse disabling IR_REMOTE also disables IR_RECEIVE?

arendst, 2018-05-13:
The latter.

You could also try and see what happens....