arendst/Tasmota #2720: Node-RED help with RF Bridge

JAComms, 2018-05-13:
Hello
I have not had much luck sending a captured code from the console from Node-RED to a fan.
I require more than 16 RF Codes and would like to send them at will from Node-RED to the fan via the RF Bridge.  Here is the received code from the console : 

 MQTT: tele/RFbridge/RESULT = {"RfReceived":{"Sync":7880, "Low":260, "High":780, "Data":"0BBCA2", "RfKey":"None"}}

If anyone has a generic example to work with or a working Flow I am sure many people would be grateful, I have spent more time than I would like to admit to trying to get this working.
If it makes any difference I am using Sonoff-Tasmota v5.6.1 and a current release RFbridge.

Thank you
Justin

Frogmore42, 2018-05-13:
I don't have an RF Bridge, but I have used Node-RED.  What have you tried that didn't work?
Also, there have been a LOT of changes since 5.6.1, so you really should upgrade.

MatzeJoerling, 2018-05-13:
`[{"id":"b5e52129.7ffc3","type":"mqtt out","z":"a6230776.dc0908","name":"","topic":"cmnd/sonoff/backlog","qos":"","retain":"","broker":"e0d5e520.4d00b8","x":480,"y":2820,"wires":[]},{"id":"6eba8d61.b4a5e4","type":"inject","z":"a6230776.dc0908","name":"RF Send 1","topic":"cmnd/gauge/Backlog","payload":"RfHigh 780; RfLow 260; RfSync 7880; RfCode 0BBCA2","payloadType":"str","repeat":"","crontab":"","once":false,"onceDelay":0.1,"x":240,"y":2820,"wires":[["b5e52129.7ffc3"]]},{"id":"e0d5e520.4d00b8","type":"mqtt-broker","z":"","broker":"localhost","port":"1883","clientid":"","usetls":false,"compatmode":false,"keepalive":"60","cleansession":true,"willTopic":"","willQos":"0","willRetain":"false","willPayload":"","birthTopic":"","birthQos":"0","birthRetain":"false","birthPayload":""}]`

Have no rfBridge but check if that works ...

wongnam, 2018-05-13:
@JAComms 
[{"id":"cb6a6cf0.82d48","type":"mqtt out","z":"b4d473ec.fa7cc","name":"","topic":"cmnd/rfbridge1/Backlog","qos":"1","retain":"false","broker":"8bd29f76.ae029","x":434.099853515625,"y":2541.583303451538,"wires":[]},{"id":"f31967a8.ade548","type":"ui_button","z":"b4d473ec.fa7cc","name":"Fan Light","group":"c22c217a.0f05b","order":8,"width":0,"height":0,"passthru":false,"label":"","color":"gold","bgcolor":"SteelBlue","icon":"fa-lightbulb-o fa-2x","payload":"rfsync 11040; rflow 350; rfhigh 1070; rfcode #CF1AA5","payloadType":"str","topic":"","x":96.09988784790039,"y":2483.5833644866943,"wires":[["cb6a6cf0.82d48"]]},{"id":"3783aaba.1aae76","type":"ui_button","z":"b4d473ec.fa7cc","name":"STOP","group":"c22c217a.0f05b","order":12,"width":0,"height":0,"passthru":false,"label":"","color":"","bgcolor":"SteelBlue","icon":"fa-stop","payload":"rfsync 11040; rflow 350; rfhigh 1070; rfcode #CF1AAC","payloadType":"str","topic":"","x":89.09984016418457,"y":2644.58340883255,"wires":[["cb6a6cf0.82d48"]]},{"id":"b19f327d.3ad42","type":"ui_button","z":"b4d473ec.fa7cc","name":"","group":"c22c217a.0f05b","order":9,"width":"2","height":"1","passthru":false,"label":"LOW","color":"","bgcolor":"SteelBlue","icon":"","payload":"rfsync 11040; rflow 350; rfhigh 1070; rfcode #CF1AA8","payloadType":"str","topic":"","x":87.99984359741211,"y":2605.5834078788757,"wires":[["cb6a6cf0.82d48"]]},{"id":"d50f1d25.88aac","type":"ui_button","z":"b4d473ec.fa7cc","name":"","group":"c22c217a.0f05b","order":10,"width":"2","height":"1","passthru":false,"label":"MED","color":"","bgcolor":"SteelBlue","icon":"","payload":"rfsync 11040; rflow 350; rfhigh 1070; rfcode #CF1AA2","payloadType":"str","topic":"","x":89.99985694885254,"y":2565.5834074020386,"wires":[["cb6a6cf0.82d48"]]},{"id":"a1fefd02.c5404","type":"ui_button","z":"b4d473ec.fa7cc","name":"","group":"c22c217a.0f05b","order":11,"width":"2","height":"1","passthru":false,"label":"HIGH","color":"","bgcolor":"SteelBlue","icon":"","payload":"rfsync 11040; rflow 350; rfhigh 1070; rfcode #CF1AA6","payloadType":"str","topic":"","x":88.9998550415039,"y":2525.583405971527,"wires":[["cb6a6cf0.82d48"]]},{"id":"1410f69b.4b3d49","type":"comment","z":"b4d473ec.fa7cc","name":"433 RF Remote Control","info":"","x":430.4998474121094,"y":2463.7501401901245,"wires":[]},{"id":"8bd29f76.ae029","type":"mqtt-broker","z":"","broker":"localhost","port":"1883","clientid":"","usetls":false,"compatmode":true,"keepalive":"60","cleansession":true,"willTopic":"","willQos":"0","willPayload":"","birthTopic":"","birthQos":"0","birthPayload":""},{"id":"c22c217a.0f05b","type":"ui_group","z":"","name":" RF Remote","tab":"349b3b40.ddb984","order":11,"disp":true,"width":"6","collapse":false},{"id":"349b3b40.ddb984","type":"ui_tab","z":"","name":"RF Bridge","icon":"rss_feed","order":4}]