arendst/Tasmota #2721: Allow setting of TIME_DST and TIME_STD values from Tasmota Console

apastuszak, 2018-05-13:
**Have you look for this feature in other issues and in the wiki?**

Yes

**Is your feature request related to a problem? Please describe.**

I am unable to use local time to set timers without recompiling Tasmota myself.  The provided binaries do not work well for me, since they're compiled with UTC+1.

**Describe the solution you'd like**
 Be able to set TIME_DST and TIME_STD from the console, the way you can other paramters.

**Describe alternatives you've considered**

I can't think of an alternative, since I was told these values can only be set at compile time.

**Additional context**

I think the binaries available for download should be as generic as possible.  Not being able to configure timezone rules after install from the console makes the supplied binaries on Github more difficult to use.

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2018-05-13:
It will have to be a compromise between code size and functionality. The problem with these parameters are that many user input errors need to be prevented. Allowing only numbers saves a lot of code space wasted on string comparison but is less user friendly...

Your approach of separate commands sounds good. Let me think about a way to store all possible values in as small as possible Settings value(s).

apastuszak, 2018-05-13:
Well, right now, users need to set it manually and compile it.  So, there is no option at all.  Perhaps someone could wrote a small python script that allows you to calculate the number that Tasmota is using and you enter that value.  Would that not prevent an increase in memory?  We could make that the v.1 deliverable and work further on refining it and making it more user friendly.  Just exposing it to the console for manipulation would be an improvement.

arendst, 2018-05-13:
@ascillato just uploaded an updated settings.h with room for STD/DST.

ascillato, 2018-05-13:
:smile: :+1:

ascillato, 2018-05-15:
Tested and work great! :+1: Thanks

Added also to the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands) the `TimeDST` and `TimeSTD `commands usage.

ascillato, 2018-05-15:
@apastuszak please test it and if works close the issue. Thanks

apastuszak, 2018-05-15:
@ascillato I will need to compile in the morning and flash something.

Also, don't see anything in the wiki.

ascillato, 2018-05-15:
Please, check it again. It is there on commands.

ascillato, 2018-05-15:
[image]

apastuszak, 2018-05-15:
I did a search for TimeDST instead of Time_DST.   It's 11:00 PM in my corner of the world, so I will test in the morning.

ascillato, 2018-05-15:
Ups. my bad. The command is TimeDST. updating the wiki

Thanks :+1:

apastuszak, 2018-05-15:
That will make my morning.  Less frustrating!  :-)

ascillato, 2018-05-15:
:laughing:

apastuszak, 2018-05-15:
I'm getting a compilation error on sonoff_minimal:

```
.pioenvs\sonoff-minimal\libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x4): undefined reference to `setup'
.pioenvs\sonoff-minimal\libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x8): undefined reference to `loop'
.pioenvs\sonoff-minimal\libFrameworkArduino.a(core_esp8266_main.cpp.o): In function `loop_wrapper()':
core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x21): undefined reference to `setup'
core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x2d): undefined reference to `loop'
collect2.exe: error: ld returned 1 exit status
*** [.pioenvs\sonoff-minimal\firmware.elf] Error 1
```

arendst, 2018-05-15:
@apastuszak expected with pio. Just start the compilation again and if still fails remove the .pioenvs folder

qingz2004, 2018-05-15:
Compiled and worked good.
Thanks!

apastuszak, 2018-05-15:
Still won't compile for me.

```
F:/Andy/Downloads/Sonoff-Tasmota-development/sonoff/xdrv_03_energy.ino: In function 'void EnergySnsInit()':
F:/Andy/Downloads/Sonoff-Tasmota-development/sonoff/xdrv_03_energy.ino:1046:75: error: no matching function for call to 'TasmotaSerial::TasmotaSerial(uint8_t&, uint8_t&, int)'
PzemSerial = new TasmotaSerial(pin[GPIO_PZEM_RX], pin[GPIO_PZEM_TX], 1);
^
F:/Andy/Downloads/Sonoff-Tasmota-development/sonoff/xdrv_03_energy.ino:1046:75: note: candidates are:
In file included from F:/Andy/Downloads/Sonoff-Tasmota-development/sonoff/xdrv_03_energy.ino:415:0:
lib\TasmotaSerial-1.2.0\src/TasmotaSerial.h:40:5: note: TasmotaSerial::TasmotaSerial(int, int)
TasmotaSerial(int receive_pin, int transmit_pin);
^
lib\TasmotaSerial-1.2.0\src/TasmotaSerial.h:40:5: note:   candidate expects 2 arguments, 3 provided
lib\TasmotaSerial-1.2.0\src/TasmotaSerial.h:38:7: note: constexpr TasmotaSerial::TasmotaSerial(const TasmotaSerial&)
class TasmotaSerial : public Stream {
^
lib\TasmotaSerial-1.2.0\src/TasmotaSerial.h:38:7: note:   candidate expects 1 argument, 3 provided
lib\TasmotaSerial-1.2.0\src/TasmotaSerial.h:38:7: note: constexpr TasmotaSerial::TasmotaSerial(TasmotaSerial&&)
lib\TasmotaSerial-1.2.0\src/TasmotaSerial.h:38:7: note:   candidate expects 1 argument, 3 provided
F:/Andy/Downloads/Sonoff-Tasmota-development/sonoff/xdrv_03_energy.ino:1048:23: error: 'class TasmotaSerial' has no member named 'hardwareSerial'
if (PzemSerial->hardwareSerial()) { ClaimSerial(); }
^
F:/Andy/Downloads/Sonoff-Tasmota-development/sonoff/xsns_15_mhz19.ino: In function 'void MhzInit()':
F:/Andy/Downloads/Sonoff-Tasmota-development/sonoff/xsns_15_mhz19.ino:287:74: error: no matching function for call to 'TasmotaSerial::TasmotaSerial(uint8_t&, uint8_t&, int)'
MhzSerial = new TasmotaSerial(pin[GPIO_MHZ_RXD], pin[GPIO_MHZ_TXD], 1);
^
F:/Andy/Downloads/Sonoff-Tasmota-development/sonoff/xsns_15_mhz19.ino:287:74: note: candidates are:
In file included from F:/Andy/Downloads/Sonoff-Tasmota-development/sonoff/xdrv_03_energy.ino:415:0:
lib\TasmotaSerial-1.2.0\src/TasmotaSerial.h:40:5: note: TasmotaSerial::TasmotaSerial(int, int)
TasmotaSerial(int receive_pin, int transmit_pin);
^
lib\TasmotaSerial-1.2.0\src/TasmotaSerial.h:40:5: note:   candidate expects 2 arguments, 3 provided
lib\TasmotaSerial-1.2.0\src/TasmotaSerial.h:38:7: note: constexpr TasmotaSerial::TasmotaSerial(const TasmotaSerial&)
class TasmotaSerial : public Stream {
^
lib\TasmotaSerial-1.2.0\src/TasmotaSerial.h:38:7: note:   candidate expects 1 argument, 3 provided
lib\TasmotaSerial-1.2.0\src/TasmotaSerial.h:38:7: note: constexpr TasmotaSerial::TasmotaSerial(TasmotaSerial&&)
lib\TasmotaSerial-1.2.0\src/TasmotaSerial.h:38:7: note:   candidate expects 1 argument, 3 provided
F:/Andy/Downloads/Sonoff-Tasmota-development/sonoff/xsns_15_mhz19.ino:289:22: error: 'class TasmotaSerial' has no member named
'hardwareSerial'
if (MhzSerial->hardwareSerial()) { ClaimSerial(); }
^
F:/Andy/Downloads/Sonoff-Tasmota-development/sonoff/xsns_17_senseair.ino: In function 'void SenseairInit()':
F:/Andy/Downloads/Sonoff-Tasmota-development/sonoff/xsns_17_senseair.ino:196:78: error: no matching function for call to 'TasmotaSerial::TasmotaSerial(uint8_t&, uint8_t&, int)'
SensairSerial = new TasmotaSerial(pin[GPIO_SAIR_RX], pin[GPIO_SAIR_TX], 1);
^
F:/Andy/Downloads/Sonoff-Tasmota-development/sonoff/xsns_17_senseair.ino:196:78: note: candidates are:
In file included from F:/Andy/Downloads/Sonoff-Tasmota-development/sonoff/xdrv_03_energy.ino:415:0:
lib\TasmotaSerial-1.2.0\src/TasmotaSerial.h:40:5: note: TasmotaSerial::TasmotaSerial(int, int)
TasmotaSerial(int receive_pin, int transmit_pin);
^
lib\TasmotaSerial-1.2.0\src/TasmotaSerial.h:40:5: note:   candidate expects 2 arguments, 3 provided
lib\TasmotaSerial-1.2.0\src/TasmotaSerial.h:38:7: note: constexpr TasmotaSerial::TasmotaSerial(const TasmotaSerial&)
class TasmotaSerial : public Stream {
^
lib\TasmotaSerial-1.2.0\src/TasmotaSerial.h:38:7: note:   candidate expects 1 argument, 3 provided
lib\TasmotaSerial-1.2.0\src/TasmotaSerial.h:38:7: note: constexpr TasmotaSerial::TasmotaSerial(TasmotaSerial&&)
lib\TasmotaSerial-1.2.0\src/TasmotaSerial.h:38:7: note:   candidate expects 1 argument, 3 provided
F:/Andy/Downloads/Sonoff-Tasmota-development/sonoff/xsns_17_senseair.ino:198:26: error: 'class TasmotaSerial' has no member named 'hardwareSerial'
if (SensairSerial->hardwareSerial()) { ClaimSerial(); }
^
F:/Andy/Downloads/Sonoff-Tasmota-development/sonoff/xsns_18_pms5003.ino: In function 'void PmsInit()':
F:/Andy/Downloads/Sonoff-Tasmota-development/sonoff/xsns_18_pms5003.ino:105:59: error: no matching function for call to 'TasmotaSerial::TasmotaSerial(uint8_t&, int, int)'
PmsSerial = new TasmotaSerial(pin[GPIO_PMS5003], -1, 1);
^
F:/Andy/Downloads/Sonoff-Tasmota-development/sonoff/xsns_18_pms5003.ino:105:59: note: candidates are:
In file included from F:/Andy/Downloads/Sonoff-Tasmota-development/sonoff/xdrv_03_energy.ino:415:0:
lib\TasmotaSerial-1.2.0\src/TasmotaSerial.h:40:5: note: TasmotaSerial::TasmotaSerial(int, int)
TasmotaSerial(int receive_pin, int transmit_pin);
^
[... 25 lines cut ...]
lib\TasmotaSerial-1.2.0\src/TasmotaSerial.h:38:7: note:   candidate expects 1 argument, 3 provided
F:/Andy/Downloads/Sonoff-Tasmota-development/sonoff/xsns_20_novasds.ino:89:26: error: 'class TasmotaSerial' has no member named 'hardwareSerial'
if (NovaSdsSerial->hardwareSerial()) { ClaimSerial(); }
^
[... cut ...]
lib\TasmotaSerial-1.2.0\src/TasmotaSerial.h:38:7: note: constexpr TasmotaSerial::TasmotaSerial(TasmotaSerial&&)
lib\TasmotaSerial-1.2.0\src/TasmotaSerial.h:38:7: note:   candidate expects 1 argument, 3 provided
F:/Andy/Downloads/Sonoff-Tasmota-development/sonoff/xsns_23_sdm120.ino:214:25: error: 'class TasmotaSerial' has no member named 'hardwareSerial'
if (SDM120Serial->hardwareSerial()) { ClaimSerial(); }
^
*C** [.pioenvs\sonoff\src\sonoff.ino.cpp.o] Error 1ompiling .pioenvs\sonoff\lib4dd\ESP8266WiFi\ESP8266WiFiMulti.cpp.o
```

arendst, 2018-05-15:
You'll need to update the TasmotaSerial library too (now 2.0.0)

ascillato, 2018-05-15:
Try on the menu platform.io the option rebuild c/c++

apastuszak, 2018-05-15:
@arendst  Need some hand-holding on how how to do that.

arendst, 2018-05-15:
Copy from the repository lib folder the (new) TasmotaSerial-2.0.0 folder to your local lib folder. Remove the current TasmotaSerial-1.2.0 folder

apastuszak, 2018-05-15:
@arendst Had both 1.2 and 2.0 folders.  Deleted 1.2 and I think it's OK.  Thanks.

apastuszak, 2018-05-15:
OK, got everything to compile.  Flashed Sonoff minimal without issue.  Flashes full Sonoff and device rebooted and I get a rapidly flashing green light.  Flashed with WIFI_MANAGER option.  It's not coming up with it's own wifi network to connect to.  I held the button down for 10 seconds to do a wipe and it still won't come up or show a wifi network.

I'm going to need to flash using the serial adapter.  Sorry for the delay.

apastuszak, 2018-05-15:
Ok, FINALLY managed to flash 5.14.0 using Visual Studio Code and Platform I/O over serial.  DST settings seem to be working fine.

Thank you much for implementing this so quickly!  Closing issue.

closed 2018-05-15 by apastuszak