arendst/Tasmota #2722: PID doesn't handle negative temperatures

wienerdogracing, 2018-05-13:
The PID portion doesn't handle a negative number correctly from the sensor.

The code on line 187 in xdrv_92_pid.ino

 if (value != NULL  &&  strlen(value) > 0  &&  isdigit(value[0]) )

should be changed to

if (value != NULL  &&  strlen(value) > 0  && isdigit(value[0]) || (value[0] == '-' && isdigit(value[1]) ) )

This checks to see if the first value of the sensor return is a digit and also check to see if the first value is a neagative sign and the 2nd value is a digit.

ascillato, 2018-05-13:
Hi, there is no such file (**xdrv_92_pid.ino**) on the repository.

Please, open the issue on the forked repository where you download that file.

And also please link that issue to here and close this issue.

Thanks :+1:

ascillato, 2018-05-14:
Hi,

The PID Repository is https://github.com/colinl/Sonoff-Tasmota/tree/pid_branch as explained on the wiki

The developer is @colinl

wienerdogracing, 2018-05-14:
Sorry I did look at that branch before but I couldn't figure out how to to post an issue.  I am not really good with GitHub so I thought maybe the issue was supposed to be posted on the main branch.  Closing now.

closed 2018-05-14 by wienerdogracing

ascillato, 2018-05-14:
@colinl is also at the forum at: https://groups.google.com/forum/#!forum/sonoffusers

Please, try there to contact him.

ascillato, 2018-05-14:
Also you can make a _Pull Request_ on his Repository doing the following:

* Browse the files of the https://github.com/colinl/Sonoff-Tasmota/tree/pid_branch
* Find the _xdrv_92_pid.ino_.
* Click on it
* When it opens, click on edit (the pen icon at the top right)
* Edit the file.
* At the bottom you will have a `Create Pull Request` Button. You will have also a comment box where you can explain the reason for that proposed change.

ascillato, 2018-05-15:
Hi @wienerdogracing 

Could you contact colinl for your issue?

colinl, 2018-05-15:
Don't worry, I am on the case. Traveling at the moment but will sort it when I am back.