arendst/Tasmota #2723: Sonoff S31 Energy Total can not be reset

qingz2004, 2018-05-13:
Firmware version: 5.12.0n

11:50:49 CMD: EnergyReset3 0
11:50:49 MQT: stat/S31/RESULT = {"EnergyReset":{"Total":0.000,"Yesterday":0.000,"Today":0.000}}
11:50:57 CMD: Status 8
11:50:57 MQT: stat/S31/STATUS8 = {"StatusSNS":{"Time":"2018-05-13T11:50:57","ENERGY":{"Total":7.987,"Yesterday":0.000,"Today":0.000,"Power":0,"Factor":0.00,"Voltage":110,"Current":0.000}}}

Jason2866, 2018-05-13:
May this issue is fixed in v.5.13. Just give this version a try

qingz2004, 2018-05-14:
Yes, you are right.
Works with 5.13.1b.

tobox, 2018-08-23:
Still fails for me in 6.1.0a

09:50:30 CMD: EnergyReset3 0
09:50:30 MQT: stat/sonoff-power-1/RESULT = {"EnergyReset":{"Total":0.000,"Yesterday":0.000,"Today":0.000}}
09:50:30 MQT: tele/sonoff-power-1/SENSOR = {"Time":"2018-08-23T09:50:30","ENERGY":{"Total":0.428,"Yesterday":0.000,"Today":0.000,"Period":0.000,"Power":0.000,"Factor":0.00,"Voltage":240.87,"Current":0.000}}

qingz2004, 2018-08-23:
Same here with 6.1.1.7.
The issue came back.

ascillato, 2018-08-23:
Hi

@qingz2004

Bug Fixed.

Can you test it?

Thanks.

qingz2004, 2018-08-23:
@ascillato Will do when I get home. And will close it if it works.

qingz2004, 2018-08-23:
Tested.
Fixed in 6.1.1.9.

closed 2018-08-23 by qingz2004

ihavenonick, 2019-03-17:
its not work in 6.4.1.22
09:09:20 CMD: EnergyReset3
09:09:20 MQT: stat/sonoff/RESULT = {"EnergyReset":{"Total":6.776,"Yesterday":0.119,"Today":0.002}}
[image]

qingz2004, 2019-03-17:
@ihavenonick The command to set Energy Total to 0 is EnergyReset3 0

ihavenonick, 2019-03-17:
Oh, thank you :-)

rb28z2, 2020-11-20:
I'm having the same issue. I've called `EnergyReset3 0` but my S31 is still showing the same values as before. This is on Tasmota 9.1.0.

cpankonien, 2020-11-25:
Same issue here on Tasmota 9.1.0. Have been able to reset all counters except Energy Total.

AlphaDE, 2021-08-07:
Same issue with 9.5.0

rbm78bln, 2021-08-16:
Can confirm, doesn't work with BlitzWolf SHP6 neither

AlphaDE, 2021-08-16:
I found a way, I had to reset the day values first, they overrided always the total values. After I reset the day value, I could reset the total value.

mrneutron42, 2021-12-06:
Same issue in 10.0.0 
In fact, EnergyReset is no longer a valid command!

10:48:21.212 CMD: EnergyReset3 0
10:48:21.218 RSL: RESULT = {"Command":"Unknown"}
10:48:55.821 CMD: EnergyReset0
10:48:55.827 RSL: RESULT = {"Command":"Unknown"}
10:49:00.286 CMD: EnergyReset1
10:49:00.292 RSL: RESULT = {"Command":"Unknown"}
10:49:03.020 CMD: EnergyReset2
10:49:03.026 RSL: RESULT = {"Command":"Unknown"}
10:49:08.229 CMD: EnergyReset
10:49:08.235 RSL: RESULT = {"Command":"Unknown"}

AlphaDE, 2021-12-06:
But https://tasmota.github.io/docs/Commands/#commands-list shows the valid commands, which did work for me on 10.0:
EnergyTotal 0
EnergyToday 0
EnergyYesterday 0

mrneutron42, 2021-12-06:
Yes, now I see a note that the EnergyReset command was deprecated in version 10.0.0.

EnergyReset<x> | x = 1..5
`With version 10, this command has been replaced, see above.`