arendst/Tasmota #2724: No control over an Amazon Echo Plus since version 5.13.1

Levaeri, 2018-05-13:
**Describe the bug**
No control over an Amazon Echo Plus since version 5.13.1

A clear and concise description of what the bug is.
In Version 5.12.0, the "Hue Emulation" in Tasmota provided direct control over an Amazon Echo Plus, without the need for a Philips Hue hub.
Since version 5.13.1 this is no longer possible, the devices lose the control function with the update and are no longer found in the Alexa app. Testing on 2 different Sonoff-Basic-Devices with the same Result.

Note:
With Version 5.12.0 was always switched correctly, but there was often, not always, a hint from Alexa, that the device with the name X does not respond.

Also, make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues and in the wiki
- [x] Hardware used : Sonoff Basic and Alexa Echo Plus
- [x] Provide the output of command ``status 0`` : 
19:53:42 CMD: status 0
19:53:42 RSL: STATUS = {"Status":{"Module":1,"FriendlyName":["Baum"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
19:53:42 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T02:06:55","StartupUTC":"2018-05-13T16:46:47","Sleep":0,"BootCount":36,"SaveCount":201,"SaveAddress":"FA000"}}
19:53:42 RSL: STATUS2 = {"StatusFWR":{"Version":"5.13.1","BuildDateTime":"2018.05.01 21:05:42","Boot":6,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
19:53:42 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["HomeRouterLosConEssid",""],"TelePeriod":300,"SetOption":["55800001","55C18000"]}}
19:53:42 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":530,"Free":472,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
19:53:42 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-2757","IPAddress":"10.10.10.122","Gateway":"10.10.10.254","Subnetmask":"255.255.255.0","DNSServer":"10.10.10.254","Mac":"68:C6:3A:80:8A:C5","Webserver":2,"WifiConfig":4}}
19:53:42 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sun May 13 18:53:42 2018","Local":"Sun May 13 19:53:42 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:11","Sunset":"20:22"}}
19:53:42 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-05-13T19:53:42"}}
19:53:42 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-05-13T19:53:42","Uptime":"0T02:06:55","Vcc":3.463,"POWER":"OFF","Wifi":{"AP":1,"SSId":"HomeRouterLosConEssid","RSSI":64,"APMac":"34:81:C4:36:0D:1D"}}}

**To Reproduce**
Steps to reproduce the behavior:
1. Install Tasmota Version 5.12.0 
2. Activate "Hue Emulation"
3. Start a Search for new Objekts in Alexa (Only on Echo Plus!) via App
4. A new Objekt was found
5. You can control this Objekt via Speech or App.
6. Upgrade to Version 5.13.1
7. Lost the Control in the App
8. Delete the Objekt
9. Start a Search for new Objekts in Alexa: Not found a new Objekt.

Workaround:
- Install Version 5.12.0 an start a scan vor new Objekts - The Sonoff was found again with normal control. 

**Expected behavior**
A clear and concise description of what you expected to happen.
I have expect the same Funktion Alexa Echo Plus with Tasmota V1.12.0
I hope, the wrong  hint from Alexa "The device with the name X does not respond" was fixed.

**Screenshots**
If applicable, add screenshots to help explain your problem.
No Screenshot avaibile.

**Additional context**
Add any other context about the problem here.

**(Please, remember to close the issue when the problem has been addressed)**

reloxx13, 2018-05-13:
[Workaround](https://github.com/esp8266/Arduino/issues/4235)
or go back to ExpressIf library 2.3.0
[ExpressIf Versions comparsion](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#wifi-issues-fw--512x-expressif-versions)

reloxx13, 2018-05-13:
Edit above: Workaroudn link corrected

Levaeri, 2018-05-13:
Thank you for your prompt reply.
I did not succeed with the search before, probably my English is a bit too bad. I just wanted to report an incident and I hope you can fix it with one of the next versions.
I'm just a user ... ;-)

ascillato, 2018-05-13:
You can try the compiled versions by reloxx13 at https://github.com/reloxx13/Sonoff-Tasmota-Modified/releases/tag/5.13.1b-mod-1.21.2

ascillato, 2018-05-13:
those are compiled using ESP8266 Board Library v2.3.0

Levaeri, 2018-05-13:
Thanks for the hint, I will test it soon...

reloxx13, 2018-05-14:
> I hope you can fix it with one of the next versions.

Tasmota cannot fix anything to it, its an alexa bug and the workaround needs to be implemented in the library wich is not hosted by tasmota.

in the workaround issue i posted is a link to the ExpressIf Git and an open issue.

wongnam, 2018-05-14:
if you compile with ESP8266 core ver.2.4.0 => need use this workaround https://github.com/arendst/Sonoff-Tasmota/issues/1639#issuecomment-360218536

if you compile with cored version 2.3.0 => no need any adjust as above.

closed 2018-06-11 by ascillato2