arendst/Tasmota #2725: Newbie trying to understant config_override

enboig, 2018-05-13:
My intention is to use sonota to upload minimal version completely configured to the device. To keep my code organized and updated I was planning to make local branches of the code for every device with configuration in user_config_override.h

Is it possible to specify the MODULE in override file?

ascillato, 2018-05-13:
Yes, use the key

module modulename

Modulename is found on sonoff_template file

Remember that the configuration is stored in eeprom meaning that after the first time you configure the device it will remember it if you flash another version.

reloxx13, 2018-05-13:
why you want to make branches for every device?
one is enough. it will be a pain to update alot of branches.

at the current state you dont even need to fork it, just clone it with git and use the override file. the file is ignored and should not affect your cloned local repo.
for updates just use git pull.

enboig, 2018-05-13:
Ok, so an override file "per device" is enough. I was searching for a simple way to update the code and keep the configuration of every device.

My concert was how to keep the configuration for each update. But I see that after first flash and configure, I can update with any configuration because it is saved. Does it also apply if configured with full version and updated with minimal version?

ascillato, 2018-05-13:
Yes

Frogmore42, 2018-05-14:
You only need a single override file.  Put in your:
1. Wi-Fi configuration 
1. MQTT configuration
1. NTP configuration
1. Timezone/DST configuration

When you commission a new device it will be connected to your Wi-Fi and MQTT so you can then configure everything else using the commands.  That way, you only have to keep track of a minimal and the current FW you want to use.  It makes it a LOT easier to manage.

ascillato, 2018-05-17:
Hi,

If this has solved your issue, please close it. Thanks :+1:

closed 2018-05-17 by enboig