arendst/Tasmota #2728: S20_EU_V1.3: After flashing, no reaction

Testi89, 2018-05-14:
I have successfully flashed the sonoff-DE.bin firmware in an S20_EU_V1.3.
Then I could successfully read the chip_id, so the chip is working.

But when I insert the plug into AC power and press shortly 5 times the button, nothing happens.

My expectation was that the button LED is blinking and that I can see an AP at 192.168.4.1. But nothing so far.
Maybe the hardware layout of V1.3 changed and the tasmota firmware is not working?

Jason2866, 2018-05-14:
Did you flash with option -DOUT?
Plz  post your flash command. For Testing try to flash Version 5.12.
In 5.13.x arduino sdk 2.4.0 is used. This makes often problems (many issues posted...).

Testi89, 2018-05-14:
This is my flash command, I am working without arduino under Linux:

$esptool.py --port /dev/ttyUSB0  write_flash 0x0  ~/Downloads/sonoff-DE.bin 
esptool.py v1.3-dev
Connecting...
Auto-detected Flash size: 8m
Running Cesanta flasher stub...
Flash params set to 0x0020
Writing 544768 @ 0x0... 544768 (100 %)
Wrote 544768 bytes at 0x0 in 47.2 seconds (92.3 kbit/s)...
Leaving...

Testi89, 2018-05-14:
Also no success with 5.12.0. I also tried a flash_erase first.

$ esptool.py --port /dev/ttyUSB0  write_flash 0x0  ~/Downloads/sonoff-DE.bin 
esptool.py v1.3-dev
Connecting...
Auto-detected Flash size: 8m
Running Cesanta flasher stub...
Flash params set to 0x0020
Writing 499712 @ 0x0... 499712 (100 %)
Wrote 499712 bytes at 0x0 in 43.3 seconds (92.3 kbit/s)...
Leaving...

Jason2866, 2018-05-14:
Yes, flashing failed because device only supports DOUT mode.
See wiki https://github.com/arendst/Sonoff-Tasmota/wiki/Esptool

Testi89, 2018-05-14:
Ok, thanks. Now it is working with DOUT.