arendst/Tasmota #2729: cmnd isn't responding

phdelodder, 2018-05-14:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://forum.digital-pig.com/) Maintenance Mode: Fixing issues, syncing content, back soon!
- [ x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x ] Hardware used : Luani HVIO board
- [x ] Provide the output of command ``status 0`` : not responding , every 300 sec I get this: tele/hvio1/STATE {"Time":"2018-05-14T14:32:13","Uptime":"2T05:50:54","Vcc":3.438,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Dxxxe","RSSI":76,"APMac":"xxx"}}

**(Please, remember to close the issue when the problem has been addressed)**

When I use MQTT to turn on power1 cmnd/hvio1/POWER1 ON, I'm not getting any feedback. Also with other commands like cmnd/hvio1/status 0 I'm not seeing anything appear in mqtt. 

I'm using version 5.13.1.

curzon01, 2018-05-14:
You are able to receive telemetry topics so mqtt might work.

- Start the device and observe the INFO1 topic, it contains the FallbackTopic device id, e.g
`MQT: tele/hvio1/INFO1 = {"Module":"Luani HVIO","Version":"5.13.1a","FallbackTopic":"DVES_123456","GroupTopic":"sonoffs"}`
- send `Status 0` command via fallback topic, e. g.
`mosquitto_pub -h yourmqttserver -u <user> -P <password> -t "cmnd/DVES_123456/Status" -m "0"` and post the result

Hint: If you have changed MQTT client to the device name (it doesn't matter how), e.g. if you has set _hvio1_ as MQTT client ID, restore it to default "DEV_%06X".

phdelodder, 2018-05-14:
Thanks for the reply, the error was a bootloop. wdt .... 

I flashed an empty file and reflashed tasmota that solved it

closed 2018-05-14 by phdelodder