arendst/Tasmota #2730: Sonoff Basic Relay Quick Flick while it is restarting

wongnam, 2018-05-14:
- In the case of Sonoff is in "ON" state, don't know if anyone here has encountered sonoff flicker quickly while it is restarting or not? I have over 10 Sonoff switch are facing this issue.

- Note that when it is in "OFF" state, no errors are encountered.

- [x] Searched the problem in issues and in the wiki

[image]

Jason2866, 2018-05-14:
Let me guess, you use HASS.io?

wongnam, 2018-05-15:
I do not use HA.

The issue will still be happen event it does not have connected to Mqtt or stand alone.

Frogmore42, 2018-05-15:
If it is a very quick on/off/on, I would not be surprised at all.  When you reset a MCU, its outputs become inputs/nothing before they become outputs again.

Normally, the device shouldn't be resetting, so this is not a problem.  While there are things iTead could have done to eliminate this, none are easy or free.

ascillato, 2018-05-15:
Hi,

I have also a Sonoff Basic with Tasmota v5.13.1c and do that flickering on reboot.

That behaviour does not come from the Tasmota Firmware. It is part of the normal boot of the ESP8285 MCU.

When the MCU is starting all the GPIOs are LOW except for D6, D7, D3, D4, TX and RX that are HIGH

The relay on the Sonoff Basic is connected to D6.

That's why this is happening.

This behaviour on boot is explained on the ESP8285 datasheet from espressif website.

ascillato, 2018-05-15:
Hi, this is because it is set to have some GPIOs to LOW on Boot. On ESPEasy they have selectable that GPIO Boot state. Please, if you want you can investigate it further to know if can be added that feature to Tasmota to make it save the actual state to the boot state.

or78rXxZ, 2018-05-15:
Did you try WifiConfig 4?

wongnam, 2018-05-15:
@or78rXxZ No, I am using Wificonfig 5 to prevent restart if wifi was lost.
My topic is mention about it flickering while resting the sonoff.
I've heard that the flickering issue is caused by SDK Lib due to GPIO must go to low and go to high during the reset progress.

curzon01, 2018-05-15:
Status 0 info always would be helpfull (then all hre a know which lib you are using and what's your configuration), also about what SDK Lib version you are talking.

closed 2018-05-16 by wongnam

RyanEwen, 2019-03-03:
I have setup a Sonoff Basic flashed with Tasmota connected to my garage door opener and use GPIO14 as Switch1 with a reed switch to determine if my garage door is closed or not. My concern is that if the power goes out or if I update the firmware my garage will open as the Sonoff restarts.

Is there any way to prevent this?

Frogmore42, 2019-03-03:
Please don't use a closed issue to ask a question.

The question you are asking, while it might seem so,  is not simple.  This is actually quite a complex topic and doing it right 100% of the time is very hard.  If all you care about is most of the time, it is easy and probably works just fine the way you have it.