arendst/Tasmota #2731: Device appears to be dead after flash to 5.13

pmknowles, 2018-05-14:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/) **(No in maintenance mode)**
- [X ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [X ] Hardware used : Sonoff TH-10 with AM2301 sensor
- [ ] Provide the output of command ``status 0`` :

**(Please, remember to close the issue when the problem has been addressed)**
The device was flashed over the air a few months ago.
Today I tried to flash from 5.9 to 5.13 using the web interface (using the default web address in the dialogue not uploading a binary). It said uploading (no miscompare error) so I waited a couple of minutes and then clicked Main Menu and got a 'Can't reach page' error.
The device shows no LEDs. I've tried turning power off and also pressing the button for 40 seconds.

Is there anyway to resurrect it or is it bricked?

Jason2866, 2018-05-14:
Many users have problems with version 5.13 which is compiled wirh arduino sdk 2.4.0
@arendst will compile a version with arduino sdk 2.3.0 this week.
Your device is provably not dead. You have to flash ist via usb - serial adapter.
You can use v.5.12 in the meantime which works for most all users

reloxx13, 2018-05-14:
for sure this issue has nothing to do with 2.4.0.

the flashing failed for whatever reason.

those esp devices are never death if flashing failed, you can always flash by usb, doesn't matter if their is a fw on the esp or not.

reloxx13, 2018-05-14:
connect to usb and see what the serial logs says if their is one outputting to see what is hapaning on the device.

Jason2866, 2018-05-14:
@reloxx13 
I dont think flashing failed. Device is unresponsible (because of 2.4.0) after restart.
But makes no difference just reflash with a good Tasmota firmware version

pmknowles, 2018-05-14:
I'm running openHAB under Windows so not sure if the 2.3/2.4 issue applies as it downloaded from the domus site - I didn't compile it. I will open it up tomorrow and see if the pins are available. THanks for the help.

mpapierski, 2018-05-24:
+1

Seems like I have simliar issue. I tried OTA with 5.13.0 on Sonoff Basic, failed, I forgot about the case until there was power outage. Most of my Sonoff Basic devices came back online just not the ones I tried to OTA.

After connecting serial port to the device they're responding with "Exception(0)". Now `esptool` can't connect to them, or esptool reports that they're reporting garbage in flash mode. Sorry for lack for exact log.

ascillato, 2018-06-13:
Hi,

If your issue was solved, please close it. Thanks :+1:

pmknowles, 2018-06-13:
I resolved it by flashing via the pins. It took several attempts to get it into flash mode.

closed 2018-06-13 by pmknowles