arendst/Tasmota #2734: KNX mapping to GA

Winni66, 2018-05-15:
I'm testing with sonoff-knx.bin version 5.13.1

Binary mapping for relays is working properly.

But I have found some strange behaviour for Sonoff TH16:
In Data to Send to Group Address it's not possible to map humidity (always temperature is shown after saving)
In Group Adresses to Receive Data from I get always temperature on GA I had mapped to reply humidity

For Sonoff POW:
There is no possibility to map information about power consumption.

Datatype for KNX:
It seems that DPT7 is used for temperature, therefore the value cannot be used directly from KNX devices, which normaly expect DTP9 if there are decimals. Maybe it might be possible to select KNX datatype too?

Nevertheless, it's a great firmware for Sonoff devices. Let's hope that OTA-setup for Tasmota will be available again in the future.

ascillato, 2018-05-15:
Hi,

> In Data to Send to Group Address it's not possible to map humidity (always temperature is shown after saving)

Yes, that was a bug but it is solved now on the Sonoff-Tasmota development branch. You can build a firmware with the features you need or wait for next release that is going to be out soon.

Also for testing I have compiles some KNX firmwares v5.13.1b [here](https://github.com/ascillato/Sonoff-Tasmota_KNX/releases/tag/v5.13.1b) if you want to try.

> There is no possibility to map information about power consumption.

Nop, It is not implemented yet.

> It seems that DPT7 is used for temperature

It uses DPT9.001 (2 bytes float). where you see that? DPT7.001 is a long integer (0-65535)

> Let's hope that OTA-setup for Tasmota will be available again

What do you mean? OTA is working.

Winni66, 2018-05-15:
Thanks for information, I upgraded to 5.13.1b:
DTP9 you are right, don't know why it didn't worked before here, I must have made something wrong.
Humidity is sent properly now.
But: Up to now it was possible to request values from KNX-side, but with the new firmware there is no answer from Sonoff device anymore.

OTA-Setup: I meant update new original firmware to Tasmota, OTA is working properly if you have Tasmota installed already

Winni66, 2018-05-15:
Sorry, didn't want to close, It's my first issue an Github ;-)

ascillato, 2018-05-15:
Hi,

I have a Temp and Humidity sensor sending data over KNX without problem.

Please, check your configuration.

As example, mine is:

[image]

Winni66, 2018-05-15:
I have only 5.13.1b, but the same setup...
It worked properly for temperature in 5.13.1 (only humidity issue).
In 5.13.1b I get no feedback anymore on requests (but humidity info). Unfortunately I'm not able to build 5.13.1c.
Think I have to way for next release ;-)
Thanks a lot for feedback!

ascillato, 2018-05-15:
between v5.13.1b and v5.13.1c there are not differences related to KNX.

That's why should be a config problem. I can read temp and humidity.

Winni66, 2018-05-15:
I just tried to upload 5.14 release, now the device get's IP but I cannot ping and cannot reach web-interface.
So it's dead for the next days. Perhaps I can access again when I'm at home and bring it nearer to the access point.
I'm only able to upload by OTA :-(

Winni66, 2018-05-15:
Wrong information from remote, my wife says, that there is now a Sonoff-Network and she tried to setup Network again (access to network ID and password when connecting to this network with mobile). But after save no connect to my network, restart with Sonoff-Network again ( I trust her, that she used correct information).
So there will be no update on this topic until next tuesday

ascillato, 2018-05-16:
Hi, can you test the actual v5.14.0a that has an improvement on KNX ?

If you don't know how to compile the last version (v5.14.0a), you can find them also at https://github.com/ascillato/Sonoff-Tasmota_KNX/releases

Thanks.

Winni66, 2018-05-16:
No I can't, because I don't get WiFi connection. The device is fixed in the garden, I have to investigate next week when I'm at home.

Winni66, 2018-05-28:
Back again, I had to buy USB connector to flash directly.
I'm using now 5.14.0a with core 2.3.0 because of my Wifi issues.
Comments to KNX functionality:
- Menu for TH16 with Temp/Humidity sensor is fine now, BUT KNX request for humidity is still sending temperatur. Updates because of changes are sent properly !
- POW enhancement (not tested all of them), menu is ok, BUT after switching the updates of currency and voltage are quite slow, is there any opportunity to optimize? Change of status behaves properly and fast !
- KNX communication enhancement: is working as described, but be careful -> may lead to ignore fast switching commands

Winni66, 2018-06-06:
I installed 5.14.0b and did a new KNX setup.
22:43:15 HTP: Configure KNX
22:43:15 KNX: aktiviert: 1, Communication Enhancement: 0
22:43:15 KNX: Physical Address: 2.0.3 
22:43:15 KNX: GA: 3
22:43:15 KNX: GA #1: Ausgang 1 zu 8/5/2
22:43:15 KNX: GA #2: Feuchtigkeit zu 8/5/20
22:43:15 KNX: GA #3: Temperatur zu 8/5/21
22:43:15 KNX: CB: 3
22:43:15 KNX: CB #1: 8/5/2 zu Ausgang 1
22:43:15 KNX: CB #2: 8/5/20 zu Reply Feuchtigkeit
22:43:15 KNX: CB #3: 8/5/21 zu Reply Temperatur

When requesting 8/5/20 I still get temperature !
This happens on console:
22:44:41 KNX: Received from 8.5.20 Befehl Read: 0 zu Reply Temperatur
why is it temperature ?
It'S properly working when data is sent periodically:
22:48:15 KNX: Temperatur sent to 8.5.21 
22:48:15 KNX: Feuchtigkeit sent to 8.5.20 
22:48:15 RSL: SENSOR = {"Time":"2018-06-06T22:48:15","SI7021":{"Temperature":27.50,"Humidity":51.60},"TempUnit":"C"}
22:48:16 CFG: in Flash gespeichert am F5, zählen 52, Bytes 3584

sending of information would be great if one could select by delta values (smaller teleperiod will cause KNX traffic!) At least for all values there should be an automatic update after toggeling relais...

ascillato, 2018-06-06:
Hi,

> When requesting 8/5/20 I still get temperature !

Will investigate.

> sending of information would be great if one could select by delta values (smaller teleperiod will cause KNX traffic!) At least for all values there should be an automatic update after toggeling relais...

For sending updates only by delta you can make that by rules. There is now the `KnxTx_Val `command. Let me first update the wiki with the new commands for using KNX on rules.

ascillato, 2018-06-08:
Fixed Bug on sensors reply.

Thanks for reporting :+1:

ascillato, 2018-06-13:
Hi, can you test the actual v6.0.0a that has that bug solved on KNX ?

If you don't know how to compile the last version (v6.0.0a), you can find it also at https://github.com/ascillato/Sonoff-Tasmota_KNX/releases

Also There is the Pull Request #2955 for Tasmota to add arithmetic commands for Tasmota. With these new commands you can implement the Delta Values you want. There is an example in that PR for that.
This PR is already implemented in https://github.com/ascillato/Sonoff-Tasmota_KNX/releases

Thanks.

Winni66, 2018-06-13:
Request-Bug is solved in your release, thanks a lot. 
I have no experience on rules. I tried the following, but it didn't seems to work:

rule
on SI7021#temperature>%var1% do backlog var1 %value%; publish stat/sonoff/temp %var1%; var2 %var1%; add1 2; sub2 2 endon
on SI7021#temperature<%var2% do backlog var2 %value%; publish stat/sonoff/temp %var2%; var1 %var2%; add1 2; sub2 2 endon
on SI7021#humidity>%var3% do backlog var3 %value%; publish stat/sonoff/humidity %var3%; var4 %var3%; add3 2; sub4 2 endon
on SI7021#humidity<%var4% do backlog var4 %value%; publish stat/sonoff/humidity %var4%; var3 %var4%; add3 2; sub4 2 endon

Thanks

ascillato, 2018-06-13:
What error is displayed?

The rule as you posted need to be all together as:

`rule on SI7021#temperature>%var1% do backlog var1 %value%; publish stat/sonoff/temp %var1%; var2 %var1%; add1 2; sub2 2 endon on SI7021#temperature<%var2% do backlog var2 %value%; publish stat/sonoff/temp %var2%; var1 %var2%; add1 2; sub2 2 endon on SI7021#humidity>%var3% do backlog var3 %value%; publish stat/sonoff/humidity %var3%; var4 %var3%; add3 2; sub4 2 endon on SI7021#humidity<%var4% do backlog var4 %value%; publish stat/sonoff/humidity %var4%; var3 %var4%; add3 2; sub4 2 endon`

And remember to turn on rules also

`rule 1`

ascillato, 2018-06-13:
If that does not work, please open a new issue and we will see there.

As you confirmed that this issue (KNX mapping) is solved. I'm closing it. Please, reopen it, if you find anything else related to this. Thanks :+1:

closed 2018-06-13 by ascillato2