arendst/Tasmota #2738: KNX ignored requests

gtortone, 2018-05-15:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x] Hardware used : Wemos D1 mini
- [x] Provide the output of command ``status 0`` :
```
20:23:04 MQT: sftmp01/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["sftmp01"],"Topic":"sftmp01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
20:23:04 MQT: sftmp01/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:23:09","StartupUTC":"2018-05-15T18:59:55","Sleep":0,"BootCount":29,"SaveCount":320,"SaveAddress":"F9000"}}
20:23:04 MQT: sftmp01/stat/STATUS2 = {"StatusFWR":{"Version":"5.13.1c","BuildDateTime":"2018-05-15T17:51:31","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
20:23:04 MQT: sftmp01/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"10.1.1.1","LogPort":514,"SSId":["gem-house","gem-house-gw"],"TelePeriod":10,"SetOption":["02008009","55818000"]}}
20:23:04 MQT: sftmp01/stat/STATUS4 = {"StatusMEM":{"ProgramSize":523,"Free":480,"Heap":13,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
20:23:04 MQT: sftmp01/stat/STATUS5 = {"StatusNET":{"Hostname":"sftmp01","IPAddress":"10.1.1.6","Gateway":"10.1.1.254","Subnetmask":"255.255.255.0","DNSServer":"10.1.1.20","Mac":"A0:20:A6:02:1E:C4","Webserver":2,"WifiConfig":2}}
20:23:04 MQT: sftmp01/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.1.1.1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_021EC4","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:23:04 MQT: sftmp01/stat/STATUS7 = {"StatusTIM":{"UTC":"Tue May 15 19:23:04 2018","Local":"Tue May 15 20:23:04 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:08","Sunset":"20:24"}}
20:23:04 MQT: sftmp01/stat/STATUS10 = {"StatusSNS":{"Time":"2018-05-15T20:23:04","Switch1":"OFF","DHT11":{"Temperature":23.0,"Humidity":37.0},"TempUnit":"C"}}
20:23:04 MQT: sftmp01/stat/STATUS11 = {"StatusSTS":{"Time":"2018-05-15T20:23:04","Uptime":"0T00:23:09","Vcc":2.995,"POWER":"OFF","Wifi":{"AP":1,"SSId":"gem-house","RSSI":44,"APMac":"10:13:31:50:6F:2D"}}}
20:23:04 MQT: sftmp01/tele/STATE = {"Time":"2018-05-15T20:23:04","Uptime":"0T00:23:09","Vcc":2.995,"POWER":"OFF","Wifi":{"AP":1,"SSId":"gem-house","RSSI":46,"APMac":"10:13:31:50:6F:2D"}}
```
Hi,
I start to play with KNX and I'm using knxd (https://github.com/knxd/knxd) on my Linux machine and
knxtool utility to send some KNX commands...

I configured a Wemos D1 mini with Sonoff-Tasmota as following:

[image]

I launched KNXd with following command line:

```
server/knxd -e 0.0.1 -D -R -S -E 15.15.0:8 -i -I eth0
```

I also added a static route for multicast network 224.0.23.12:

```
route add -net 224.0.23.12 netmask 255.255.255.255 eth0
```

Using knxtool I'm able to turn off/on the relay on D7 pin

```
tools/knxtool off ip:127.0.0.1 2/2/8
or
tools/knxtool on ip:127.0.0.1 2/2/8
```

The problem is that 'sometimes' if I quickly switch on and off the relay the status of relay does not 
change from last status... it seems that the KNX request is ignored from Sonoff-Tasmota;

I verify that KNX frame is received correctly by KNXd because I'm using knxtool in a different
shell to 'spy' KNX messages...

```
tools/knxtool vbusmonitor1 ip:localhost
```

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-05-15:
Hi,

First of all, you have a problem in your config.

If you send Temperature AND Humidity to the same group address you will not be able to know which one you are receiving there. You need to set different Group Address for those.

Just as example I share a piece of my configuration.yaml from HomeAssistant:

```yaml
sensor:  
  - platform: knx
    name: Outside Temperature
    address: '4/1/1'
    type: 'temperature'
  - platform: knx
    name: Outside Humidity
    address: '4/1/2'
    type: 'humidity
```

There you see that HomeAssistant is receiving data from different group address. You need to separate those, if not you will see only the last value sent that will be humidity.

> The problem is that 'sometimes' if I quickly switch on and off the relay the status of relay does not
> change from last status... it seems that the KNX request is ignored from Sonoff-Tasmota

Multicast have some issues on some wifi routers and much more with wifi repeaters. The problem is not that Tasmota is ignoring. Is that is not receiving the telegram. I understand that you can see some telegrams that should be sent to Tasmota on other device but also that depends on how that device manage the IGMP with the router. ESP is very picky with some IGMP Router versions and the router forgets to send data to the ESP. 
Besides this, you are using a compiled version with `"Core":"2_4_1"` that has some bugs and is not very stable. That's why it is recommended to use v2.3.0 that is more stable.

Try to see if in your wifi router's configuration you have:
* Multicast: on
* IGMP v3: on
* IGMP Snoop: off

That should improves the communication

gtortone, 2018-05-16:
Thanks a lot for your explanations ! I have some elements to further investigation on this subject...

closed 2018-05-16 by gtortone

ascillato, 2018-05-16:
:+1:

> if I quickly switch on and off the relay the status of relay does not change

Soon I will release a little enhacement for KNX communication for that.

ascillato, 2018-05-16:
Ah, If you press the button and that one is configured as multipress, it will not react very fast to changes waiting for you to press fast several times. See Button Multipress on the wiki.

gtortone, 2018-05-17:
I tested it with 5.14.0a (git pull 5 minutes ago) and core 2.3.0 but the behaviour is always the same I described in the issue... sorry about this bad news...

it could be a fault of my router that is 'branded' from my ADSL company and it has not options about IGMP and Multicast (or they are hidden from home-user)