arendst/Tasmota #2740: [Sonoff Basic] How to get TRIGGER (3.3v for 1 second) to only send MQTT message on GPIO14 ?

hvanmegen, 2018-05-15:
I'm trying to figure out a way to have a simple Radar sensor (The RCWL-0516) that goes high (3.3v) on GPIO14 only send a message to my MQTT broker and NOT switch my relay (that powers some LED lighting).

Having it set up as button 2, 3 or 4 doesn't send an MQTT message, and having it set up as a switch doesn't work either; this only switches my relay on and off whenever I move...

I tried looking it up in the documentation, but I'm not entirely sure how to handle this.. isn't there a simple sensor like a PIR that pulls high when it detects motion and then drops back to low a second or so later after the trigger has been sent?

Which one should I use? I can't seem to find the correct one.

hvanmegen, 2018-05-15:
Note: all relay switching logic is handled externally by MQTT clients; I only want to use the Sonof basic as a simple MQTT enabled relay AND a single GPIO pin MQTT message sender whenever the pin goes high and back low again

wongnam, 2018-05-15:
Wiki has all about what you need

https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic

Think about switchtopic and switchmode

hvanmegen, 2018-05-16:
Aaah!! I have it switch switch2 to MQTT only by changing the SwitchTopic and SwitchMode!

Perfect, thanks for the help!

curzon01, 2018-05-18:
@hvanmegen - beside oneself with joy don't forget to close issues if it it solved - you opened it, you close it :wink:

closed 2018-06-08 by ascillato2

danst0, 2019-01-12:
The wiki link is dead and I still do not get it. I changed the GPIO14 now to Switch2 but all messages in MQTT/console are still the same and power goes low after two seconds or so.

meingraham, 2019-01-12:
@ascillato - can this page -(https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic) be restored?