arendst/Tasmota #2741: PIR help working Great but stop it sending the OFF command for a delay time

MYLE-01, 2018-05-16:
I have a "Mini IR Infrared Pyroelectric PIR"
the 3.3 vcc version

have setup as per the wemos Wiki  happy camper

It turn ON when I walk into room   but then It send the OFF command couple of sec later
is there a way to put a delay in to send the off command say 5 min later ??

TOBYBEAR69, 2018-05-16:
The Mini PIRs typically do not have manual adjustments for delay time and sensitivity so you have to do it via software (which I would recommend even if you had hardware adjustments available).  Since the main point of the Tasmota firmware is to provide functionality via MQTT messaging I highly suggest you get familiar with Node Red.  It is a relatively simple drag and drop interface for creating custom automatons and can greatly enhance the flexibility of you automations.  It is free and open source software.  See nodered.org for more information.  There are also a decent amount of YouTube videos that will provide you with the basics.  You will need three nodes at a minimum:  MQTT Inlet that passes in a message when the sensor fires, a Delay Node that you can set to however long you want the light to stay on, and then an MQTT Outlet Node to send a message to turn off the light when the Delay Node fires its message after your selected time interval.  A possible second option is to use the recently added rules functionality of the Tasmota firmware.  Have not thought it all the way thru because I think you will be better off in the long run going the NodeRed route.....but it would go something like....create a rule that only responds to switch on messages (ignores the off) and then set a timer that triggers another rule that turns it off.  You probably need to use SwitchTopic set to 2 so that you control the relay manually (see commands topic).

MYLE-01, 2018-05-16:
@TOBYBEAR69  thanks

Have Home Assistant running with Just started playing with node red was just starting to think about going down that track

MYLE-01, 2018-05-16:
Got it working in node red
[image]

here is the Node red Commands
[{"id":"52127a09.656d34","type":"mqtt in","z":"52e056f1.f36ff8","name":"Office Motion","topic":"stat/SONOFF-FAN/POWER3","qos":"2","broker":"d3f7744e.a9cb78","x":110,"y":140,"wires":[["23361fe8.1754b"]]},{"id":"9e920ab6.f68048","type":"debug","z":"52e056f1.f36ff8","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":430,"y":100,"wires":[]},{"id":"cfe6a421.aa0598","type":"delay","z":"52e056f1.f36ff8","name":"","pauseType":"delay","timeout":"1","timeoutUnits":"minutes","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":420,"y":320,"wires":[["e681dcb8.4aa1c"]]},{"id":"e7c5d7f4.3c23e8","type":"mqtt out","z":"52e056f1.f36ff8","name":"","topic":"nodered/motion","qos":"","retain":"","broker":"d3f7744e.a9cb78","x":840,"y":80,"wires":[]},{"id":"23361fe8.1754b","type":"function","z":"52e056f1.f36ff8","name":"","func":"if (msg.payload ==\"ON\") {\n    msg.reset = true;\n    msg.payload = \"ON\"\n}\nelse{\n    \n    msg.payload = \"ON\"\n}\n\nreturn msg;","outputs":1,"noerr":0,"x":270,"y":220,"wires":[["e7c5d7f4.3c23e8","9e920ab6.f68048","cfe6a421.aa0598"]]},{"id":"96a290b2.7908f","type":"mqtt out","z":"52e056f1.f36ff8","name":"","topic":"nodered/motion","qos":"","retain":"","broker":"d3f7744e.a9cb78","x":840,"y":140,"wires":[]},{"id":"e681dcb8.4aa1c","type":"function","z":"52e056f1.f36ff8","name":"","func":"if (msg.payload ==\"ON\") {\n    msg.payload = \"OFF\"\n}\nelse{\n    msg.payload = \"OFF\"\n}\n\nreturn msg;","outputs":1,"noerr":0,"x":590,"y":240,"wires":[["96a290b2.7908f","9e920ab6.f68048"]]},{"id":"d3f7744e.a9cb78","type":"mqtt-broker","z":"","name":"","broker":"localhost","port":"1883","clientid":"","usetls":false,"compatmode":true,"keepalive":"60","cleansession":true,"willTopic":"","willQos":"0","willPayload":"","birthTopic":"","birthQos":"0","birthPayload":""}]

its even got a reset so as it see more motion  the OFF Delay is reset each time.

closed 2018-05-16 by MYLE-01