arendst/Tasmota #2744: Sonoff Dual R2: Multiple restart using rules

gd-99, 2018-05-16:
Program Version | 5.13.1. Core/SDK Version | 2_4_1/2.2.1(cfd48f3)
I have set-up a timer (16) to restart the device using rules:
**Timer:**
`Timer16 {"Arm":1,"Mode":0,"Time":"12:00","Days":"1111111","Repeat":1,"Action":3}`
**Rule:**
```
Rule on clock#timer=16 do restart 1 endon
Rule 1 - enable rule
Rule 5 - enable oneshot
```
This nearly works. The device restarts as expected at the time set in timer 16. However, as soon as the device restarts and the clock resets to the correct time the device then restarts again often this happens 6 times before it stops restarting and carries on functioning correctly.

Connecting up to a serial console I see these lines towards the end of the usual start up message:
00:00:00 Project sonoff Sonoff#2 (Topic sonoff, Fallback DVES_XXXXXX, GroupTopic sonoffs) Version 5.13.1-2_4_1
00:00:00 WIF: Connecting to AP1 <my SSID> in mode 11N as sonoff-xxxx...
00:00:12 WIF: Connected
00:00:12 DNS: Initialized
...
...
**13:48:53 RUL: CLOCK#TIMER=16 performs "restart 1"**
**13:48:53 RSL: RESULT = {"Restart":"Restarting"}**
**13:48:54 APP: Restarting**
(ignore the time, I did this to test the issue so not at 12:00)

This happens with Rule 5, oneshot enabled or not. Using the Web interface and clicking on the red restart button, causes the device to reset once - i.e no multi resets as above.

I think (looking for confirmation) the issue is this sequence:
1) the timer using the rule, resets the device as expected. 
2) after restart the device gets the correct time from the internet
3) the timer function looks for up and coming timer events 
4) as the device restarts within less than a minute so the "current" timer event is still "in time" and the timer re-runs the restart rule.
5) repeat the above until the timer has bumped on beyond the time for the reset timer event.

Once passed the cycle of resets the device is as solid as a rock.

I have also tried setting the rule as:
`Rule` on clock#timer=16 do restart 99 endon 
so that I do not keep writing to flash just for my timed restart - assume this is the best way of running a routine restart /reboot.

Looking for pointers on how to stop the cycle of restarts please.

arendst, 2018-05-16:
Your sequence is spot on!

The one-shot is not handled by a restart. I could change the code by using a memory location which survives the restart and if one-shot is set it should not execute again.

I'll investigate.

arendst, 2018-05-16:
Tested and Fixed in development v5.14.0a.

gd-99, 2018-05-16:
Confirmed working on Sonoff Dual R2 with thanks.

I did however have some issues trying to compile v5.14.0a using the Arduino IDE.
At compile I received this error: TasmotaSerial::TasmotaSerial(uint8_t&, uint8_t&, int), then a whole load of trace information.

I managed to successfully compile and then upload by commenting out the whole **Serial Sensor** section of **user_config.h**.

I don't use the serial sensors (yet) so it doesn't affect me however thought it useful to report my experience.

Many thanks for sorting out the timer multi restart so quickly.

arendst, 2018-05-16:
You need tasmotaserial-2.0.0 (and delete the older version) from the lib folder.

Anyway. It works so Iclose this issue.

gd-99, 2018-05-16:
Ok thanks will check this tomorrow.

Closing issue

closed 2018-05-16 by gd-99