arendst/Tasmota #2745: Sonoff Dual R2 instability using IwIP: V2 Lower Memory - restart cycle

gd-99, 2018-05-16:
Tasmota: 5.13.1
Arduino IDE: 1.8.5 (portable)
ESP8266 Community: 2.4.1

Following the [Tasmota wiki guide](https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE) to compile using the Arduino IDE, at the point Configuring Arduino IDE, [ESP8266 Board version 241](https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE#esp8266-board-version-241-1) the advice is to compile with the IwIP variant set to v2 Lower Memory. I found following this advice my device was very unstable and would go into a restart cycle.

By changing the IwIP variant to **v1.4 Higher Bandwidth**, the Sonoff dual R2 would start up normally and is rock solid without any hint of the restart cycling seen when using the V2 Lower Memory. 

I do not know what memory hit I am taking by using the higher bandwidth. I also didn't try using the V2 Higher Bandwidth option.

My symptom was very clear and not as described in issue #2605 but wonder if changing the IwIP variant would help that issue as well?

arendst, 2018-05-16:
Thnx for sharing.

Just did a compile and noticed v1.4 program size is even 1k smaller.

For those wanting to test this in platformio the following change needs to be made to platformio.ini
```
build_flags =
  -Wl,-Tesp8266.flash.1m0.ld
  -DVTABLES_IN_FLASH
;  -DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
;  -DUSE_CONFIG_OVERRIDE
```
This will enable the default lwIP v1.4 Higher Bandwidth option.

reloxx13, 2018-05-16:
note:
those are all options:

```
; -DPIO_FRAMEWORK_ARDUINO_LWIP_HIGHER_BANDWIDTH    ;v1.4 Higher Bandwidth (default)
; -DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY         ;v2 Lower Memory
; -DPIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH   ;v2 Higher Bandwidth
```

im using DPIO_FRAMEWORK_ARDUINO_LWIP_HIGHER_BANDWIDTH and cannto complain (together with 2.3.0).

arendst, 2018-05-18:
Running tests for some days now and they are just fine.

I will enforce the default option in pio and Arduino IDE to be PIO_FRAMEWORK_ARDUINO_LWIP_HIGHER_BANDWIDTH for 2.4.1

closed 2018-07-10 by ascillato2