arendst/Tasmota #2747: RGB(W) effects and default startup color/brightness

hulkhaugen, 2018-05-16:
Sorry if this already exists, but i haven't been able to find anything in here about it nor with google. I've seen a few forum posts about people adding effect_list: 1, 2, 3, etc... in HA, but no reference to how that works. I have a few RGB WiFi bulbs that are currently flashed with stegenhof's AiLight firmware, but i'm not happy with it's current limited ability. I really like Tasmota for my Sonoff devices and think/hope it also (can) support these features.

1. First of all, i'm assuming that i have to soldier wires to the bulbs again to flash this firmware. Correct?

2. Are effects included in Tasmota, such as rainbow etc? I found a few references to NeoPixelBus, but that seems to be a different project. Is it included, or do i have to include it manually? Or isn't it related at all?

3. Can i set a default color and brightness for when i power on the device? If i go to bed with some sexy colors or effects on the night stand light, i would want it to go back to warmwhite and a given brightness when i power it on in the morning. Is that possible?

ascillato, 2018-05-16:
Hi,

AiLight is supported, see wiki (https://github.com/arendst/Sonoff-Tasmota/wiki)

> First of all, i'm assuming that i have to soldier wires to the bulbs again to flash this firmware. Correct?

It depends. Some devices have pins to connect to. See Hardware preparation on wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Hardware-Preparation)

> Are effects included in Tasmota, such as rainbow etc? I found a few references to NeoPixelBus, but that seems to be a different project. Is it included, or do i have to include it manually? Or isn't it related at all?

Yes, effects are there. See Commands on wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Commands)

> Can i set a default color and brightness for when i power on the device? If i go to bed with some sexy colors or effects on the night stand light, i would want it to go back to warmwhite and a given brightness when i power it on in the morning. Is that possible?

Yes, you can. See `SetOption15`, `FADE`, `SCHEME 1` and other Commands on wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Commands)

For your reference (see in wiki) the configuration for **Home Assistant**, in _configuration.yaml_ can be something like:

(if you set on Tasmota the mqtt topic as **led** for example - can be anything you like)

```yaml
light:
# My Sexy Bedroom Light
    - platform: mqtt
      name: "RGB Bedroom Light"
      retain: true
      state_topic: "stat/led/POWER"
      command_topic: "cmnd/led/POWER"
      availability_topic: "tele/led/LWT"
      qos: 1
      payload_on: "ON"
      payload_off: "OFF"
      payload_available: "Online"
      payload_not_available: "Offline"
      optimistic: false
      brightness_state_topic: "stat/led/Dimmer"
      brightness_command_topic: "cmnd/led/Dimmer"
      brightness_value_template: "{{ value_json.Dimmer }}"
      brightness_scale: 100      
      rgb_state_topic: "stat/led/Color"
      rgb_command_topic: "cmnd/led/Color2"
      rgb_command_mode: hex
      rgb_command_template: "{{ '%02x%02x%02x' | format(red, green, blue)}}"
      effect_state_topic: "stat/led/Scheme"
      effect_command_topic: "cmnd/led/Scheme"
      effect_value_template: "{{ value_json.Scheme }}"
      effect_list:
          - 0
          - 1
          - 2
          - 3
          - 4
          - 5
          - 6
          - 7
          - 8
          - 9
          - 10
          - 11
          - 12
```

ascillato, 2018-05-16:
If this solves your issue, please close it. Thanks.

ascillato, 2018-05-16:
If you find in the wiki that some information related to the AILight need to be improved with images and so on, please add it. Thanks.

hulkhaugen, 2018-05-17:
Thanks, I'm sure I got what I needed. My hardware isn't available at the moment, but will try when I get back home. Again, thanks so much!

closed 2018-05-17 by hulkhaugen

Kepete, 2019-09-07:
Old issue but I would like to reopen it as I have the same question as your 3rd. 
> Can i set a default color and brightness for when i power on the device? If i go to bed with some sexy colors or effects on the night stand light, i would want it to go back to warmwhite and a given brightness when i power it on in the morning. Is that possible?

I want the light to go back to white when I toggle it from my physical switch. How to configure tasmota like this? I had a look at the proposed SetOption15, FADE, SCHEME 1 and other Commands on wiki but didn't manage to ge tit working.

ascillato, 2019-09-08:
You can use a rule

g-off-white, 2019-11-06:
> You can use a rule

I have been trying to do this with no success.

It should be 
rule1 on system#boot do color ffffffff endon
{"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":472,"Rules":"on system#boot do color #ffffffff endon"}
I have also tried using switch1#boot but the light insists on coming up as color ffffff00 when powered up.  This should be trivial and these bulbs are useless without any white LEDs.

ascillato, 2019-11-06:
@g-off-white 

If you run in the console the command `color #ffffffff`, it works as expected?

g-off-white, 2019-11-06:
Further
If the bulb has a rgb color and white values, only the rgb component is remembered through power cycle.
If the bulb color is set to a white only configuration it remembers it.

ascillato, 2019-11-06:
Please, open a new issue with the troubleshooting template **completed** in order to have all the information needed to find the problem.

g-off-white, 2019-11-06:
The color functionality is working well from the console.  After power cycling color #ffffffff  comes back as color #ffffff00

ascillato, 2019-11-06:
Good, you are narrowing the problem. Please, write all this in a new issue with all the requested information of the troubleshooting template.

g-off-white, 2019-11-06:
I will raise a new issue about the startup color oddity.
Later ... time for bed.