arendst/Tasmota #2748: Wiring Directly to Esp8266ex on Sonoff

Joshncodesongithub, 2018-05-16:
Arduino IDE 1.8.5 Win 10
Sonoff Basic
ESP8266 2.4.0 
Tasmota 5.13.1
DSD TECH USB to TTL Serial Converter CP2102

It shows here https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Basic#serial-connection they soldered a wire directly to the esp8266ex on the Sonoff. Has anyone actually tried this and is it a more reliable way to flash the Sonoff? 

I can't get anymore Sonoffs to flash using a DSD TECH USB to TTL Serial Converter CP2102. Had one work with using pins and piece of paper to hold them in place. I connected jumper wires from the converter and flashed tasmota. However, all the others I've tried to flash won't show up in Arduino IDE under Ports. Keep getting error COM1 not connected. The read led on the converter flashes once when I connect power but doesn't stay on.

Joshncodesongithub, 2018-05-17:
Sorry I should have also mentioned that I soldered the header pins on in my last attempt to flash. It resulted in the same problem. Thinking of using pogo pins to make better contact. Looking at the holes in the sonoff you can see there's complete metal lining in each hole.

Frogmore42, 2018-05-19:
Since you are using Windows, go to Device Manager and find which COM port your adapter really is.  It is unlikely to really be COM1.  That is likely the issue you are seeing and not the wiring.  The error message seems to be saying it can't even open the COM port let alone talk to the device.

closed 2018-05-19 by Joshncodesongithub