arendst/Tasmota #2749: MQTT Websocket

loxK, 2018-05-16:
Some MQTT broker like mosquitto, do suppport MQTT over websocket. 

Wouldn't it be faster/ more responsive and less "ressource hungry" ?

Frogmore42, 2018-05-17:
What resources concern you?
What speed issues have you seen?
Web sockets aren't necessarily less resource intensive, nor faster.

Frogmore42, 2018-05-17:
Web sockets make it easy for browser apps to use networking/TCP.  Since Tasmota is not running in a browser, there is no reason to use web sockets.  To do so would require more resources and likely be slower.

Or, did you have something else in mind?

ionciubotaru, 2018-05-17:
try Paho, for me works like a charm for years.

```
client = new Paho.MQTT.Client(mqttBrokerURI, "browser-" + (new Date().getTime()));
client.onConnectionLost = onConnectionLost;
client.onMessageArrived = onMessageArrived;
function mqttConnect() {
  client.connect({
  //    userName: mqttUsername,
  //    password: mqttPassword,
    cleanSession: false,
    onSuccess: onConnect
  });
}
function onConnect() {
 console.log("connected");
 client.subscribe("+/+/LWT");
 client.subscribe("stat/+/RESULT");
 client.subscribe("tele/+/STATE");
}
function onConnectionLost(responseObject) {
  if (responseObject.errorCode !== 0) console.log("connection lost:" + responseObject.errorMessage);
}
.............................
```

closed 2018-07-10 by ascillato2

chkr1011, 2019-07-07:
I would also like to see an option to use MQTT over WebSockets instead. The reason is that some brokers are only available over WebSockets. 

For example: When hosting an Azure App Service only HTTP endpoints are available and thus only MQTT over WebSockets will work. The ESP32 IDF already supports MQTT over WebSockets.

edwinspire, 2023-08-05:
Does Tasmota currently support MQTT over WebSocket?

sfromis, 2023-08-05:
No, Tasmota uses the standard MQTT protocol.