arendst/Tasmota #2753: [Update Tasmota WIKI] Wemos D1 Mini - BME680 sensor added

RaymondMouthaan, 2018-05-17:
Hello all,

Today I received my BME680 - Temperature, Humidity, Pressure and Gas Sensor. I've updated the [Tasmota Wemos Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos%20D1%20Mini) accordingly.

Cheers,
Ray

FabianHM, 2018-05-17:
What does the unit "Kilo Ohm" have to do with Gas? Is this Serial Output?

FabianHM, 2018-05-17:
@RaymondMouthaan : Yes, this is the correct anwear of my Question, but I meant something else ;-)
I just ordered this sensor for my project Heating-Room. I have planned with the sensor to monitor whether possibly gas leaks in the heating rom and this is the sensor apparently not suitable.

My hope was an output similar to the rove - multichannel gas sensor (which was too expensive for me)
https://github.com/arendst/Sonoff-Tasmota/pull/1245
[image]
Does anyone have an idea how I can use the BME 680 for my needs?
Greetings Fabian

closed 2018-06-04 by ascillato

moneybag, 2018-10-04:
What about IAQ ?

djscolla, 2019-08-02:
Please DELETE the photo of your fantasy. Because in google image this photo is in 1° page. 
And I buy it! For nothings... ty