arendst/Tasmota #2754: Tasmota 15.14.0a - MQTT not reachable after a router reboot

jeylites, 2018-05-17:
**Describe the bug**

A clear and concise description of what the bug is.

I had a couple T1 flashed to the latest 5.14.0a,  everything seem working until I had to reboot my wifi router. Soon after I realise these switches were not reachable on Homekit. I had to reset the switches only than it come back online. Never had this issue on version 15.13.1a

During the offline phenomenon, web browser, KNX, MQTT broker are all connected and working. Just can't control via home automation software.

Also, make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues and in the wiki
- [ ] Hardware used : 

Sonoff T1 touch

Core: 2.30

Tasmota: 5.14.0a  (May 16, 2017)
TasmotaMQTT platform

Uploader: Platform Io  V 1.26.1 OSX

- [ ] Provide the output of command ``status 0`` :

```
16:54:59 KNX: Received from 2.2.1 Command Write: 0 to Output 1
16:54:59 MQT: stat/jey1/RESULT = {"POWER1":"OFF"}
16:54:59 MQT: stat/jey1/POWER1 = OFF
16:54:59 KNX: Received from 2.2.1 Command Write: 0 to Output 1
16:54:59 MQT: stat/jey1/RESULT = {"POWER1":"OFF"}
16:54:59 MQT: stat/jey1/POWER1 = OFF
16:56:06 MQT: tele/jey1/STATE = {"Time":"2018-05-17T16:56:06","Uptime":"0T02:41:55","POWER1":"OFF","POWER2":"ON","POWER3":"ON","Wifi":{"AP":1,"SSId":"SMILE","RSSI":68,"APMac":"00:72:63:51:68:58"}}
17:01:06 MQT: tele/jey1/STATE = {"Time":"2018-05-17T17:01:06","Uptime":"0T02:46:55","POWER1":"OFF","POWER2":"ON","POWER3":"ON","Wifi":{"AP":1,"SSId":"SMILE","RSSI":68,"APMac":"00:72:63:51:68:58"}}
17:02:00 MQT: tele/jey1/UPTIME = {"Time":"2018-05-17T17:02:00","Uptime":"0T02:47:49"}
17:02:56 WIF: Connect failed as AP cannot be reached
17:02:56 WIF: Connecting to AP2 salt in mode 11N as jey1-4683...
17:03:03 WIF: Connect failed as AP cannot be reached
17:03:04 WIF: Connect failed as AP cannot be reached
17:03:04 WIF: Connecting to AP1 SMILE in mode 11N as jey1-4683...
17:03:11 WIF: Connect failed as AP cannot be reached
17:03:12 WIF: Connect failed as AP cannot be reached
17:03:12 WIF: Connecting to AP2 salt in mode 11N as jey1-4683...
17:03:19 WIF: Connect failed as AP cannot be reached
17:03:20 WIF: Connect failed as AP cannot be reached
17:03:21 WIF: Connecting to AP1 SMILE in mode 11N as jey1-4683...
17:03:27 WIF: Connected
17:06:07 MQT: tele/jey1/STATE = {"Time":"2018-05-17T17:06:07","Uptime":"0T02:51:56","POWER1":"OFF","POWER2":"ON","POWER3":"ON","Wifi":{"AP":1,"SSId":"SMILE","RSSI":70,"APMac":"00:72:63:51:68:58"}}
17:10:47 WIF: Connecting to AP1 SMILE in mode 11N as jey1-4683...
17:10:54 WIF: Connect failed as AP cannot be reached
17:10:55 WIF: Connecting to AP2 salt in mode 11N as jey1-4683...
17:11:02 WIF: Connect failed as AP cannot be reached
17:11:03 WIF: Connect failed as AP cannot be reached
17:11:03 WIF: Connecting to AP1 SMILE in mode 11N as jey1-4683...
17:11:07 MQT: tele/jey1/STATE = {"Time":"2018-05-17T17:11:07","Uptime":"0T02:56:56","POWER1":"OFF","POWER2":"ON","POWER3":"ON","Wifi":{"AP":1,"SSId":"SMILE","RSSI":100,"APMac":"00:72:63:51:68:58"}}
17:11:10 WIF: Connect failed as AP cannot be reached
17:11:11 WIF: Connect failed as AP cannot be reached
17:11:11 WIF: Connecting to AP2 salt in mode 11N as jey1-4683...
17:11:18 WIF: Connect failed as AP cannot be reached
17:11:19 WIF: Connect failed as AP cannot be reached
17:11:19 WIF: Connecting to AP1 SMILE in mode 11N as jey1-4683...
17:11:26 WIF: Connect failed as AP cannot be reached
17:11:27 WIF: Connect failed as AP cannot be reached
17:11:27 WIF: Connecting to AP2 salt in mode 11N as jey1-4683...
17:11:34 WIF: Connect failed as AP cannot be reached
17:11:35 WIF: Connect failed as AP cannot be reached
17:11:36 WIF: Connecting to AP1 SMILE in mode 11N as jey1-4683...
17:11:47 WIF: Connected
17:14:24 CMD: status 0
17:14:24 MQT: stat/jey1/STATUS = {"Status":{"Module":30,"FriendlyName":["JEY 1","JEY 12","JEY 13"],"Topic":"jey1","ButtonTopic":"0","Power":6,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
17:14:24 MQT: stat/jey1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T03:00:13","StartupUTC":"2018-05-17T06:14:11","Sleep":0,"BootCount":11,"SaveCount":190,"SaveAddress":"F6000"}}
17:14:24 MQT: stat/jey1/STATUS2 = {"StatusFWR":{"Version":"5.14.0a","BuildDateTime":"2018-05-16T23:31:46","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
17:14:24 MQT: stat/jey1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SMILE","salt"],"TelePeriod":300,"SetOption":["0A008009","55818000"]}}
17:14:24 MQT: stat/jey1/STATUS4 = {"StatusMEM":{"ProgramSize":465,"Free":536,"Heap":16,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
17:14:24 MQT: stat/jey1/STATUS5 = {"StatusNET":{"Hostname":"jey1-4683","IPAddress":"192.168.0.26","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"60:01:94:98:B2:4B","Webserver":2,"WifiConfig":4}}
17:14:24 MQT: stat/jey1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.200","MqttPort":1883,"MqttClientMask":"26","MqttClient":"26","MqttUser":"admin","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:14:24 MQT: stat/jey1/STATUS7 = {"StatusTIM":{"UTC":"Thu May 17 09:14:24 2018","Local":"Thu May 17 17:14:24 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":8}}
17:14:24 MQT: stat/jey1/STATUS10 = {"StatusSNS":{"Time":"2018-05-17T17:14:24"}}
17:14:24 MQT: stat/jey1/STATUS11 = {"StatusSTS":{"Time":"2018-05-17T17:14:24","Uptime":"0T03:00:13","POWER1":"OFF","POWER2":"ON","POWER3":"ON","Wifi":{"AP":1,"SSId":"SMILE","RSSI":74,"APMac":"00:72:63:51:68:58"}}}
```

**To Reproduce**
Steps to reproduce the behavior:

Try controlling Sonoff after a router reboot with a home automation software. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-05-17:
Hi,

Please, search if you have any other device with
`"MqttClientMask":"26","MqttClient":"26"`

It is recommended to leave those as default so the last part is going to be different for all devices. If you have the same values for other devices they will have a problem registering again to MQTT server.
For Example, mine are:
`"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3910F2"`

and that on the MQTT menu looks like:

[image]

`DVES_%06X` is the default and that automatically generates: `DVES_3910F2` that is an unique ID.

jeylites, 2018-05-17:
@ascillato 

I recompiled to reflect the changes suggested, still the same. The problem seems similar to #2760 

I downgraded to 5.13.1a and it's working now.

ascillato, 2018-05-18:
May be it is command `SetOption26`. See in wiki.

> SetOption26 | 0 / off | (default) Keep using POWER without postfix for single power devices

jeylites, 2018-05-18:
No joy, after a power reboot of MQTT broker, router, controller. All the sonoff touch flashed with 15.14.0a are not reachable via MQTT but are reachable from the web browser, MQTT broker despite having `SetOption26 | 0 / off ` . The one flashed 5.13.1a are working.

@ascillato Is it working for you? 

I even reinstall mosquito thinking there's an id conflict. The only way it seems to recover MQTT comms is a reset on of sonoff touch - web or physical button

curzon01, 2018-05-18:
Try `WifiConfig 5` and repeat the test

ascillato, 2018-05-18:
@jeylites yes it is working for me

try `SetOption26 1`

you have to see if your mqtt broker expects `POWER1 ON` or `POWER ON`, that's why the `SetOption26` command.

jeylites, 2018-05-19:
Somehow, things got fixed on the latest release download yesterday. 

I didn’t had to change Setoption26, it worked right after flashing. 

Also homebridge  Json config has always been set to 

Power1
Power2
Power3

jeylites, 2018-05-19:
@curzon01 

WiFi option 5, you referring to WIFI_RETRY 

I had that all the time

closed 2018-05-19 by jeylites

curzon01, 2018-05-19:
@jeylites the difference between your settings WIFI_RETRY (=cmnd `WifiConfig 4`) and WIFI_WAIT(=`WifiConfig 5`) is that your settings try to connect to the other AP instead of same AP it has before lost connection. If you have always a second AP with available when first AP is gone thats ok but if you have only one AP available WIFI_WAIT (or cmnd `WifiConfig 5`) might be the better setting.

I have set part of my Tasmota client to WIFI_WAIT instead of WIFI_RETRY because if the AP for these clients are unavailable these clients should not try to connect to the second AP. They should connect to the same AP again.

jeylites, 2018-05-19:
@curzon01 

Thanks for the explanation, I always wonder the difference between all configuration. It will be nice to have an explanation on github explaining the difference

curzon01, 2018-05-22:
I think the most of them are already described on [Commands Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands).

For example the Wifi modes see [Commands Wifi](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#wifi), scroll down to `WifiConfig` 1-5.

Unfortuntely many users starting directly changing values in user_config.h instead of playing with the very flexible possibility of the build in commands and finally use hardcoded user_config.h settings.