arendst/Tasmota #2755: Erratic behavior while MQTT server offline

kajmaj, 2018-05-17:
I noticed that if is MQTT server switched off (offline)  (server where Homeassistant+MQTT server run is off) some Sonoffs behaves erratic (randomly ON/OFF) 
I am speaking mainly about 1 of 2 Inching Self-locking WiFi Wireless Switch 5V 12V both of them have the same setting.
It is a problem with a garage door in my case.
Anybody faced the same issue?

ascillato, 2018-05-17:
Hi,

We need more information to solve your issue. Please, provide the troubleshooting template for one of the devices that is working erratically:

> Make sure these boxes are checked [x] before submitting your issue - Thank you!
> 
> - [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
> - [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
> - [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
> - [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
> - [ ] Hardware used : 
> - [ ] Provide the output of command ``status 0`` :
> 
> **(Please, remember to close the issue when the problem has been addressed)**
> 

And please also provide the output of the console when this behaviour is observed. Thanks

kajmaj, 2018-05-17:
I learned, that tis behavior has no relation to MQTT server offline but to device reset and consequent switch toggle.
Strange is that both devices related to the issue have the same configuration, but problem is with one piece only.

Status 0 report:
`
18:28:57 MQT: Connect failed to 192.168.1.222:1883, rc -2. Retry in 10 sec
18:29:07 CMD: Status 0
18:29:07 RSL: stat/sonoff_3/STATUS = {"Status":{"Module":1,"FriendlyName":["sonoff_3"],"Topic":"sonoff_3","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
18:29:07 RSL: stat/sonoff_3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:06:04","StartupUTC":"2018-05-17T17:23:03","Sleep":50,"BootCount":58,"SaveCount":90,"SaveAddress":"F6000"}}
18:29:07 RSL: stat/sonoff_3/STATUS2 = {"StatusFWR":{"Version":"5.13.1b","BuildDateTime":"2018-05-15T10:25:54","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
18:29:07 RSL: stat/sonoff_3/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HugoNET",""],"TelePeriod":300,"SetOption":["00000009","55818000"]}}
18:29:07 RSL: stat/sonoff_3/STATUS4 = {"StatusMEM":{"ProgramSize":484,"Free":516,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
18:29:07 RSL: stat/sonoff_3/STATUS5 = {"StatusNET":{"Hostname":"sonoff_3-7394","IPAddress":"192.168.1.60","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:9F:7C:E2","Webserver":2,"WifiConfig":1}}
18:29:07 RSL: stat/sonoff_3/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.222","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9F7CE2","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:29:07 RSL: stat/sonoff_3/STATUS7 = {"StatusTIM":{"UTC":"Thu May 17 17:29:07 2018","Local":"Thu May 17 18:29:07 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
18:29:07 RSL: stat/sonoff_3/STATUS10 = {"StatusSNS":{"Time":"2018-05-17T18:29:07"}}
18:29:07 RSL: stat/sonoff_3/STATUS11 = {"StatusSTS":{"Time":"2018-05-17T18:29:07","Uptime":"0T00:06:04","Vcc":3.393,"POWER":"OFF","Wifi":{"AP":1,"SSId":"HugoNET","RSSI":52,"APMac":"F4:F2:6D:9B:94:A2"}}}
18:29:08 MQT: Attempting connection...`

Frogmore42, 2018-05-19:
You are using core 2.4.0 on the one that is having issues.  That is known to cause resets on some devices.

kajmaj, 2018-05-19:
Well, I´ve returned back to 2.3.0 and it behaves in the same way.

`05:40:20 CMD: status 0
05:40:21 MQT: stat/sonoff_3/STATUS = {"Status":{"Module":1,"FriendlyName":["sonoff_3"],"Topic":"sonoff_3","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
05:40:21 MQT: stat/sonoff_3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:44","StartupUTC":"2018-05-19T04:36:37","Sleep":0,"BootCount":20,"SaveCount":75,"SaveAddress":"FB000"}}
05:40:21 MQT: stat/sonoff_3/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-05-15T15:29:54","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
05:40:21 MQT: stat/sonoff_3/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HugoNET",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
05:40:21 MQT: stat/sonoff_3/STATUS4 = {"StatusMEM":{"ProgramSize":526,"Free":476,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
05:40:21 MQT: stat/sonoff_3/STATUS5 = {"StatusNET":{"Hostname":"sonoff_3","IPAddress":"192.168.1.60","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:9F:7C:E2","Webserver":2,"WifiConfig":1}}
05:40:21 MQT: stat/sonoff_3/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.107","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9F7CE2","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
05:40:21 MQT: stat/sonoff_3/STATUS7 = {"StatusTIM":{"UTC":"Sat May 19 04:40:21 2018","Local":"Sat May 19 05:40:21 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:03","Sunset":"20:30"}}
05:40:21 MQT: stat/sonoff_3/STATUS10 = {"StatusSNS":{"Time":"2018-05-19T05:40:21"}}
05:40:21 MQT: stat/sonoff_3/STATUS11 = {"StatusSTS":{"Time":"2018-05-19T05:40:21","Uptime":"0T00:03:44","Vcc":3.112,"POWER":"OFF","Wifi":{"AP":1,"SSId":"HugoNET","RSSI":52,"APMac":"08:96:D7:2A:01:D8"}}}`

Strange is that second device is on 2.4.0 and works flawlessly.

Frogmore42, 2018-05-19:
You have a retained message that is causing the problem.  HA is known to have this issue.  I don't use HA, but search the issues for retained messages and you will see many from people who use HA.

kajmaj, 2018-05-19:
I´ve switched off server where HA and MQTT run. It means no retained messages available at all.
If I access device via web (IP) and Restart, it restarts and toggle.
It is the same with HA/MQTT server off or on.
It is the same with both 2.4.0 and 2.3.0 cores
Tried several different FW 5.12.0, 5.13.0, 5.14.0 - the same
`MQT: stat/sonoff_3/RESULT = {"PowerRetain":"OFF"}` Tried "ON" - the same
`MQT: stat/sonoff_3/RESULT = {"PowerOnState":3}` Tried "0" - the same
`MQT: stat/sonoff_3/RESULT = {"PulseTime1":10}` 

If HA is on, switch setting is following:
`  - platform: mqtt
    name: "Garage"
    command_topic: "cmnd/sonoff_3/power"
    state_topic: "stat/sonoff_3/POWER"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF" `

ascillato2, 2018-06-27:
Hi,

Have you managed to solve your issue?

kajmaj, 2018-06-27:
Hello,
unfortunatelly not.  I´ve tried different FW, all possible combinations of `PowerRetain` and `PowerOnState`
No progress at all.
If I restart via web (IP) either plain Restart or restart after FW change or anything else what requires Restart, it restarts and toggles.

Frogmore42, 2018-06-27:
Provide the log file showing the issue with the current fw.

kajmaj, 2018-07-01:
I'm going to flash ver. 6.0 a.s.a.p and will provide report afterwards.

kajmaj, 2018-07-07:
6.1.0 flashed, the same behavior :(
It is strange, that pulse causing garage door opening is not registered.
Log after reboot:
`00:00:00 Project sonoff sonoff_3 (Topic garage, Fallback DVES_9F7CE2, GroupTopic sonoffs) Version 6.1.0-2_3_0
00:00:00 WIF: Connecting to AP1 HugoNET in mode 11N as garage...
00:00:03 WIF: Connected
00:00:03 DNS: Initialized
00:00:03 HTP: Web server active on garage.local with IP address 192.168.1.60
00:00:04 MQT: Attempting connection...
00:00:04 MQT: Connected
00:00:04 MQT: tele/garage/LWT = Online (retained)
00:00:04 MQT: cmnd/garage/POWER = 
00:00:04 MQT: tele/garage/INFO1 = {"Module":"Sonoff Basic","Version":"6.1.0","FallbackTopic":"DVES_9F7CE2","GroupTopic":"sonoffs"}
00:00:04 MQT: tele/garage/INFO2 = {"WebServerMode":"Admin","Hostname":"garage","IPAddress":"192.168.1.60"}
00:00:04 MQT: tele/garage/INFO3 = {"RestartReason":"Software/System restart"}
00:00:05 MQT: stat/garage/RESULT = {"POWER":"OFF"}
00:00:05 MQT: stat/garage/POWER = OFF
08:34:40 MQT: tele/garage/STATE = {"Time":"2018-07-07T08:34:40","Uptime":"0T00:00:12","Vcc":3.112,"POWER":"OFF","Wifi":{"AP":1,"SSId":"HugoNET","RSSI":60,"APMac":"08:96:D7:2A:01:D8"}}
`
STATUS 0:
`08:40:37 CMD: status 0
08:40:37 MQT: stat/garage/STATUS = {"Status":{"Module":1,"FriendlyName":["sonoff_3"],"Topic":"garage","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
08:40:37 MQT: stat/garage/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:06:09","StartupUTC":"2018-07-07T06:34:28","Sleep":50,"BootCount":82,"SaveCount":174,"SaveAddress":"FA000"}}
08:40:37 MQT: stat/garage/STATUS2 = {"StatusFWR":{"Version":"6.1.0","BuildDateTime":"2018-07-07T08:28:44","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
08:40:37 MQT: stat/garage/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HugoNET","indebuurt2"],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
08:40:37 MQT: stat/garage/STATUS4 = {"StatusMEM":{"ProgramSize":476,"Free":524,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","00402B94","0C000000","21021796","00000000"]}}
08:40:37 MQT: stat/garage/STATUS5 = {"StatusNET":{"Hostname":"garage","IPAddress":"192.168.1.60","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"60:01:94:9F:7C:E2","Webserver":2,"WifiConfig":1}}
08:40:37 MQT: stat/garage/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.107","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9F7CE2","MqttUser":"admin","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
08:40:37 MQT: stat/garage/STATUS7 = {"StatusTIM":{"UTC":"Sat Jul 07 06:40:37 2018","Local":"Sat Jul 07 08:40:37 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 02:00:00 2018","Timezone":2}}
08:40:37 MQT: stat/garage/STATUS10 = {"StatusSNS":{"Time":"2018-07-07T08:40:37"}}
08:40:37 MQT: stat/garage/STATUS11 = {"StatusSTS":{"Time":"2018-07-07T08:40:37","Uptime":"0T00:06:09","Vcc":3.112,"POWER":"OFF","Wifi":{"AP":1,"SSId":"HugoNET","RSSI":56,"APMac":"08:96:D7:2A:01:D8"}}}`

Because garage system requires "pulse" to act, I set up PulseTime:
`08:39:27 MQT: stat/garage/RESULT = {"PulseTime1":"10 (Active 0)"}`

ascillato2, 2018-08-18:
Closing this issue as there is no feedback. Please, ask to reopen if needed. Thanks

closed 2018-08-18 by ascillato2