arendst/Tasmota #2758: Sonoff RF Bridge - can we make rules on RfReceived

jcheger, 2018-05-17:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ x] Hardware used : Sonoff RF Bridge
- [ x] Provide the output of command ``status 0`` : {"StatusFWR":{"Version":"5.14.0a","BuildDateTime":"2018-05-17T12:03:39","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}

**(Please, remember to close the issue when the problem has been addressed)**
Is it possible to create rules on received codes of rfkeys ? If yes, can somebody tell the syntax ?

Without any relevant documentation, I did try all kind of syntax I could imagine, but without any success
- rule on button1#state do publish hello world endon
- rule on rfkey1 do publish hello world endon
- rule on tele-rfkey1 do publish hello world endon
- rule on rfreceived#rfkey=1 do publish hello world endon
- rule on tele-rfreceived#rfkey=1 do publish hello world endon
- rule on rfreceived#Data=AA4DE2 do publish hello world endon

arendst, 2018-05-17:
Not available (yet).

jcheger, 2018-05-17:
The idea behind this question is to use 1 door contact and 2 PIR to power on the light. Any detection resets the timer, and once it's over, it powers the lights off (if I did forget to do it manually). The goal is to do it without any 3rd party equipment, such as an MQTT broker.

closed 2018-05-28 by arendst

Gasxet, 2019-03-11:
Dear all firs of all my compliments... I did Tasmotize almost all my devices in home, you did a great job.
This is my first time I'm looking for help and thank you in advance for any. Sorry if I maybe request in a wrong way...
I put rules in the RF bridge for some PIR sensor connected, tried to put multiple in a single Rule (Only 3 are accepted), but Only the first par of the rule seems to work, can you help?, thank you , this is the code and only yhe Rf_Bridge/PSal works none of the Rf_Bridge/PIng and/or Rf_Bridge/PCuc:
`Rule1 on RFReceived#data=E846AE do backlog publish2 Rf_Bridge/PSal 1; ruletimer1 5 endon on rules#timer=1 do publish2 Rf_Bridge/PSal 0 endon on RFReceived#data=E9E1DE do backlog publish2 Rf_Bridge/PIng 1; ruletimer2 5 endon on rules#timer=2 do publish2 Rf_Bridge/PIng 0 endon on RFReceived#data=E8329E do backlog publish2 Rf_Bridge/PCuc 1; ruletimer3 5 endon on rules#timer=3 do publish2 Rf_Bridge/PCuc 0 endon`

Bedzaa, 2022-12-04:
Dears, after some time I decided to have rules on Tasmotized RF Bridge. My idea: if PIR sensor sends the data string, the light will switch on. 

This rule however does not do anything:

Rule1
on RfReceived#Data=D332FE do publish cmnd/ Sonoff_basic_R3_03/POWER ON endon

Could anybody help? what am I doing wrong?
Thank You

sfromis, 2022-12-04:
Better fix the typo, the topic cannot contain a space.
`publish cmnd/ Sonoff_basic_R3_03/POWER ON`

Did you activate the rule? Read the console with `Weblog 3` active to see what happens.