arendst/Tasmota #2760: All sonoffs stopped working after update to 5.14

titomontes, 2018-05-17:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x ] Hardware used : Sonoff Basic/Homeassistant (Hass.io)
- [x ] Provide the output of command ``status 0`` :

```
18:29:41 MQT: patio/sonoff3/STATUS = {"Status":{"Module":1,"FriendlyName":["Reflector Patio"],"Topic":"sonoff3","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
18:29:42 MQT: patio/sonoff3/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff-minimal.bin","RestartReason":"Software/System restart","Uptime":"0T01:29:19","StartupUTC":"2018-05-17T16:00:22","Sleep":0,"BootCount":68,"SaveCount":385,"SaveAddress":"F8000"}}
18:29:42 MQT: patio/sonoff3/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-05-15T15:29:54","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
18:29:42 MQT: patio/sonoff3/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["Montes01","indebuurt2"],"TelePeriod":300,"SetOption":["55800009","55818000"]}}
18:29:42 MQT: patio/sonoff3/STATUS4 = {"StatusMEM":{"ProgramSize":526,"Free":476,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
18:29:42 MQT: patio/sonoff3/STATUS5 = {"StatusNET":{"Hostname":"sonoff3-4821","IPAddress":"192.168.1.13","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"80.58.61.250","Mac":"2C:3A:E8:07:B2:D5","Webserver":2,"WifiConfig":3}}
18:29:42 MQT: patio/sonoff3/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_07B2D5","MqttUser":"tito1","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:29:42 MQT: patio/sonoff3/STATUS7 = {"StatusTIM":{"UTC":"Thu May 17 17:29:42 2018","Local":"Thu May 17 18:29:42 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:05","Sunset":"20:27"}}
18:29:42 MQT: patio/sonoff3/STATUS10 = {"StatusSNS":{"Time":"2018-05-17T18:29:42","Switch1":"ON"}}
18:29:42 MQT: patio/sonoff3/STATUS11 = {"StatusSTS":{"Time":"2018-05-17T18:29:42","Uptime":"0T01:29:20","Vcc":3.134,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"Montes01","RSSI":64,"APMac":"34:97:F6:61:67:C0"}}}
```

Issue: After updating firmware (going minimal to normal OTA upgrade), all sonoffs have stopped working over MQTT. Whwnwver I use the slider on HA, the switch turns on and can´t be switched off from HA (seems it doesn´t retain the state). Web interface works OK.  i haven´t changed anything in HA nor the MQTT server or in the firmware itself. 

This was the previous HA yaml entry: 

```
- platform: mqtt
  name: "Reflector Patio"
  command_topic: "patio/sonoff3/cmnd/power"
  state_topic: "patio/sonoff3/POWER"
  qos: 0
  optimistic: false
  payload_on: "ON"
  payload_off: "OFF"
  retain: true
```

Thanks a lot for your help

Piet1702, 2018-05-17:
Same for me. Compiled it fresh yesterday (5.14a) for one Basic for testing. Same effect in HA like you got. Interesting note: Automation still works. Switch turned off caused by automation even the HA Web interface showed the switch was off the whole time.

VDRainer, 2018-05-17:
Change the cmnd and stat topics in HA to POWER1 and it works again.
The question is, bug or feature?

ascillato, 2018-05-18:
Hi,

@VDRainer great you find the problem. It is a feature. see command SetOption26 in wiki.

`SetOption26 | 0 / off | (default) Keep using POWER without postfix for single power devices`

curzon01, 2018-05-18:
It's also documented in [Release Changelog](https://github.com/arendst/Sonoff-Tasmota/tree/v5.14.0):
> Add SetOption26 to enforce use of indexes even when only one relay is present (#1055)

titomontes, 2018-05-18:
Ok, fixed it. Flashing wiki should be updated though, there area many not so savvy minds that will hit a wall with that feature.

Thanks a lot for your work Theo.

ascillato, 2018-05-18:
Now Tasmota uses
    `POWER1`
  instead of
    `POWER`

So as to be standarized when you have more than one output like
    `POWER1`
    `POWER2`
    `POWER3`
    `POWER4`

Before this was:
    `POWER`
    `POWER2`
    `POWER3`
    `POWER4`

`SetOption21 1` enables that previous way.

localhost61, 2018-05-18:
@ascillato you made a tipo 

> *SetOption21* 1 enables that previous way.

-> SetOption26

ascillato, 2018-05-18:
Ups. Thanks :+1:

closed 2018-06-02 by ascillato

Kedryn, 2018-07-27:
What if, on a nodemcu without relays, i keep getting "POWER" instead of "POWER1" when using SetOptions26 1?
I'm on 6.1.1.

11:33:46 MQT: homeassistant/garage_sensors/RESULT = {"SetOption26":"ON"}
11:34:33 MQT: homeassistant/garage_sensors/RESULT = {"POWER":"ON"}
11:34:33 MQT: homeassistant/garage_sensors/POWER = ON
11:34:34 MQT: homeassistant/garage_sensors/RESULT = {"POWER":"OFF"}
11:34:34 MQT: homeassistant/garage_sensors/POWER = OFF

Kedryn, 2018-07-27:
OMG. i found out. I HAD to put a Relay1 on a unconnected pin to fake it, and now it works. 
Why do i have to do that? I wanna use switches (i have an IR and a RADAR sensor on two pins)
 without having relays, only to send MQTT messages...