arendst/Tasmota #2762: Execute commands on local WiFi

anoop-r-kulkarni, 2018-05-18:
Hi, 

Please forgive my ignorance and please direct me to where I can find this information, if already available.

I have a use-case:
1) The devices are connected on a Local WiFi (which is connected to the Internet through a router)
2) In the best case, MQTT works and there are no issues to execute commands
3) IF the router fails, I intend to provide users an app, connected to the same WiFi network from where users should be able to carry out basic functionality.
4) In the above scenario, there is no MQTT. 

So the question, is there a way for me to issue commands purely on the WiFi? Can think of using HTTP requests.. but is that the only way? or I have a hole in my understanding?

Thanks and regards
~anoop

ascillato, 2018-05-18:
Yes. There are some ways.

Rest API
KNX

Please see commands on the wiki

Frogmore42, 2018-05-19:
Connect to the serial interface and see what Tasmota thinks is happening.

Make sure you get the results of status 0.

ascillato, 2018-06-02:
Hi @anoop-r-kulkarni 

Now it is also the command `websend`

If your issue was solved, please close it. Thanks :+1:

anoop-r-kulkarni, 2018-06-02:
I used the REST api. DId not know about websend, so will upgrade to that. Thanks for the leads.

Best,
~anoop

closed 2018-06-02 by anoop-r-kulkarni