arendst/Tasmota #2763: SR04 issue on Tasmota 5.14.0

alexjoni, 2018-05-18:
I recently added a SR04 distance sensor to a Sonoff basic, and it seems to work ok the first couple of minutes.
After about 5 minutes the readout is 283cm, and it gets stuck to that value.
If I restart the sonoff it works again for 5 minutes then it turns again to 283cm..

Is there a way to debug this further?
Maybe trigger an Sr04Init() function more often?

Regards,
Alex

Frogmore42, 2018-05-19:
It could also be the SR04 sensor that is bad.  Do you have another one?

Frogmore42, 2018-05-19:
Since it seems like you are getting the maximum distance, it seems like the sensor is not working well.  Most of them need 5V to work, so that might be the issue.

wongnam, 2018-05-19:
i had tried before, over 300cm the value will unreadable with Tasmota even your SR04t sensor can read up to 400cm.

alexjoni, 2018-05-19:
I did use 5V to power it. I'm not sure how clean the 5V on a sonoff basic is, I'll try to add a cap for filtering.
I'll also order a new sensor and see if it makes any difference.

helmar74, 2018-06-18:
Hi, I've also an issue with 5.14.0 and a HC-SR04.
First of all the sensor is OK. I tried it with an other arduino and it's measuring fine!
I connected the sensor as mentioned here: 
[https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini-and-HC-SR04-Ultrasonic-Sensor](url)

The problem ist, that the sensor is measuring not correct woth Tasmota FW.
The distance is about 145 cm. 
But within Tasmota the measuring jumps between 96 and 144 cm
Via MQTT sometimes (the correct) 144 cm are sent, somteimes 96 cm.
The SEnsor has free "sight" to the ceiling, so no reflection should occur
So the measuring isn't ok.

Anyone an idea, what to try?

ascillato2, 2018-07-16:
Hi

A lot of improvements has been made for sensors in last release. Please try it. 

Closing this issue. Please, reopen if needed. Thanks

closed 2018-07-16 by ascillato2