arendst/Tasmota #2764: Compilation problem 'XsnsCall' was not declared in this scope

majherek, 2018-05-18:
Hi,

When I compile vanilla 5.14.0a everything is OK.

When I usw
`#define USE_CONFIG_OVERRIDE                      // Uncomment to use user_config_override.h file. See README.md
`
with my user_config_override.h I got error:
```
C:\Users\majherek\Documents\Arduino\sonoff\sonoff.ino: In function 'void MqttDataHandler(char*, byte*, unsigned int)':

sonoff:562: error: 'XsnsCall' was not declared in this scope

       XsnsCall(FUNC_COMMAND);

                            ^

C:\Users\majherek\Documents\Arduino\sonoff\sonoff.ino: In function 'boolean MqttShowSensor()':

sonoff:1446: error: 'XsnsCall' was not declared in this scope

   XsnsCall(FUNC_JSON_APPEND);

                            ^

C:\Users\majherek\Documents\Arduino\sonoff\sonoff.ino: In function 'void PerformEverySecond()':

sonoff:1501: error: 'XsnsCall' was not declared in this scope

       XsnsCall(FUNC_PREP_BEFORE_TELEPERIOD);

                                           ^

sonoff:1521: error: 'XsnsCall' was not declared in this scope

   XsnsCall(FUNC_EVERY_SECOND);

                             ^

C:\Users\majherek\Documents\Arduino\sonoff\sonoff.ino: In function 'void StateLoop()':

sonoff:1826: error: 'XsnsCall' was not declared in this scope

   XsnsCall(FUNC_EVERY_50_MSECOND);

                                 ^

C:\Users\majherek\Documents\Arduino\sonoff\sonoff.ino: In function 'void setup()':

sonoff:2472: error: 'XsnsCall' was not declared in this scope

   XsnsCall(FUNC_INIT);

                     ^

C:\Users\majherek\Documents\Arduino\sonoff\settings.ino: In function 'void SettingsSaveAll()':

settings:188: error: 'XsnsCall' was not declared in this scope

   XsnsCall(FUNC_SAVE_BEFORE_RESTART);

                                    ^

C:\Users\majherek\Documents\Arduino\sonoff\xdrv_10_rules.ino: In function 'void RulesEvery50ms()':

xdrv_10_rules:351: error: 'XsnsNextCall' was not declared in this scope

         XsnsNextCall(FUNC_JSON_APPEND);  // ,"INA219":{"Voltage":4.494,"Current":0.020,"Power":0.089}

                                      ^
[... 282 lines cut ...]
#endif

#ifdef WEB_SERVER
#undef WEB_SERVER
#define WEB_SERVER 2
#endif

#ifdef USE_DISCOVERY
#undef USE_DISCOVERY
#endif

#ifdef LATITUDE
#undef LATITUDE
#define LATITUDE               22.33032         // [Latitude] Your location to be used with sunrise and sunset
#endif

#ifdef LONGITUDE
#undef LONGITUDE
#define LONGITUDE              22.90932          // [Longitude] Your location to be used with sunrise and sunset
#endif

#endif  // _USER_CONFIG_OVERRIDE_H_
```

I use Arduine IDE (2.4.1)

majherek, 2018-05-21:
Hi,

I found problem.

When I uncomment USE_MQTT_TLS and comment USE_WEBSERVER vanilla 5.14.0a stops compiling with error mentioned above. I think it is a bug. 

What shall I do to use MQTT with TLS support?

majherek, 2018-05-21:
Is is similar to #1692

arendst, 2018-05-21:
Do it in user_config.h and NOT in user_config.override.h

See header information in your user_config_override.h line 41

majherek, 2018-05-21:
I have in user_config.h
```
//#define USE_CONFIG_OVERRIDE                      // Uncomment to use user_config_override.h file. See README.md
#define USE_MQTT_TLS                             // Use TLS for MQTT connection (+53k code, +15k mem)
//#define USE_WEBSERVER                            // Enable web server and wifi manager (+66k code, +8k mem)

```
And my  user_confi_.override.h is empty.

Problem still exitst.

majherek, 2018-05-23:
Hi,

can anybody check if the vanilla 5.14.0a is compiling with:
```
//#define USE_CONFIG_OVERRIDE                      // Uncomment to use user_config_override.h file. See README.md
#define USE_MQTT_TLS                             // Use TLS for MQTT connection (+53k code, +15k mem)
//#define USE_WEBSERVER                            // Enable web server and wifi manager (+66k code, +8k mem)
```
in user_config.h

I will be grateful for your help.

ascillato, 2018-05-23:
Hi,

I tested last Tasmota Version v5.14.0a (2018-MAY-23) and compiling it under platformio as wiki explain with all default except `#define USE_MQTT_TLS` and `//#define USE_WEBSERVER`, It works for me.

So may be there is a problem with your libraries or compiler

majherek, 2018-05-24:
Thanks @ascillato.

But I use Arduino IDE with esp8266 2.4.1.

I made some tests.

1) Download 5.13.1 in zip format from github. Copy platform.txt and boards.txt, albo replace all libraries from source zip. I made 2 modifications in user_config.h.
```
//#define USE_CONFIG_OVERRIDE                      // Uncomment to use user_config_override.h file. See README.md
#define USE_MQTT_TLS                             // Use TLS for MQTT connection (+53k code, +15k mem)
//#define USE_WEBSERVER                            // Enable web server and wifi manager (+66k code, +8k mem)
``` 
And everything is compiling. 

2) I do the same with 5.14.0a (copy boards.txt, platform.txt, remove old libraries, copy new one from zip file, 2 changes in user_config.h). And there is still error:
```
Arduino:1.8.5 (Windows Store 1.8.10.0) (Windows 10), Płytka:"Generic ESP8266 Module, Serial, 80 MHz, ck, 26 MHz, 40MHz, DOUT, 4M (1M SPIFFS), 2, v1.4 Higher Bandwidth, Disabled, None, Only Sketch, 115200"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.10.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.10.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\majherek\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.10.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.10.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\majherek\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.10.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\majherek\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:UploadTool=esptool,CpuFrequency=80,ResetMethod=ck,CrystalFreq=26,FlashFreq=40,FlashMode=dout,FlashSize=4M1M,led=2,LwIPVariant=Prebuilt,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=115200 -ide-version=10805 -build-path C:\Users\majherek\AppData\Local\Temp\arduino_build_461214 -warnings=none -build-cache C:\Users\majherek\AppData\Local\Temp\arduino_cache_667707 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\majherek\Documents\ArduinoData\packages\esp8266\tools\esptool\0.4.13 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\majherek\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Users\majherek\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\0.2.0 -verbose C:\Users\majherek\Documents\Arduino\sonoff\sonoff.ino
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.10.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.10.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\majherek\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.10.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.10.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\majherek\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.10.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\majherek\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:UploadTool=esptool,CpuFrequency=80,ResetMethod=ck,CrystalFreq=26,FlashFreq=40,FlashMode=dout,FlashSize=4M1M,led=2,LwIPVariant=Prebuilt,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=115200 -ide-version=10805 -build-path C:\Users\majherek\AppData\Local\Temp\arduino_build_461214 -warnings=none -build-cache C:\Users\majherek\AppData\Local\Temp\arduino_cache_667707 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\majherek\Documents\ArduinoData\packages\esp8266\tools\esptool\0.4.13 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\majherek\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Users\majherek\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\0.2.0 -verbose C:\Users\majherek\Documents\Arduino\sonoff\sonoff.ino
Using board 'generic' from platform in folder: C:\Users\majherek\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1
Using core 'esp8266' from platform in folder: C:\Users\majherek\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1
Detecting libraries used...
"C:\Users\majherek\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\majherek\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/include" "-IC:\Users\majherek\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/lwip/include" "-IC:\Users\majherek\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\majherek\AppData\Local\Temp\arduino_build_461214/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC" -DLED_BUILTIN=2  -DESP8266 "-IC:\Users\majherek\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266" "-IC:\Users\majherek\Documents\ArduinoData\packa [... cut ...] w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_GENERIC" -DLED_BUILTIN=2  -DESP8266 "-IC:\Users\majherek\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266" "-IC:\Users\majherek\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.1\variants\generic" "-IC:\Users\majherek\Documents\Arduino\libraries\esp-knx-ip-0.5.1" "-IC:\Users\majherek\Documents\ArduinoData\packages\esp8

ascillato, 2018-05-24:
Hi

I compile the last Tasmota version that is in github (v5.14.0b) with arduino, and it also works

[image]

So, I can not reproduce the issue neither with Arduino nor platformio

Try uninstall arduino and install it again.

As you can see on the attached screenshot, I use Arduino 1.8.2

Or try Platformio.

arendst, 2018-05-25:
You might by now get a hunch that the windows store solution doesn't seem to work...

Start all over following the wiki - Arduino IDE and become one of many happy users.

majherek, 2018-05-27:
Hi @arendst ,

I am unable do run Arduino IDE from ZIP file [(Arduino Forum)](https://forum.arduino.cc/index.php?topic=549751.0).

So I install Arduino IDE 1.8.2 from installator from arduino.cc. I configure it (as it is described in WIKI). And vanilla 5.14.0b is compiling, but when I do use MQTT TLS it stops compilling. I try 2.3.0, 2.4.0, 2.4.1 - it does not matter. 

I use about 20 sonoff and 3 wemos d1 mini. I start using Tasmota from 5.9. It stops working for me from 5.14. Tastmota 5.13.1 is compiling (regardless of Arduino IDE - Win store, standalone installation, etc.).

arendst, 2018-05-27:
OK. When disabling the webserver it indeed throws two compile errors regarding xsns_23_sdm120 and xsns_25_sdm630. When I correct these two error it compiles just fine with TLS enabled and Webserver disabled.

I'll update github now...

ascillato, 2018-05-27:
I understand now why was not working for you and was working ok for me. I
did not include the sdm sensor.

majherek, 2018-05-27:
Hi all,
I emphasized each time that I made only two changes to the pure source version (vanilla). 

However, the most important thing is that the problem has been fixed ;-) 

Now everything is working fine in any Arduino IDE (installed from arduino.cc and from windows store).

closed 2018-05-27 by majherek