arendst/Tasmota #2765: DHT22 implemented?

Dome2075, 2018-05-18:
Is there no support for DHT 22? 

i only see DHT 11 in the config.

when i connect the DHT22 and cofigure DHT11, i'll get any rubisch..

thanks for help

ascillato, 2018-05-18:
Yes, it is implemented.

Please, use `AM2301` option

See wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Sensor-Configuration)

Extract from wiki:

> Single wire sensor
> The following single wire sensors are supported:
> 
> DHT11 Temperature and Humidity - DHT11 (1) in group 1
> DHT21 Temperature and Humidity - AM2301 (2) in group 1
> AM2301 Temperature and Humidity - AM2301 (2) in group 1
> DHT22 Temperature and Humidity - DHT22 (3) in group 1
> AM2302 Temperature and Humidity - DHT22 (3) in group 1
> AM2321 Temperature and Humidity - DHT22 (3) in group 1
> Itead SI7021 Temperature and Humidity - SI7021 (3) in group 1
> DS18B20 Temperature - DS18x20 (4) in group 2

ascillato, 2018-05-18:
Please close the issue if your problem was solved. Thanks

closed 2018-05-18 by Dome2075