arendst/Tasmota #2768: NodeMcu V3 cant flash

hwarang098tkd, 2018-05-18:
Hi and congrats for the fantastic work.
I have flashed a lot of modules but now i a have a problem.
I tried to flash my nodemcu v3 with atom (as always) and i cant. 
I tried to flash it and i did according to the terminal (wrote Success) but when the module restart had the same softaware and configurations as before the flashing. then i flash a small program with Arduino IDE and worked fine(i theck it and had the new code). And here is the strange -> i flashed it again with atom and returned to the same configurations as it was at first.
I cant understand this. "Like a gost code".
Some help please!!!!

Frogmore42, 2018-05-19:
There are also commands to erase and reset settings in the wiki.

hwarang098tkd, 2018-05-19:
ok i did it. I download esptoo.py and erase_flash to node and then flash with atom with [env:wemos-d1-mini].... Is this the right software? i want to use this module for sensors(temperature, motion and photoresistor via analog A0). and a second question : after the last update (5.14) i don't know how to enable analog A0.

Jason2866, 2018-05-21:
@hwarang098tkd 
NodeMCU and Wemos are programmed the same way.
So yes you are on the right way. Have fun!