arendst/Tasmota #2774: OTA uprade to 5.14.0 fails with blinkning green light

apastuszak, 2018-05-19:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : Sonoff Basic
- [ ] Provide the output of command ``status 0`` :

**(Please, remember to close the issue when the problem has been addressed)**

I am unable to flash 5.14.0 on my switches using OTA.  I did a switch last week and after reboot, I the flashing green light telling me I got an incomplete flash.

This afternoon I did another switch.  I flashed Sonoff Minimal and after reboot, I got the flashing green light again.

To further experiment, I took another switch and downgraded it from 5.13.1 to 5.12.0 using the minimal followed by sonoff method and that worked fine.  As soon as I flashed 5.14.0 minimal in order to upgrade, the switch rebooted and I am once again getting a flashing green light on the switch and no connectivity.

The first switch that failed last week I was able to upgrade using a serial to USB adapter and jumper wires, so I will do the same for the other two.  But I don't want to do any more OTA attempts until I figure out what's going on.

Since I can't connect to the switch, and the flashing light followed by occasional reboots indicates a bad flash, I don't even know how to begin to troubleshoot the issue.

reloxx13, 2018-05-19:
Blinking green lights means your wifi config is gone and (default) the wps manager is on for a few mins and than the device restarts.

Looks like you flashed soetime ago with another CFG_HOLDER and the new default config holder is clearing your config.

apastuszak, 2018-05-19:
Well, I don't see any new wifi networks with the name Sonoff in them on my network.

I took one of my switches and flashed it using serial to USB adapter and it came up just fine in 2.4.0.  Between getting it all setup, giving /dev/ttyUSB0 write permissions, and a 15 year old nagging me for lunch, the whole process took about a half hour.

I went to get switch #2 and the green light was now solid.  I went to the switch URL and it was on Tasmota minimal firmware and I was able to upgrade it to Tasmota full.

But the switch was in that blinking rebooting mode for probably a good 10-15 min, and I don't know how long it was that way before I noticed it was solid.

The device is in the basement along with my router, and the two devices are about 5 feet apart from it, so I doubt it' signal strength issue.

ajanulis, 2018-05-21:
@apastuszak , I am pretty sure that IT WAS SIGNAL STRENGHT issue - I am unable to connect to it on the same table - it's just 2 feet :) - if you compiled using last Arduino core - it causes a lot of troubles with wifi disappearing and reappearing (sometimes). Compile using 2.3.0 and everything should be OK

ajanulis, 2018-05-21:
@apastuszak, but you always can downgrade in case of failure

apastuszak, 2018-05-21:
Of the two that I had upgraded, one was on 5.131 using 2.4.1.  I directly
updated that to 5.14.0 and it went into the flash boot loop and I had flash
serially.  The other one, I downgraded to the released 5.12.0 from Github
and then upgraded to 5.14.0.  That one also went into the flashing light
loop but at some point connected to wifi and let me do a full upgrade.  So,
that's my strategy going foward.  Downgrade to 5.12.0 and then upgrade to
5.14.0, and always use Arduino 2.3.0.

Jason2866, 2018-05-21:
There is really no need to downgrade. Just compile sonoff minimal and flash it if new firmware is to big
for direct flash.
Compile firmware with arduino v.1.5.3 (2.3.0). This is the one and only version that works always...
Just follow this, and you will get a reliable setup with 5.14.x

ascillato2, 2018-06-27:
Hi

Have you managed to solve your issue?

benjaminprojas, 2018-06-29:
@apastuszak are you using MQTT? If so, it is probably related to: https://github.com/arendst/Sonoff-Tasmota/issues/1955. This has nothing to do with OTA, but rather that it cannot reconnect back to your wifi network after the upgrade.  Currently the last working version for me is 5.12.0m. I have not been able to get any newer version to work.

closed 2018-07-10 by ascillato2