arendst/Tasmota #2775: Unable to send values from sensor (WEMOS D1 mini + ds18b20) to home assistant via MQTT

Bellumatt, 2018-05-19:
Hi all,
Sorry, but I am a little bit lost.
The following is a part of my configuration.yaml file for home assistant:

mqtt:
  broker: 192.168.1.15
  port: 1883
  username: myusername
  password: mypassword 

sensor:
#weather predictions
  - platform: yr
#wemos temperatura stanze
  - platform: mqtt
       state_topic: "Temperatura/home/emma/temperature"
       name: "WEMOS-2P-EMMA"
        protocol: 3.1
        unit_of_measurement: "°C"
        value_template: "{{ value_json['DS18B20']['Temperature'] }}"

I can’t make it work with home assistant. I have the icon and the graph template in my dashboard, but no temperature is shown.
What am I missing...
Thanks
D

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://forum.digital-pig.com/)
- [X] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [X] Hardware used : WEMOS d1 mini
- [X] Provide the output of command ``status 0``

19:23:33 CMD: Status 0=
19:23:33 MQT: Temperatura/home/emma/temperature/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"home/emma/temperature","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
19:23:33 MQT: Temperatura/home/emma/temperature/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:58:13","StartupUTC":"2018-05-19T17:25:20","Sleep":0,"BootCount":33,"SaveCount":47,"SaveAddress":"F5000"}}
19:23:33 MQT: Temperatura/home/emma/temperature/STATUS2 = {"StatusFWR":{"Version":"5.13.1","BuildDateTime":"2018-05-01T21:09:13","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
19:23:33 MQT: Temperatura/home/emma/temperature/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WiFi",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
19:23:33 MQT: Temperatura/home/emma/temperature/STATUS4 = {"StatusMEM":{"ProgramSize":530,"Free":472,"Heap":16,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":3}}
19:23:33 MQT: Temperatura/home/emma/temperature/STATUS5 = {"StatusNET":{"Hostname":"home/emma/temperature-5109","IPAddress":"192.168.1.57","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"MyMACadd","Webserver":2,"WifiConfig":3}}
19:23:33 MQT: Temperatura/home/emma/temperature/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.15","MqttPort":1883,"MqttClientMask":"WEMOS-2P-EMMA","MqttClient":"WEMOS-2P-EMMA","MqttUser":"myusername","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
19:23:33 MQT: Temperatura/home/emma/temperature/STATUS7 = {"StatusTIM":{"UTC":"Sat May 19 18:23:33 2018","Local":"Sat May 19 19:23:33 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:03","Sunset":"20:30"}}
19:23:33 MQT: Temperatura/home/emma/temperature/STATUS10 = {"StatusSNS":{"Time":"2018-05-19T19:23:33","DS18B20":{"Temperature":27.6},"TempUnit":"C"}}
19:23:33 MQT: Temperatura/home/emma/temperature/STATUS11 = {"StatusSTS":{"Time":"2018-05-19T19:23:33","Uptime":"0T00:58:13","Vcc":2.921,"Wifi":{"AP":1,"SSId":"myWiFi","RSSI":52,"APMac":"APMAC"}}}

ascillato, 2018-05-19:
your `state_topic `is wrong

please, read the wiki at https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant#dht22-sensor

Bellumatt, 2018-05-20:
Thank.
Solved!
:-)

closed 2018-05-20 by Bellumatt