arendst/Tasmota #2776: Pass SCHEME using Tasmota WS2812 led with HomeAssistant

Giorgiofox, 2018-05-19:
Hello I'm using Tasmota firmware with WS2812 with HomeAssistant, and it work perfectly.
I'm also using Google Home with vocal commands.. there are any way to pass SCHEME number using Google Assistant?
Thanks in advance

Giorgio

ascillato, 2018-05-19:
yes, configure home assistant to send a mqtt message to stat/[yoursonofftopic]/Scheme.

See wiki on homeassistant at https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant#led-ws2812b

Giorgiofox, 2018-05-19:
Yes, this work! But.. what vocal command I need to use with Google Home ?

thanks!
Giorgio

ascillato, 2018-05-19:
Sorry, that is google home configuration, not Tasmota. You have to configure new commands on HomeAssistant. Please see the HomeAssistant configuration at https://www.home-assistant.io and their forums at https://community.home-assistant.io/

If you find an answer with them, please also share here. Thanks

ascillato, 2018-05-19:
Here you have a workaround: https://community.home-assistant.io/t/how-to-activate-scene-via-voice-command-google-home/51962/6

It is very interesting. I will do something like that at home :wink: 

Thanks for share your ideas!

As this is not a Tasmota Software Bug, please close this issue. Thanks :+1:

closed 2018-05-19 by Giorgiofox