arendst/Tasmota #2780: Feature Requests for Rules

ascillato, 2018-05-19:
**Have you look for this feature in other issues and in the wiki?**
YES

**Is your feature request related to a problem? Please describe.**
_A clear and concise description of what the problem is._

1 - As on rules there are 5 Variables stored in RAM named as `VAR1 `to `VAR5 `and called as `%VAR1%`, what about of having also a `MEM1 `to `MEM5 `stored in EEPROM and called as `%MEM1%` ?

These would be very useful for storing SET_POINT Values for Thermostats and also for saving the last status for next boot.

2- Another nice feature would be to be able to use Variables on the trigger of the event like the following example rule:
`rule on tele-am2301_12#temperature>%var1% do power1 on endon`
by now, it is only possible to do the following:
`rule on tele-am2301_12#temperature>22 do power1 on endon`

This would add flexibility for example on changing a set point from mqtt or from other internal event.

**Describe the solution you'd like**
_A clear and concise descriptidon of what you want to happen._
* Add to rules variables MEM1 to MEM5 and be stored in EEPROM.
* Add to rules the possibility of using %var1% and %mem1% on the first part of the rule.

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._
none

**Additional context**
_Add any other context or screenshots about the feature request here._

---------------------------------------------

**Thermostat Example. Unfinished.**

Due to current limitations:

* Allways starts enabled
* The Set Point (upper limit and lower limit) are fixed

Inital config:

* available physical button as button2
* relay1 as controller
* button1 on any unused pin
* relay2 on any unused pin

Inital config on console:

rule 1                 <- turn on rules
teleperiod 60     <- check temp every minute
setoption26 1    <- use power1 on mqtt messages
setoption0 0      <- dont save relay status on eeprom
poweronstate 0 <- start all relays off

Rules:

On boot enable thermostat and start watch dog timer to check temp sensor connection. Inform Status
`rule on system#boot do backlog event enabled=1; ruletimer1 70; event getenabled; event message=Init endon`

Send Log Messages
`on event#message do publish stat/sonoff/log %value% endon`

Set thermostat enable variable
`on event#enabled do var1 endon`

View thermostat enable variable
`on event#getenabled do publish stat/sonoff/ENABLED %var1% endon`

An available button is configured as button2 to set thermostat ON or STAND BY
`on button2#state do backlog event togglevar1 endon`

`on event#togglevar1 do backlog event toggling1=%var1% endon`
`on event#toggling1=0 do backlog event enabled=1 endon`
`on event#toggling1=1 do backlog event enabled=0 endon`

Check temp sensor connection. If fails, set to off and turn off thermostat. Inform new status. Also continue checking
`on rules#timer=1 do backlog event enabled=0; ruletimer1 70; power1 0; event getenabled; event message=SENSOR_FAILURE endon`

Resets checking timer if temperature is connected
`on tele-am2301-12#temperature do ruletimer1 70 endon`

Thermostat control - upper limit and lower limit
`on tele-am2301-12#temperature>25 do power1 0 endon`
`on tele-am2301-12#temperature<24 do power1 %var1% endon`

Thermostat can be turned on by:

* pushing button2
* by command on local console: `event enabled=1`
* by command on any console: `publish cmnd/topic/EVENT ENABLED=1`
* or MQTT at: `cmnd/topic/EVENT ENABLED=1`

Thermostat can be turned Off by:

* pushing button2
* by command on local console: `event enabled=0`
* by command on any console: `publish cmnd/topic/EVENT ENABLED=0`
* or MQTT at: `cmnd/topic/EVENT ENABLED=0`

To get the status:

* by command on local console: `event getenabled`
* by command on any console: `publish cmnd/topic/EVENT GETENABLED`
* by MQTT: `cmnd/topic/EVENT GETENABLED`

   answer at: `stat/[topic]/ENABLED`

Everything together:

INITIAL CONFIG:

`backlog rule 1; teleperiod 60; setoption26 1; setoption0 0; poweronstate 0`

RULES:

`rule on system#boot do backlog event enabled=1; ruletimer1 70; event getenabled; event message=Init endon on event#message do publish stat/sonoff/log %value% endon on event#enabled do var1 endon on event#getenabled do publish stat/sonoff/ENABLED %var1% endon on button2#state do backlog event togglevar1 endon on event#togglevar1 do backlog event toggling1=%var1% endon on event#toggling1=0 do backlog event enabled=1 endon on event#toggling1=1 do backlog event enabled=0 endon on rules#timer=1 do backlog event enabled=0; ruletimer1 70; power1 0; event getenabled; event message=SENSOR_FAILURE endon on tele-am2301-12#temperature do ruletimer1 70 endon on tele-am2301-12#temperature>25 do power1 0 endon on tele-am2301-12#temperature<24 do power1 %var1% endon`

---------------------------------------------

**(Please, remember to close the issue when the problem has been addressed)**

wongnam, 2018-05-20:
It would be nice if also implement 2 more rules
- on button1#state=doubleclick do publish cmnd/sonoff1/power2 endon
- on button1#state=hold do power2 on endon

Thanks

arendst, 2018-05-20:
I'll release the requested features.

Alas, your rule set will fail as the total length exceeds 512 characters ...

ascillato, 2018-05-20:
Tested new commands VAR and MEM and works really good

Thanks Theo !   :+1:

closed 2018-05-20 by ascillato

tjareson, 2019-01-29:
Is there anything that can be done to support this feature request?
My use case is using it as local controller for an RFID controlled lock. RFID reader is connected to serial port of sonoff. If the server is not responding to the MQTT msg of the serial bridge certain RFID-IDs can be stored locally on sonoff with rules to open the door as fallback option.