arendst/Tasmota #2781: Sonoff Pow V2 CSE7759 Reading fluctuates

HobbytronicsPK, 2018-05-20:
Hi.
 Thank you so much for such a splendid Sonoff third party firmware. I am having issue with my Sonoff Pow. It is **Version 2 with CSE7759** in it.

**Main problem:**
**Value to current and Power fluctuate significantly** with a constant charging of my Laptop. With official itead firmware, it was giving Power of 27 Watts while in Tasmota firmware, Power is fluctuating from 39-47 watts.

I suspect, Tasmota firmware does not support Sonoff Pow V2 devices with CSE7759 IC in it. I have changed module under configuration to Sonoff Pow R2 and Sonoff S31 which also have same CSE chip. But changing the module results in no output under Voltage, Current, Power etc.

arendst, 2018-05-20:
I guess you pow V2 is just a plain old pow V1.

If it was a V2 then you would never see any reading when you select module Pow as the energy measurement  hardware is totally different.

To verify if it is a V2 select module Pow V2 and enable weblog option 4 for some seconds. The output of the serial interface  should show the CSE7759 data.

HobbytronicsPK, 2018-05-20:
@arendst I am talking about Sonoff Ver 2.0 not R2.

arendst, 2018-05-20:
Yes so your statement

> Thank you so much for such a splendid Sonoff third party firmware. I am having issue with my Sonoff Pow. It is Version 2 with CSE7759 in it. <

is not true.

arendst, 2018-05-20:
Did you calibrate your Pow after installing Tasmota?

HobbytronicsPK, 2018-05-20:
After calibration. Now reading is smooth. Thank you @arendst for prompt reply.

closed 2018-05-20 by HobbytronicsPK