arendst/Tasmota #2785: Sonoff Pow, No energy data on MQTT

HobbytronicsPK, 2018-05-20:
Hi,

I have successfully implemented Tasmota Firmware on my Sonoff Pow Ver. 2 (Not R2). I have setup mosquitto MQTT broker and Nodered. Turning "off" and "on" commands are working on topic "cmnd/sonoff/power" and device is responding to specific command by turning relay off and on. Energy data is shown on web interface.

**Problem:** 
Energy data is not being published to MQTT. I have subscribed to # on nodered mqtt. All other topics such as "stat/sonoff/POWER" are receiving response. But there is no energy data is published on stat/sonoff/SENSOR topic.

curzon01, 2018-05-21:
- pls give `Status 0` info needed as requested
- have a look on Web GUI Console and observer whats happening within 5 minutes.

HobbytronicsPK, 2018-05-22:
Here is the Console output.
`00:00:00 Project sonoff Sonoff (Topic DVES_EA9716, Fallback DVES_EA9716, GroupTopic sonoffs) Version 5.14.0-2_4_1
00:00:00 WIF: Connecting to AP1 Ahmed in mode 11N as DVES_EA9716-5910...
00:00:08 RSL: stat/DVES_EA9716/RESULT = {"POWER":"OFF"}
00:00:08 RSL: stat/DVES_EA9716/POWER = OFF
01:51:50 WIF: Connected
01:51:50 DNS: Initialized
01:51:50 HTP: Web server active on DVES_EA9716-5910.local with IP address 192.168.50.221
01:51:51 MQT: Attempting connection...
01:51:51 MQT: Connected
01:51:51 MQT: tele/DVES_EA9716/LWT = Online (retained)
01:51:51 MQT: cmnd/DVES_EA9716/POWER = 
01:51:51 MQT: tele/DVES_EA9716/INFO1 = {"Module":"Sonoff Pow","Version":"5.14.0","FallbackTopic":"DVES_EA9716","GroupTopic":"sonoffs"}
01:51:51 MQT: tele/DVES_EA9716/INFO2 = {"WebServerMode":"Admin","Hostname":"DVES_EA9716-5910","IPAddress":"192.168.50.221"}
01:51:51 MQT: tele/DVES_EA9716/INFO3 = {"RestartReason":"Power on"}
01:51:52 MQT: stat/DVES_EA9716/RESULT = {"POWER":"OFF"}
01:51:52 MQT: stat/DVES_EA9716/POWER = OFF
01:51:59 RSL: tele/DVES_EA9716/STATE = {"Time":"2018-05-22T01:51:59","Uptime":"0T00:00:18","Vcc":3.481,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Ahmed","RSSI":88,"APMac":"00:15:6D:81:A6:82"}}
01:51:59 RSL: tele/DVES_EA9716/SENSOR = {"Time":"2018-05-22T01:51:59","ENERGY":{"Total":0.052,"Yesterday":0.000,"Today":0.052,"Period":0,"Power":0,"Factor":0.00,"Voltage":198,"Current":0.000}}
01:52:09 RSL: tele/DVES_EA9716/STATE = {"Time":"2018-05-22T01:52:09","Uptime":"0T00:00:28","Vcc":3.481,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Ahmed","RSSI":92,"APMac":"00:15:6D:81:A6:82"}}
01:52:09 RSL: tele/DVES_EA9716/SENSOR = {"Time":"2018-05-22T01:52:09","ENERGY":{"Total":0.052,"Yesterday":0.000,"Today":0.052,"Period":0,"Power":0,"Factor":0.00,"Voltage":198,"Current":0.000}}
01:52:19 RSL: tele/DVES_EA9716/STATE = {"Time":"2018-05-22T01:52:19","Uptime":"0T00:00:38","Vcc":3.481,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Ahmed","RSSI":100,"APMac":"00:15:6D:81:A6:82"}}
01:52:19 RSL: tele/DVES_EA9716/SENSOR = {"Time":"2018-05-22T01:52:19","ENERGY":{"Total":0.052,"Yesterday":0.000,"Today":0.052,"Period":0,"Power":0,"Factor":0.00,"Voltage":199,"Current":0.000}}
01:52:29 RSL: tele/DVES_EA9716/STATE = {"Time":"2018-05-22T01:52:29","Uptime":"0T00:00:48","Vcc":3.480,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Ahmed","RSSI":92,"APMac":"00:15:6D:81:A6:82"}}
01:52:29 RSL: tele/DVES_EA9716/SENSOR = {"Time":"2018-05-22T01:52:29","ENERGY":{"Total":0.052,"Yesterday":0.000,"Today":0.052,"Period":0,"Power":0,"Factor":0.00,"Voltage":199,"Current":0.000}}
01:52:39 RSL: tele/DVES_EA9716/STATE = {"Time":"2018-05-22T01:52:39","Uptime":"0T00:00:58","Vcc":3.481,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Ahmed","RSSI":92,"APMac":"00:15:6D:81:A6:82"}}
01:52:39 RSL: tele/DVES_EA9716/SENSOR = {"Time":"2018-05-22T01:52:39","ENERGY":{"Total":0.052,"Yesterday":0.000,"Today":0.052,"Period":0,"Power":0,"Factor":0.00,"Voltage":199,"Current":0.000}}
01:52:49 RSL: tele/DVES_EA9716/STATE = {"Time":"2018-05-22T01:52:49","Uptime":"0T00:01:08","Vcc":3.483,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Ahmed","RSSI":86,"APMac":"00:15:6D:81:A6:82"}}
01:52:49 RSL: tele/DVES_EA9716/SENSOR = {"Time":"2018-05-22T01:52:49","ENERGY":{"Total":0.052,"Yesterday":0.000,"Today":0.052,"Period":0,"Power":0,"Factor":0.00,"Voltage":199,"Current":0.000}}
01:52:59 RSL: tele/DVES_EA9716/STATE = {"Time":"2018-05-22T01:52:59","Uptime":"0T00:01:18","Vcc":3.482,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Ahmed","RSSI":84,"APMac":"00:15:6D:81:A6:82"}}
01:52:59 RSL: tele/DVES_EA9716/SENSOR = {"Time":"2018-05-22T01:52:59","ENERGY":{"Total":0.052,"Yesterday":0.000,"Today":0.052,"Period":0,"Power":0,"Factor":0.00,"Voltage":199,"Current":0.000}}
01:53:09 RSL: tele/DVES_EA9716/STATE = {"Time":"2018-05-22T01:53:09","Uptime":"0T00:01:28","Vcc":3.480,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Ahmed","RSSI":90,"APMac":"00:15:6D:81:A6:82"}}
01:53:09 RSL: tele/DVES_EA9716/SENSOR = {"Time":"2018-05-22T01:53:09","ENERGY":{"Total":0.052,"Yesterday":0.000,"Today":0.052,"Period":0,"Power":0,"Factor":0.00,"Voltage":199,"Current":0.000}}
01:53:19 RSL: tele/DVES_EA9716/STATE = {"Time":"2018-05-22T01:53:19","Uptime":"0T00:01:38","Vcc":3.481,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Ahmed","RSSI":98,"APMac":"00:15:6D:81:A6:82"}}
01:53:19 RSL: tele/DVES_EA9716/SENSOR = {"Time":"2018-05-22T01:53:19","ENERGY":{"Total":0.052,"Yesterday":0.000,"Today":0.052,"Period":0,"Power":0,"Factor":0.00,"Voltage":199,"Current":0.000}}
01:53:29 RSL: tele/DVES_EA9716/STATE = {"Time":"2018-05-22T01:53:29","Uptime":"0T00:01:48","Vcc":3.481,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Ahmed","RSSI":88,"APMac":"00:15:6D:81:A6:82"}}
01:53:29 RSL: tele/DVES_EA9716/SENSOR = {"Time":"2018-05-22T01:53:29","ENERGY":{"Total":0.052,"Yesterday":0.000,"Today":0.052,"Period":0,"Power":0,"Factor":0.00,"Voltage":199,"Current":0.000}}
01:53:39 RSL: tele/DVES_EA9716/STATE = {"Time":"2018-05-22T01:53:39","Uptime":"0T00:01:58","Vcc":3.480,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Ahmed","RSSI":98,"APMac":"00:15:6D:81:A6:82"}}
01:53:39 RSL: tele/DVES_EA9716/SENSOR = {"Time":"2018-05-22T01:53:39","ENERGY":{"Total":0.052,"Yesterday":0.000,"Today":0.052,"Period":0,"Power":0,"Factor":0.00,"Voltage":198,"Current":0.000}}
01:53:49 RSL: tele/DVES_EA9716/STATE = {"Time":"2018-05-22T01:53:49","Uptime":"0T00:02:08","Vcc":3.481,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Ahmed","RSSI":88,"APMac":"00:15:6D:81:A6:82"}}
[... cut ...]
02:17:30 RSL: tele/DVES_EA9716/SENSOR = {"Time":"2018-05-22T02:17:30","ENERGY":{"Total":0.052,"Yesterday":0.000,"Today":0.052,"Period":0,"Power":0,"Factor":0.00,"Voltage":206,"Current":0.000}}
02:17:40 RSL: tele/DVES_EA9716/STATE = {"Time":"2018-05-22T02:17:40","Uptime":"0T00:25:59","Vcc":3.483,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Ahmed","RSSI":90,"APMac":"00:15:6D:81:A6:82"}}
02:17:40 RSL: tele/DVES_EA9716/SENSOR = {"Time":"2018-05-22T02:17:40","ENERGY":{"Total":0.052,"Yesterday":0.000,"Today":0.052,"Period":0,"Power":0,"Factor":0.00,"Voltage":205,"Current":0.000}}`

curzon01, 2018-05-22:
Unfortunately stll missing `Status 0` so I can only assume some things...

but have a look
`01:51:59 RSL: tele/DVES_EA9716/SENSOR = {"Time":"2018-05-22T01:51:59","ENERGY":{"Total":0.052,"Yesterday":0.000,"Today":0.052,"Period":0,"Power":0,"Factor":0.00,"Voltage":198,"Current":0.000}}`

there it is.

The periodical sensor info is send by tele topic with the device name ( tele/DVES_EA9716...) - see [MQTT-Features](https://github.com/arendst/Sonoff-Tasmota/wiki/MQTT-Features):
>Telemetry data will be sent by prefix tele like tele/sonoff/SENSOR {"Time":"2017-02-16T10:13:52", "DS18B20":{"Temperature":20.6}}

HobbytronicsPK, 2018-05-24:
@curzon01 I have set ntpserver via command `NtpServer1 asia.pool.ntp.org`. Now NTP Server is setup as time is now correct. But there is noting received on MQTT under "**tele/DVES_EA9716/SENSOR**" and "**tele/DVES_EA9716/STATE**". However, device is publishing data to other topics.

> tele/DVES_EA9716/UPTIME : msg.payload : string[52]
"{"Time":"2018-05-24T11:02:00","Uptime":"0T00:04:39"}"

> tele/DVES_EA9716/INFO1 : msg.payload : string[95]
"{"Module":"Sonoff Pow","Version":"5.14.0","FallbackTopic":"DVES_EA9716","GroupTopic":"sonoffs"}"

> tele/DVES_EA9716/INFO2 : msg.payload : string[84]
"{"WebServerMode":"Admin","Hostname":"DVES_EA9716-5910","IPAddress":"192.168.50.222"}"

**Web Log Level is set to 4 More Debug. Have a look onto console log now.**
[Console_Output_1.txt](https://github.com/arendst/Sonoff-Tasmota/files/2034836/Console_Output_1.txt)

closed 2018-05-24 by HobbytronicsPK

Mantikor, 2018-06-10:
Sonoff POW with Tasmota 5.14.0 has same problem, if I build FW with Arduino studio - sensor not publish data in MQT, only RST. If i flash prebuild FW with OTA, all works fine.

honk0504, 2018-06-10:
Hi there. I just got my first Sonoff POW and installed 5.12.0. No values for voltage and current. I upgraded to 5.14.0 ... same here; no values for voltage and current. I now installed pre-built sonoff-allsensors.bin. unfortunatly ...  no values for voltage and current. 
Any suggestions?