arendst/Tasmota #2787: motion sensor on ailight

Dickesplumpel, 2018-05-20:
Hello everybody. I have soldered a motion sensor rcwl-0516 to an ailight bulb, Gpio1, GND and 3,3V. It works fine but all external switches I set to Gpio1 are switching the bulb so I can not use it for switching a dummy in domoticz which leaves the light on longer after motion is detected. When the bulb is switched by domoticz (another motion sensor) the motion sensor of the bulb "overwrites" the domoticz command (domticz turns on and the motion sensor of the bulb turns off when trigger gets low) - not shure if its easy to follow  ;-)

Anyone get this combo working?

Jason2866, 2018-05-21:
Same logic setup as #2792

Dickesplumpel, 2018-05-21:
As far as I can see: no! But maybe I oversea s.th.
I have many "dummy -relais" working for switches on not connected GPIOs because the mqtt message domoticz/in (and tele/xxx/POWER) is send immediatly, so reaction time is quick, but in my case when i set Switch2 and Relay2 in the ailight Switch2 switches Relay1, same with 3 and 4. 
I think I try the hc-sr501 (Vcc on H-Pin and it works with 3,3V) with maximum trigger time (should be around 3-5 minutes) and switchmode FOLLOW. Smarter is the hidden rcwl-0561, but: Nice looking is useless when it doesn`t work!

:-)) Now I see whats wrong: tele/xxx/POWER is send not POWER1. I think thats too much trouble for Theo to change the code only for me - I will try it on my own - thank you for answering.

Jason2866, 2018-05-21:
See option 26 !

Dickesplumpel, 2018-05-23:
Thank you very much for your answer. when I say "SetOption26 on" ( or 1 )  Ailight responds: "command unknown"

Jason2866, 2018-05-23:
Plz post status 0 for help

Dickesplumpel, 2018-05-24:
09:39:53 CMD: status 0

09:39:53 MQT: stat/Lampe1/STATUS = {"Status":{"Module":27,"FriendlyName":["Sonoff"],"Topic":"Lampe1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}

09:39:53 MQT: stat/Lampe1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"2T13:10:09","StartupUTC":"2018-05-21T18:29:44","Sleep":50,"BootCount":68,"SaveCount":1168,"SaveAddress":"F4000"}}

09:39:53 MQT: stat/Lampe1/STATUS2 = {"StatusFWR":{"Version":"5.13.1a","BuildDateTime":"2018.05.13 10:52:04","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}

09:39:53 MQT: stat/Lampe1/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["arminseins",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}

09:39:53 MQT: stat/Lampe1/STATUS4 = {"StatusMEM":{"ProgramSize":536,"Free":464,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}

09:39:53 MQT: stat/Lampe1/STATUS5 = {"StatusNET":{"Hostname":"Lampe1-1681","IPAddress":"192.168.1.10","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"68:C6:3A:BD:A6:91","Webserver":2,"WifiConfig":5}}

09:39:53 MQT: stat/Lampe1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_BDA691","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}

09:39:53 MQT: stat/Lampe1/STATUS7 = {"StatusTIM":{"UTC":"Thu May 24 07:39:53 2018","Local":"Thu May 24 09:39:53 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2,"Sunrise":"05:57","Sunset":"21:36"}}

9:39:53 MQT: stat/Lampe1/STATUS10 = {"StatusSNS":{"Time":"2018-05-24T09:39:53"}}

09:39:53 MQT: stat/Lampe1/STATUS11 = {"StatusSTS":{"Time":"2018-05-24T09:39:53","Uptime":"2T13:10:09","Vcc":3.451,"POWER":"OFF","Dimmer":50,"Color":"007F1100","HSBColor":"128,100,50","Channel":[0,50,7,0],"Scheme":3,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"arminseins","RSSI":50,"APMac":"28:C6:8E:B3:C8:9C"}}}

Dickesplumpel, 2018-05-24:
Hey - cool - works!! Switch2 doesn't switch relay1. So far so good, thank you very much. 
And now I have to play with the rules to send a domoticz/in command when switch2 goes from off to on because the old way with dummy relais doesn't work here - no relais2 appears when called.
Something like:
"on Switch2#ON do publish domoticz/in {…}"
may work, haven't studied the rules of the rules so far, will play with it in the near future. 
Many thanks for spending your time and knowledge to me, bye!

Dickesplumpel, 2018-05-25:
Hello again. Setoption26 doesn't "work in the device". The messages are all right, domoticz/in command is send but Switch2 still switches Relais1. I've switched back to 5.13.1a and my "dummy Relais" doing a wonderful Job.

closed 2018-05-25 by Dickesplumpel