arendst/Tasmota #2789: Support for Shelly 2 ?

h-tro, 2018-05-20:
Hi Theo,

can tasmota get support for Shelly 2?
https://shelly.cloud/shelly2/
Shelly 2 is very small wifi device with 2 relays, power meter and should also use ESP8266.
In most cases, it fits in a German flush mount box (60 mm Ø and 40 mm depth), including the previous switch and is therefore invisible.
I think there are many users who would like to use shelly 2 with tasmota.

curzon01, 2018-05-21:
Interesting device (not such low priced as Sonoff devices)
but might be not as simple as it looks because
> Has an integrated precise power meter

reloxx13, 2018-05-23:
i think the power meter would not work anymore.

read about other devices which power meter and it did not work after flashing tasmota.

h-tro, 2018-06-20:
I have opened a shelly2.

It contains an **ESP8266EX** with **16 MB flash memory** (WINBOND 25Q16BVFIG)
and a **power monitoring IC** MCP39F501 (https://www.microchip.com/wwwproducts/en/MCP39F501)

h-tro, 2018-06-30:
Hi senfkorn,

yes, I can confirm the assignment of the pins on the backside of the board.
I have the same result.

tmarquespt, 2018-08-05:
Hi, anyone successfully installed tasmota on Shelley? Even without the power meter feature?

cko-skverlag, 2018-08-17:
Where is the need of flashing the shelly (1)? Doesn't it work without cloud connection as the sonoff do? Thanks

curzon01, 2018-08-19:
If you do not like having your home devices in a foreign cloud system, Tasmota is a one of the best alternative firmware.

And to get it work you need to flash.

cko-skverlag, 2018-08-19:
But I thought the Shelly system comes without a cloud service?

Kommende Veranstaltung:
Symposium Taktische Lagen im Rettungsdienst<https://www.taktik-rettungsdienst.de/> 28.11.2018 - Düsseldorf
4. Zukunftsforum Rettungsdienst 28.–29.01.2019 - Hamburg

Tickets und weitere Informationen: <http://www.tickets.skverlag.de> https://tickets.skverlag.de

Stumpf + Kossendey Verlagsgesellschaft mbH | Postfach 13 61 | 26183 Edewecht
Tel.: +494405918119 | Fax: +49 4405 9181-79
E-Mail: cko@skverlag.de<mailto:cko@skverlag.de> | Internet: <http://www.skverlag.de/> www.skverlag.de<https://>

Handelsregister: AG Oldenburg B 120057
Geschäftsführer: Christoph Kossendey
USt.-ID-Nr.: DE 117828392
Steuer-Nr.: 69/200/02772

JPe2, 2018-08-20:
https://github.com/arendst/Sonoff-Tasmota/issues/2789#issuecomment-411327105
@h-tro   Did you manage to flash the Shelly-1 with Tasmota firmware already?

cko-skverlag, 2018-08-20:
> Antworten an: arendst/Sonoff-Tasmota <reply@reply.github.com>
>
> #2789 (comment)<https://github.com/arendst/Sonoff-Tasmota/issues/2789#issuecomment-411327105>

No – and i don’t understand why you would want to flash it? Isn’t there a free webserver onboard? From what i have read there is no need to connect the device to a cloud. But i’m not a technician.

Christoph

Kommende Veranstaltung:
Symposium Taktische Lagen im Rettungsdienst<https://www.taktik-rettungsdienst.de/> 28.11.2018 - Düsseldorf
4. Zukunftsforum Rettungsdienst 28.–29.01.2019 - Hamburg

Tickets und weitere Informationen: <http://www.tickets.skverlag.de> https://tickets.skverlag.de

________________________________
Stumpf + Kossendey Verlagsgesellschaft mbH | Postfach 13 61 | 26183 Edewecht
Tel.: +494405918119 | Fax: +49 4405 9181-79
E-Mail: cko@skverlag.de<mailto:cko@skverlag.de> | Internet: <http://www.skverlag.de/> www.skverlag.de<https://>

Handelsregister: AG Oldenburg B 120057
Geschäftsführer: Christoph Kossendey
USt.-ID-Nr.: DE 117828392
Steuer-Nr.: 69/200/02772

kuzba, 2018-08-22:
Here is nice video, showing tasmota on shelly1 with ready to flash binary. https://www.youtube.com/watch?v=J20hxfUTP9I

ascillato, 2018-08-23:
Let's centralize all shelly requests in one issue #3559

tmarquespt, 2018-08-23:
I have succefully flashed tasmota on the shelly1 with the .bin from link that kuzba posted.
The wifi manager mode didn't started (or at least I couldn't use it) so I had to manualy set it up with serial commands using Termite.
That .bin file already has options for the shelly1 and shelly2 so I believe the guy from TheHookUp had them both working. I'm going to flash shelly2 and let you know.

kuzba, 2018-08-23:
Are you able to run shelly again from mains (eg 230V)? I somehow destroyed the device (shelly1), it runs only from 3.3V pin header. But I probably swapped L and N wire, not sure if this can cause the trouble. So I am without toy now.

benzino77, 2018-08-23:
@tmarquespt let us know how to solder wires to flush shelly 2

tmarquespt, 2018-08-23:
@benzino77  senfkorn already exaplined the pinout on the 5th post of this page. I'm still thinking on how to do it as I would avoid soldering such small pins. I'll probably make a connector with some needles and flash it without soldering.
@kuzba  Mine is working. Changing L with N should make no difference as this is AC I have mine with a plug to the wall socket so I have no ideia which is the L or the N. Why do you say it's not working? It has no light. Try to connect with Termite (Mains OFF, gpio out) and send an AP command in order to see if it responds. Beaware that the connection scheme is different from the sonoff as the current does not go trough the relay by default and you have to inject it (as explained in the picture of the shelly manual included in the box)

kuzba, 2018-08-23:
@tmarquespt when running from mains, device gets really hot and does not connect to AP. When running from 3.3V thru pin header, connects to AP and responds to commands. I guess power supply is KO. PCB schematics will help.

tmarquespt, 2018-08-23:
@kuzba That is strange indeed. I can feel some temperature on mine but nothing out of the ordinary. It just might be an issue with the unit, you should try to return it and get a new one.

FYI: Theo just made Tasmota compatible with Shelly "Only INPUT available in v6.1.1.9 as Button1n option. See changelog."

arendst, 2018-08-23:
In the video GPIO05 is selected as Switch1.

I prefer to use Button1 as it will allow using the webpage for toggling the relay.

Can anyone confirm that the following config works just as great:
```
  { "Shelly1",         // Shelly 1 (ESP8266 - 2MB)
     GPIO_SWITCH1,     // GPIO00 Switch
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     0,
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     GPIO_REL1,        // GPIO04 Relay (0 = Off, 1 = On)
     GPIO_KEY1_NP,     // GPIO05 Button
     0, 0, 0, 0, 0, 0, // Flash connection
     0, 0, 0, 0, 0, 0
  }
```

h-tro, 2018-08-23:
Compiling error 6.1.1.9

benzino77, 2018-08-23:
> ...senfkorn already exaplined the pinout on the 5th post of this page...

That's true, but one or two pictures will be just great.

papaloewe, 2018-08-23:
Here is the Webpage where the video comes from:
http://www.thesmarthomehookup.com/the-shelly-1-smart-relay-is-it-better-than-the-sonoff-basic/

ethan4, 2018-08-23:
In this video he says the pull-up resistors have to be disabled, can that be done as part of the module preset? https://youtu.be/J20hxfUTP9I?t=416

tmarquespt, 2018-08-23:
@benzino77  Sure :)
@senfkorn / @h-tro Can you confirm that this is correct?

[image]

tmarquespt, 2018-08-23:
@ethan4 It should be working on the last version of Tasmota v6.1.1.9. If you want to make that change it has to be on the source before compile. You cannot change that after flashing. You might just flash the .bin file that is on the description of that video. Then you just select Shelly1 Module from configuration.

ethan4, 2018-08-23:
@tmarquespt I'll try that, but that means there will have to be a different .bin for this device always right? Can the ability to change that be added and then made customisable in the UI?

Not everyone using Tasmota is competent to compile it themselves.

h-tro, 2018-08-23:
**@arendst** 
I have the following configuration for my shelly1:

{ "Shelly1",   	         // Shelly1 (ESP8266 - 2MB)
     GPIO_KEY1,          // GPIO00 Button
     GPIO_USER,          // GPIO01 Serial TXD and Optional sensor
     0,
     GPIO_USER,           // GPIO03 Serial RXD and Optional sensor
     GPIO_REL1,            // GPIO04 Relay 1 (0 = Off, 1 = On)
     GPIO_SWT1_NP,     // GPIO05 External switch input 1
     0, 0, 0, 0, 0, 0,         // Flash connection
     0, 0, 0, 0, 0, 0
  },

After a first test it seems to work.
I will test it further and report in the next few days.

tmarquespt, 2018-08-24:
It seems to work on shelly**2** (the .bin from the video)
I was able to flash and have the relays clicking.

[image]

arendst, 2018-08-24:
@tmarquespt using your shelly2 can you confirm that when you perform ``toggle 1`` that indeed output ``NO1`` is switched?

I ask as I suspect the relays are connected reversed based on (ancient) information from Olimex (https://github.com/OLIMEX/olimex-iot-firmware-esp8266) and my interpretation.

So I would expect that when you perform ``Toggle 1`` output ``NO2`` would be switched when using the pre-compiled binary from thehookup.

arendst, 2018-08-24:
I have Shelly1 and 2 on order. Will dig in Shelly 2 for energy measuring once arrived....

tmarquespt, 2018-08-24:
@arendst At the moment I cannot as my Shelly2 is not working (my fault). But I tried both relays with a lamp and the toggle button in the web interface worked. I'm not 100% sure, but I think they were ok (Toggle1-sw1). Unfortunatly at this point have to wait until the new units arrive :(

h-tro, 2018-08-24:
@arendst 
Now I have shelly1 installed in a 40mm deep flush box behind the existing switch.
The switch is a normal light switch (Maintained Switch)
I use my tasmota configuration as described above in switchmode 0. 
My light switch works well with tasmota and shelly1

Here is a picture of the installation. 
(Attention: 230 V AC can kill you)
[image]

salopette, 2018-08-24:
Great!
Can you give me maybe correct worked bin file?

kuzba, 2018-08-24:
Great, can you pls share bin. I will test as well.

ethan4, 2018-08-24:
I've got some Shelly 1's on the way too so will test once they arrive.

h-tro, 2018-08-25:
Here is my bin file.
I think there will be a new tasmota version of Theo soon.
Then you should use the original bin-file from Theo.

[HT_SWfirmware.zip](https://github.com/arendst/Sonoff-Tasmota/files/2320445/HT_SWfirmware.zip)

salopette, 2018-08-26:
https://github.com/h-tro

Hi, I've successfully flopped your file, the switching also works, but the Shelly always goes on and off, always after about 1 minute without me doing anything

`07:58:19 RSL: STATE = {"Time":"2018-08-26T07:58:19","Uptime":"0T00:05:13","Vcc":3.201,"POWER":"OFF","Wifi":{"AP":2,"SSId":"KabelTv","RSSI":78,"APMac":"34:31:C4:CC:10:16"}}
07:58:19 RSL: SENSOR = {"Time":"2018-08-26T07:58:19","Switch1":"ON"}
07:59:14 RSL: RESULT = {"POWER":"ON"}
07:59:14 RSL: POWER = ON
08:01:08 RSL: RESULT = {"POWER":"OFF"}
08:01:08 RSL: POWER = OFF
08:02:00 RSL: UPTIME = {"Time":"2018-08-26T08:02:00","Uptime":"0T00:08:54"}
08:03:01 RSL: RESULT = {"POWER":"ON"}
08:03:01 RSL: POWER = ON
08:03:07 APP: Serielles Logging deaktiviert
08:03:19 RSL: STATE = {"Time":"2018-08-26T08:03:19","Uptime":"0T00:10:13","Vcc":3.192,"POWER":"ON","Wifi":{"AP":2,"SSId":"KabelTv","RSSI":96,"APMac":"34:31:C4:CC:10:16"}}
08:03:19 RSL: SENSOR = {"Time":"2018-08-26T08:03:19","Switch1":"ON"}
08:03:25 RSL: RESULT = {"POWER":"OFF"}
08:03:25 RSL: POWER = OFF
08:04:02 RSL: RESULT = {"POWER":"ON"}
08:04:02 RSL: POWER = ON
08:04:54 RSL: RESULT = {"POWER":"OFF"}
08:04:54 RSL: POWER = OFF
08:05:04 RSL: RESULT = {"POWER":"ON"}
08:05:04 RSL: POWER = ON
08:06:25 RSL: RESULT = {"POWER":"OFF"}
08:06:25 RSL: POWER = OFF
08:08:19 RSL: STATE = {"Time":"2018-08-26T08:08:19","Uptime":"0T00:15:13","Vcc":3.211,"POWER":"OFF","Wifi":{"AP":2,"SSId":"KabelTv","RSSI":92,"APMac":"34:31:C4:CC:10:16"}}
08:08:19 RSL: SENSOR = {"Time":"2018-08-26T08:08:19","Switch1":"ON"}
08:09:14 RSL: RESULT = {"POWER":"ON"}
08:09:14 RSL: POWER = ON
08:09:29 RSL: RESULT = {"POWER":"OFF"}
08:09:29 RSL: POWER = OFF`

h-tro, 2018-08-26:
**@arendst , @salopette ,**
My shelly 1 configuration from above is not stable.
Last night, the light in the hall turned on and off.
It seems GPOI 0 must not be configured as Button (GPIO_KEY) or Switch (GPIO_SWT).
I have changed my configuration and will keep watching.

salopette, 2018-08-26:
Can you share your new .bin?
Will then watch it too!

eazuka, 2018-08-26:
@arendst,
can't wait for you to receive your Shelly 1 & 2. As always, i'm interested in the energy measuring feature and of-course the relay both working well. So i will be following your progress. Thanks for your great work on Tasmota.

h-tro, 2018-08-27:
**@salopette**
With my HT_SWfirmware02.zip it is now stable.
Does it work for you too?

thehookup, 2018-08-27:
@arendst Hey Theo, Sorry I've been absent in this thread, didn't realize there topic had been revivied.  What can I do to help you with the shelly stuff?  I've got my shelly1 and shelly2 flashed with tasmota 6.1.1 working with the .bin file that I posted in the video.  I can also send you some info about the power monitoring that I got from the guys at Shelly.  I was going to try to do the integration myself to save you some work, but as I got deeper into it I figured I should probably let you do it since I was having to modify some core files.  

Anyways, as you can probably tell, I love Tasmota and am super grateful for all your hard work.

arendst, 2018-08-28:
@thehookup considering the amount of activity I thought why not add the shelly as they are nice small products too.

I ordered two of them so I can do some hands-on testing before I want to release the final config. I'm still not convinced to configure either button or switch...

I'm interested in the energy monitoring too. I found some old Olimex stuff on github but if you have the latest info I would like to have it too pls. You can contact me using the contact link on github.

arendst, 2018-08-28:
@h-tro Thx, that's useful information.

kiwijunglist, 2018-08-29:
So can I flash tasmota on a shelly 2, then wire it up to 2x toggle mains powered switches behind my light switch, a switch operating each relay. And can I do this without modifying the device?  Thanks.

thehookup, 2018-08-30:
@kiwijunglist the shelly 2 doesn't have it's GPIO broken down into headers, so you'll need to solder in wires to the pads to do the tasmota flash.

h-tro, 2018-08-30:
**@arendst** 
If GPIO 0 or GPIO3 is configured as a switch or button, the relay is not stable. The relay randomly turns on and off.
I think in a module type Shelly1 or Shelly2, GPIO 0, GPIO 1 and GPIO 3 should not be configured, not even as GPIO_USER. Then there are no problems.

Shelly also works with mains voltage. For this reason, GPIO 0, GPIO 1 and GPIO 3 in a module type Shelly1 should not be configurable as input or output because the pins are external.

thehookup, 2018-08-30:
@arendst @h-tro 

I agree with @h-tro , these are the profiles that I had in my .bin file:

`  { "Shelly 1",         // Any ESP8266/ESP8285 device like WeMos and NodeMCU hardware (ESP8266)
     0,                 // GPIO00 
     0,                 // GPIO01 
     0,                 // GPIO02 
     0,                 // GPIO03 
     GPIO_REL1,         // GPIO04 Relay O and I
     GPIO_SWT1,         // GPIO05 SW pin
     0, 0, 0, 0, 0, 0,  // Flash connection
     0,                 // GPIO12 
     0,                 // GPIO13 
     0,                 // GPIO14 
     0,                 // GPIO15
     0,                 // GPIO16 
     0                  // ADC0
  },
  { "Shelly 2",         // Any ESP8266/ESP8285 device like WeMos and NodeMCU hardware (ESP8266)
     GPIO_USER,        // GPIO00
     GPIO_USER,        // GPIO01 
     GPIO_USER,        // GPIO02
     GPIO_USER,        // GPIO03
     GPIO_REL1,        // GPIO04
     GPIO_REL2,        // GPIO05
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_SWT1,        // GPIO12
     GPIO_USER,        // GPIO13
     GPIO_SWT2,        // GPIO14
     GPIO_USER,        // GPIO15
     GPIO_USER,        // GPIO16
     0                 // ADC0
  }`

h-tro, 2018-08-30:
**@thehookup** 
If you are using the latest developer version of Tasmota, you can use GPIO_SWT1**_NP** or GPIO_SWT2_**NP**.

Can you confirm that the relay of Shelly1 is not stable when GPIO 0 or GPIO 3 is configured as switch or button?
You can test this if you switch to module type Generic and manually configure the GPIO's.

ethan4, 2018-09-01:
Just some testing feedback,

I'm running HT_SWfirmware02.zip and it's working perfectly with switchmode 0 set. No issues with connectivity or operation :)

arendst, 2018-09-02:
Received my shellies yesterday.

Released in Tasmota today but without energy monitoring what I still need to investigate. 

The shelly 1 works fine with both the switch between SW and Neutral or SW and Life.

@aetjansen I haven't had the time to investigate but perhaps others have.

h-tro, 2018-09-02:
**@arendst** 
I just flashed developer version 6.2.0.1 on Shelly1.
It works. I will keep watching. Light switch corridor that I use daily.

arendst, 2018-09-02:
Just blew up my shelly 2 when connecting SW1 input to, what I thought Neutral....

[image]

The guide said you should connect SW1 to Life but I thought, hey it works on Shelly1 why not on Shelly2...

Stupid. Stick to the guide for both Shelly1 and Shelly2: **Connect the SW only to L (Live) wire**

My mistakes were that I swapped life and neutral as inputs to shelly2 while connecting SW1 to N.

So much for implementing the energy monitoring :-(

thehookup, 2018-09-02:
Ouch, yeah.  Since the switch pins are essentially supposed to be GND they need to share the same ground reference as the 8266 (which is the L pin). 

I made this diagram to help people out: https://i.imgur.com/w1T18Ld.jpg

eazuka, 2018-09-02:
@arendst lol, thank God you live to report this.

h-tro, 2018-09-02:
230V mains voltage is dangerous.
There is no difference between Shelly1 and Shelly2.
If -L- and -N- are connected to the correct input terminals, SW (Shelly1) and S1 or S2 (Shelly2) must receive -L- from external switches.

digiblur, 2018-09-03:
Thanks for the commit on the Shelly templates.  Could we get the TX, RX and GPIO0 pins added as well?  Of course the Generic works fine.  

[image]

arendst, 2018-09-03:
As for both Shellys the ESP8266 Gnd is directly connected to the AC Life (L) wire I won't provide TX, RX and GPIO0 support.

The Shelly2 does not have user accessable GPIOs and the TX and RX are being used by energy monitoring

kiwijunglist, 2018-09-03:
@thehookup 

For Shelly 2 can i just hold the wires against the pins for flashing like i do with the sonoffs rather than soldering?

With the Shelly 2 and tasmota can i control both the relays using a mains voltage toggle switch?

Thanks.

digiblur, 2018-09-03:
@arendst No problem, I'll just use generic.  I could see the issue if someone used a metal switch or something else odd, but it's around the same danger level with the Sonoff basic's lack of proper isolation in my eyes which is why I use the capacitive buttons behind a face plate to remove any direct human contact with the Sonoff and/or Shelly.

[43 more replies]

closed 2018-09-21 by ascillato2