arendst/Tasmota #2792: Tasmota - d1 mini

lscosta90br, 2018-05-21:
Dear Ones,

I'm configuring a d1-mini with 2 components:
- reed switch
- relay

The purpose is to control whether the garage door is open or closed and open the garage door through the relay.
in configure module / module parameters:
- Module type: 18 Generic
- D1 GPIO5: 17 Relay1
- D5 GPIO14 Sensor: 09 Switch1

The relay is triggered without problem.
The reed switch does not provide any door open information.
How do I look like an open or closed door information console?
With the informed results of the console I will make the decisions in the automation software

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :

**(Please, remember to close the issue when the problem has been addressed)**

MYLE-01, 2018-05-21:
add a sec relay

D5 = switch2 the D6 relay2 and know you will get MQTT message from the switch

have a look at how I tasmota my fan
https://community.home-assistant.io/t/tasmota-the-desktop-fan/53523

I put a motion sensor on D2 witch is a ON/OFF but had to add a relay3 to the RX  then it would send the MQTT message

lscosta90br, 2018-05-21:
MYLE-01,

It does the following:

[image]

I open and close the reed switch and nothing happens on the console:

`00:00:00 Project sonoff Garage Relay  (Topic d1-mini-02, Fallback DVES_5A09B3, GroupTopic sonoffs) Version 5.14.0-2_3_0
00:00:00 WIF: Connecting to AP1 magneto_v3_bgn in mode 11N as d1-mini-02-2483...
00:00:03 WIF: Connected
00:00:03 DNS: Initialized
00:00:03 HTP: Web server active on d1-mini-02-2483.local with IP address 192.168.2.121
00:00:05 MQT: Attempting connection...
00:00:05 MQT: Connected
00:00:05 MQT: tele/d1-mini-02/LWT = Online (retained)
00:00:05 MQT: cmnd/d1-mini-02/POWER = 
00:00:05 MQT: tele/d1-mini-02/INFO1 = {"Module":"Generic","Version":"5.14.0","FallbackTopic":"DVES_5A09B3","GroupTopic":"sonoffs"}
00:00:05 MQT: tele/d1-mini-02/INFO2 = {"WebServerMode":"Admin","Hostname":"d1-mini-02-2483","IPAddress":"192.168.2.121"}
00:00:05 MQT: tele/d1-mini-02/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 MQT: stat/d1-mini-02/RESULT = {"POWER1":"ON"}
00:00:06 MQT: stat/d1-mini-02/POWER1 = ON
00:00:06 MQT: stat/d1-mini-02/RESULT = {"POWER2":"OFF"}
00:00:06 MQT: stat/d1-mini-02/POWER2 = OFF
00:00:06 MQT: stat/d1-mini-02/RESULT = {"POWER1":"OFF"}
00:00:06 MQT: stat/d1-mini-02/POWER1 = OFF
13:05:29 MQT: tele/d1-mini-02/STATE = {"Time":"2018-05-21T13:05:29","Uptime":"0T00:00:13","Vcc":2.752,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"magneto_v3_bgn","RSSI":96,"APMac":"64:66:B3:C2:49:D1"}}
13:05:29 MQT: tele/d1-mini-02/SENSOR = {"Time":"2018-05-21T13:05:29","Switch2":"ON"}
13:10:03 MQT: stat/d1-mini-02/RESULT = {"POWER1":"ON"}
13:10:03 MQT: stat/d1-mini-02/POWER1 = ON
13:10:04 MQT: stat/d1-mini-02/RESULT = {"POWER1":"OFF"}
13:10:04 MQT: stat/d1-mini-02/POWER1 = OFF
13:10:05 MQT: stat/d1-mini-02/RESULT = {"POWER2":"ON"}
13:10:05 MQT: stat/d1-mini-02/POWER2 = ON
13:10:05 MQT: stat/d1-mini-02/RESULT = {"POWER2":"OFF"}
13:10:05 MQT: stat/d1-mini-02/POWER2 = OFF
13:10:32 MQT: tele/d1-mini-02/STATE = {"Time":"2018-05-21T13:10:32","Uptime":"0T00:05:16","Vcc":2.751,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"magneto_v3_bgn","RSSI":96,"APMac":"64:66:B3:C2:49:D1"}}
13:10:32 MQT: tele/d1-mini-02/SENSOR = {"Time":"2018-05-21T13:10:32","Switch2":"ON"}
13:11:07 MQT: stat/d1-mini-02/RESULT = {"POWER1":"ON"}
13:11:07 MQT: stat/d1-mini-02/POWER1 = ON`

Note the continuous physical configuration:

Relay - is on 5v, GND, D1
Reed switch - is on 3.3V, D5

What I understood from your text was for:
- create switch2 instead of swtich1 (D5)
- D6 => Relay2
What did I do wrong ? :)

Jason2866, 2018-05-21:
You need a virtual relay for your reed switch to get an independent mqtt message from relay1.
With the suggested setup you get a power2 on off message when the reed contact is closed (on) or open (off).

lscosta90br, 2018-05-21:
Dears, 

The power2 only appears at startup.
I move the reed switch image or close it and I do not see any log message on the console.
What am I doing wrong ?

[image]

[image]

[image]

ascillato, 2018-05-21:
So as to use switch2, you need to define also switch1. If you are not going to use switch1, put it on an unused pin.

I recommend to use directly switch1 and rules for the logic.

See the commands for switchmode and switchtopic to have the behaviour you want. 

Also add some ruletimer for extra protection if the door is taking too much for opening

lscosta90br, 2018-05-21:
I can not solve it !!! (I.e.

Someone would have a step step: ((
I'm lost !!

ascillato, 2018-05-21:
Hi,

There are some information missing. The relay is just giving a signal to a control that opens the door?
How then you close it?

I assume (JUST FOR EXAMPLE) that is it just a signal and not a direct control of the motor, and what you want is only to open the door and to know if the door is closed or not closed.

There are several ways to solve that automation. One is the following example.

**1-Reed Switch connection**

The reed switch is a normal close switch, meaning that with the magnet near to it, it will connect its conctacts.

As it is connected on your diagram it has a 3.3v state when the magnet is close and floating when it is far from the magnet. The floating state can not be read by the esp8266. you have to pull up or down with a resistor. You need to connect the reed switch like pushbuttons are conected.

```
+ 3.3v -----------------RESISTOR-------------- PIN on ESP8266
                                       |
                                       |
                                REED SWITCH
                                       |
                                       |
 GROUND --------------------------------
```

With that connection you will see on the esp8266 the status of the reed switch

**2-PIN Configuration**

Set the relay1 to your pin where the relay is
Set the switch1 to your pin where the reed switch is

I recommend to disable saving on eeprom relay status and also to be outputs set to off at startup for safety. Also be sure of the right switchmode selected. Also, as I assume it is just a signal, after you send the command to open (relay on), the relay goes off after 2 seconds.

On console: `backlog setoption0 0; poweronstate 0; switchmode1 0; pulsetime1 20`

**3-Control Logic**

Using rules:

* To Report the status change of the reed switch to a home automation system
console: `rule on switch1#status do publish stat/sonoff/garagedoor %value% endon`

* turn on rules
console: `backlog rule 1; rule 4 `

**4-Check**

to see the status of the switch
console: `status 8`

ascillato, 2018-05-21:
That is only an example. You will have to adapt it to your needs

MYLE-01, 2018-05-22:
Dont look at the Switch look at the Relay2

you should be seen a POWER2 ON/OFF in the MQTT log
the  switch will set the relay to ON/OFF but tasmota will send the ON/OFF of the relay

hope this make cents

MYLE-01, 2018-05-22:
Here is my set up
[image]

Switch3 is the motion sensor (your would be the Reed switch)
I have a Relay3 pointing to the RX PIN BUT NO RELAY

then when i get motion Switch3 turn on and Tasmota turn on the Relay3 and Tasmota send a MQTT message
[image]
I have node-red Looking at the POWER3 topic and doing its thing

lscosta90br, 2018-05-22:
Thank you all

What was doing wrong was connecting the reed switch module after I put a resector everything worked ..
I thank everyone who helped me

Regards,
Leandro Costa

ascillato, 2018-05-22:
> disable saving on eeprom relay status

on console type: `setoption0 0`

All the help about commands is on the wiki at https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

closed 2018-06-13 by ascillato2

RO-ny9, 2019-04-26:
> Dear,
> 
> I'm configuring a d1-mini with 2 components:
> 
> * reed switch
> * relay
> 
> The purpose is to control whether the garage door is open or closed and open the garage door through the relay.
> in configure module / module parameters:
> 
> * Module type: 18 Generic
> * D1 GPIO5: 17 Relay1
> * D5 GPIO14 Sensor: 09 Switch1
> 
> The relay is triggered without problem.
> The reed switch does not provide any door open information.
> How do I look like an open or closed door information console?
> With the informed results of the console I will make the decisions in the automation software
> 
> Note:
> 
> * Relay - is on 5v, GND, D1
> * Reed switch - is on 3.3V, D5
> 
> Make sure these boxes are checked [x] before submitting your issue - Thank you!
> 
> * [ ]  Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
> * [ ]  Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
> * [ ]  Searched the problem in the forum (https://forum.digital-pig.com/)
> * [ ]  Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
> * [ ]  Hardware used :
> * [ ]  Provide the output of command `status 0` :
> 
> **(Please, remember to close the issue when the problem has been addressed)**

Hi, please, can you send me your code from this issue? Wemos D1 mini + Reed switch magnet together? :)