arendst/Tasmota #2795: Control one device with another

mizrachiran, 2018-05-21:
Hi Guys, 

I would like to know if it's possible to control 1 Tasmota device with another.
(will be used to control living room lights from 2 different locations in the house).

What i had in mind is that 1 device will be connected and will 'really' control the light. the other will only have switches defined.

While I can set the state topic to be the same, it doesn't seem right as the topic used to report LWT and state.

Is such scenario already supported by Tasmota?
Any suggestions?

Thanks!!!

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ X] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :

**(Please, remember to close the issue when the problem has been addressed)**

wongnam, 2018-05-21:
Try use SwitchTopic https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic

- default Switch1 press it will send a default or custom topic(depending to your config) via MQTT with default postfix is POWER1 (ed: cmnd/2nd_sonoff/POWER1, Switch2 if pressed it will send command via MQTT with default postfix is POWER2
- i just wonder to know if any way to use Switch1 from the 1st_sonoff to remote the 2nd_sonoff Relay2

wongnam, 2018-05-21:
ok i found rule can fix my issue : use Switch1 from the 1st_sonoff to remote the 2nd_sonoff Relay2

mizrachiran, 2018-05-21:
Thank you @wongnam.
I looked at the documentation, SwitchTopic looks promising so far...

One question, is it possible to control one device with another without any third party setup? (OpenHAB \ HomeAssistnat). it would be easier to maintain if the 'remote' device would know the other topic and control it directly. (in such case it will have to hold the updated state as well).

Thanks,
Ran

wongnam, 2018-05-21:
in this case I recommend use Rules for more powerful and flexible.

https://github.com/arendst/Sonoff-Tasmota/wiki/Rules

note: Disable ButtonTopic as it overrides rules for buttons: ButtonTopic 0

wongnam, 2018-05-21:
Look at switchmode

wongnam, 2018-05-21:
SwitchMode 3

mizrachiran, 2018-05-21:
Using SwitchMode 3, will it send toggle command?

wongnam, 2018-05-21:
Yes. Send “TOGGLE”

In case of using onboard Button use ButtonTopic

mizrachiran, 2018-05-21:
@wongnam thank you so much for the assistance.
I will test it today and update this case accordingly.

Ran

wongnam, 2018-05-21:
Eg: type in console.
Case 1: sonoff1 connect an external button on GPIO14 + GND to remote a sonoff2
- SwitchTopic sonoff2
- SwitchMode 3

Case2: use an onboard button on sonoff1 to remote sonoff2
- ButtonTopic sonoff2

wongnam, 2018-05-22:
@mizrachiran Try knx feature, it's more than what you expect

https://github.com/arendst/Sonoff-Tasmota/wiki/KNX-Features

Especially it does not need to use mqtt broker

mizrachiran, 2018-05-22:
I just tried the SwitchTopic, it seems to be working.
The only thing that i don't understand is why the mqtt topic that sent from this device is not addressed to specific relay.
it sends "cmnd/MySwitchTopic/POWER" while I was expecting POWER1, as I set this GPIO to Switch1.
Is that intentional?

ascillato, 2018-05-22:
In Tasmota, POWER means POWER1. If you want Tamota to explicit POWER1 use on the console: `setoption26 1`

Please, see commands in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Commands)

mizrachiran, 2018-05-23:
That make sense.
But why when I set this GPIO to Switch2 \ Switch3 it still sends the same command (POWER)?

mizrachiran, 2018-05-23:
I solved it by defining non existing Relays in the module config.
when only switches were defined, it's always send POWER1.

@wongnam, @ascillato, Should I open bug for that or is it by design?

wongnam, 2018-05-23:
i think it is feature limited if you use ButtonTopic to send command to the same device or another device relay 2, relay 3.. 
- Try to use rule, Rule can publish a command like cmnd/sonoff/POWER3
    try this. Rule on Button2#State do Publish cmnd/sonoff/POWER3 %Value% endon
    note: Disable ButtonTopic as it overrides rules for buttons: ButtonTopic 0
- KNX also very easy to remote another device by button.

wongnam, 2018-05-23:
if you defined Switch2 Switch3 then change the word "Button" to "Switch" in the rule.

maxen0815, 2018-05-23:
Hi guys,
I'm reading this topic since I'm interested in this function, too. Can you tell me if there also is a solution which is able to run without MQTT broker in between?
The idea is to have one tasmota controlled button directly sending an HTTP request within the same network to trigger another tasmota device like power switch. Could call is Wifi remote control...

BR maxen

mizrachiran, 2018-05-23:
@maxen0815, this scenario is possible using KNX \ Rules.

wongnam, 2018-05-23:
@Ran because of limited issue so Tasmota implement rules.
See this article “A problem with this solution is that on a Sonoff 4CH all four buttons will be sending the same MQTT topic using only a different Power index number like cmnd/buttontopic/power3 toggle.

By using a rule a single button can now send any MQTT message allowing much more flexibility.” that I capture it from wiki.

@maxen0815 Go with KNX, yep no need MQTT broker.

mizrachiran, 2018-05-23:
This is actually what I want, that every switch \ button will send the same topic with different index.
What's happening is that all of the buttons trigger the same command...

maxen0815, 2018-05-23:
Thanks for these quick responses! I'll give KNX a try.

wongnam, 2018-05-23:
Because your devices do not have a relative relay assign to a GPIO, If 1 button go with 1 relay that you will see it will send same topic with different index related to the relay number.

wongnam, 2018-05-23:
My screen shot FYR, i hope it can help.
[image]
[image]

mizrachiran, 2018-05-23:
Got it, that was my workaround - to define non existing relays.
However, if I want a specific device to act as a remote only it should be possible to configure only switches & buttons and achieve the same outcome.

wongnam, 2018-05-23:
I don’t expect it will work with simple  buttontopic method If you don’t assign relays.
It will work if you go with Rules :)

mizrachiran, 2018-06-02:
Closing this thread.
There are 3 options for those who want a Tasmota device to act as a remote control:
1. KNX, fast and very customizable.
2. MQTT define non existing relays and set StateTopic, will result in many redundant mqtt messages.
3. Rules - although it's working as expected, i find this option harder to maintain.

I ended up using KNX and it's working great so far.

closed 2018-06-02 by mizrachiran

gabryk91, 2018-12-01:
@mizrachiran I flashed latest buil 6.3.0.14 on a sonoff basic but I can't see the knx menu, how to activate it? Thanks

iliyang, 2020-04-08:
Sorry to revive an old thread, but I'm looking for a way to communicate between Tasmota devices as directly as possible, i.e. relying only on an active WiFi connections. I haven't explored all possible options yet but it seems that the `WebSend` command (documented in [this table](https://tasmota.github.io/docs/Commands/#wi-fi)) can be used to send commands from one Tasmota device to another? I'm thinking this could be particularly useful for implementing two-way light switching without the need for an MQTT broker on another middle man. Does this make sense? Is there a better way to achieve such direct communication?

lalo-uy, 2020-04-08:
Is doable with rules.
In order to do that you will need fixed ip or local names working  in the
router.

iliyang, 2020-04-08:
That's exactly my idea. Thanks for the confirmation! Tasmota is awesome :)

ascillato, 2020-04-08:
Direct communication between Tasmotas can be achieved by:

* Rules and websend command (HTTP API)
* by MQTT publising in the cmnd/topic/ of the other device (MQTT)
* KNX (just enable the protocol. no extra hardware required - see docs at KNX)

pcdiem, 2020-04-08:
A new option comes with a PR that was merged last night. You can use device groups to control one or more Tasmota devices directly using UDP. See [Device Groups](https://tasmota.github.io/docs/Device-Groups/). The latest development release includes the DevGroupSend command that lets you send updates to a device group from a command.