arendst/Tasmota #2796: sonoff basic as generic not booting

rey90, 2018-05-21:
Hello

i have bought an sonoff basic. flashed the newest tasmota and changed the moduletype to generic, because i wan´t to connect an SR04.

[image]

After connecting the SR04 the sonoff is not booting - anyone an idea?

wongnam, 2018-05-21:
make sure you have to disable SerialLog (SerialLog 0) if you use GPIO1 and GPIO3 for SR04

wongnam, 2018-05-21:
No need to change the module to Generic.
note that: do not connect SR04 direct to Sonoff Basic that the echo signal output will damage the sonoff due to SR04 using 5V.

rey90, 2018-05-21:
i have not connect the sr04 directly. i have bought JSN-sr04t which comes with an board and should also run with 3,3v. If i don´t change the module to generic , were should i connect the sr04? i will need 3,3v + tri + echo + gnd

wongnam, 2018-05-21:
Sonoff Basic still have GPIO1 and GPIO3

sê wiki https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Basic

[image]

rey90, 2018-05-21:
sorry my mistake. changed the mode back and also find this settings. but the module is still not booting up

Jason2866, 2018-05-21:
Does the module boot normal without the sensor?
If yes connect the sensor to gpio 1 and gpio 14.
Maybe the sensor iritates serial in at boot time....

rey90, 2018-05-22:
ok thanks . i have connected the sensor to gpio1 and gpio 14 and now the module is booting . But the value shown is completly wrong. TRI and ECH are connected in the correct way. Is there any option to debug this?

rey90, 2018-05-23:
ok thanks for this information - i have also found an article about problems with 3,3v. Could anyone quickly help me to find 5v on the sonoff basic.

Jason2866, 2018-05-23:
Take a look here
https://www.itead.cc/wiki/Sonoff
You will find a schematic diagramm here

reloxx13, 2018-05-23:
scrape this line on the board:
[image]

turn the board over and solder a bridge like here:
[image]

solder in a pin to the hole you just soldered a bridge to.

more info:
[image]
[image]

wongnam, 2018-05-24:
@reloxx13 Beware, do not connect Echo output directly to the GPIO directly due to the voltage level is not compatible(5V vs 3.3V).

Possible way is using a zenner diode to cut voltages above 3.3V
[image]

rey90, 2018-05-24:
thanks for your help. i will test it

rey90, 2018-06-08:
just an update from my site: i have tried to take out the 5V without success, but this is till my bad soldering skills - will go on

ascillato2, 2018-07-16:
Hi

Seems that your question has been addressed. Closing this issue. Please, reopen if needed. Thanks

closed 2018-07-16 by ascillato2

Solraven, 2020-08-05:
firstly thank you. 
with the info on this page and a few other sources i have installed tasmota on esp01, connected it to a sr04t using a logic level converter, with 5v source connected to the sr04t side. i had given up on this project a year or so ago, i was going to try teaching myself and started off with blink and got frustrated, i was thinking i did something wrong and thought i had fried all 3 of my esp01's. at this point, i honestly think i just could not get them to boot because i didnt jump vcc and ch_pd. and than when i tried to reflash it it would fail, probably forgot to wipe first. my father was ill, and i didnt have the time or patience. than my father passed, and this stuff got tossed in a box until now.

the distance on the webgui comes and goes, any ideas why? is there a way without rewriting to change the distance to be something else, possibly a percentage between two set values, or even inject an equation to get Gallons?

the goal is to mount it in a pvc pipe and put that down in a water tank, to measure water level. at this point i could use some direction on where to go with this. the goal is get the level on a daily basis with minimal effort. possibly displayed on an lcd, or being able to ask alexa for the level. 

 when i ran power up the hill to the tank i put a cat5e cable going to a small poe router that is hooked to my asus ac68r router. i was reading up on setting up merlin on the ac68r as im under the impression that would allow me to set up a mqtt broker. are there better solutions? i was playing around with wemo emulation but i read it doesn't support sensors in the forums, tho https://tasmota.github.io/docs/Alexa/ states specificaly that hue does not support sensors but does not say anything about wemo. can alexa get info from a mqtt broker to answer a query? could i get a graphical level by loading tasmota_display on a second module and attaching an lcd? would i only get a digital distance displayed? does a esp01 have the required ports to utilize an lcd?

any help is appreciated?