arendst/Tasmota #2797: Advice for Newbie please

Slowsteve, 2018-05-21:
Thyankyou for taking the time to create all this great work.
I have been following your instructions and have added the generic board. I have v 1.8.5 and there is no option for 1M(no SPIFFS), the nearest seem to be 1M(64K SPIFFS) or 512K(noSPIFFS) which shall I choose?
Am I required to enter wi fi and mqtt broker details at this stage ; i'm a new comer and haven't yet learnt about mqtt. If I need to, van you give me any quick ponters please?

andrethomas, 2018-05-22:
I'm guessing something went wrong with your esp8266 board support install - mine seems to show several variants as follows

https://snag.gy/rOMH3u.jpg

As bare minimum I would recommend you at least configure the wifi ap you're using for your wireless network as to make it easier to connect to the device.

See user_config.h and edit
#define STA_SSID1              ""        // [Ssid1] Wifi SSID
#define STA_PASS1              ""      // [Password1] Wifi password

All the rest can be done through the web interface when it has connected to your wifi network.

andrethomas, 2018-05-22:
> I think everything after the // is your comment so I just need to insert my wifi ssid and password between the '' '' and leave everything else in the file unchanged?

Correct!

The simplest I found was to create a portable install of the arduino ide and then add esp8266 support and finally the tasmota source and libraries to it and then compile.

The important step is to not open the normal arduino ide but the portable version from its own executable and then open the sonoff sketch from within that arduino instance. This way the configuration of arduino for your tasmota development is isolated from all other arduino related development.

If you like I can push a zip file with a single folder somewhere for the current version that includes everything you need in a single folder.

It really isnt rocket science once you get your head wrapped around it.

Cheers

Slowsteve, 2018-05-22:
Wow, thanks again. Although it's taking me a long time to do something that looked straightforward, I think I'm learning other stuff along the way. I'm going to do a bit more googling to learn about some of the terms you use, particularly, "portable version from its own executable and then open the sonoff sketch from within that arduino instance." Your offer to 'push' a zip file  sounds like a great offer but I'll need time to find out how that works.

andrethomas, 2018-05-22:
Created a repo for you https://github.com/andrethomas/arduino_esp8266_tasmota.git

clone it to c:\

run the arduino.exe inside it and open the sketch in portable\sketches\sonoff

Should work just fine

andrethomas, 2018-05-22:
Here you go

https://thehackbox.org/downloads/arduino_esp8266_tasmota.7z

Extract it so that it is located in c:\arduino_esp8266_tasmota and the arduino.exe file is in that folder.

Run the arduino.exe file and it should open the included version of sonoff-tasmota (V5.14.0a)

In short, its a portable version of arduino1.8.5 downloaded from arduino.cc

I then added ESP8266 board support from [https://github.com/esp8266/Arduino] (v2.4.10)

I then added the source and libraries from Sonoff-Tasmota from [https://github.com/arendst/Sonoff-Tasmota]

These are copied into arduino_esp8266_tasmota\portable\sketchbook\libraries and arduino_esp8266_tasmota\portable\sketchbook\sonoff respectively as per instructions found at https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE

The rest should work out perfectly.

Cheers and good luck

Slowsteve, 2018-05-23:
I had various errors when flashing such as espcomm_upload_mem failed and espcomm_open failed and after more googling I traced the problem to be low voltage due to  a poor connection on my brand new 
usb to uart board. Annoying. Flashing was then done! Yeah! The rf bridge didn't connect to my wireless so I tried
3 short presses: Start Wifi smartconfig allowing for SSID and Password configuration using an Android mobile phone with the ESP8266 SmartConfig app. The LED will blink during the config period. A single button press during this period will abort and restart sonoff.
                 but this failed.
Then I did
4 short presses: Start Wifi manager providing an Access Point with IP address 192.168.4.1 and a web server allowing the configuration of Wifi.
                  it worked and I set up my wifi (even though I had edited the user config file previously). 
Elation! Thankyou so much for all your help in getting there.

I was then exploring and tapped on firmware update OTA which was a mistake as I hadn't made mqtt settings. When looking in console there is a repeated attempt to connect
 17:55:04 MQT: Attempting connection...
17:55:05 DNS: Query done. MQTT services found 0
17:55:05 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
17:55:16 MQT: Attempting connection...
17:55:17 DNS: Query done. MQTT services found 0
17:55:17 MQT: Connect failed to :1883, rc -2. Retry in 10 sec     etc
I've looked for a command to exit this but can't find one and am afraid to simply dabble and risk breaking things. 
I promise I won't keep pestering and have tried to research.

reloxx13, 2018-05-23:
wiki -> supported commands
ctrl+f mqtt

andrethomas, 2018-05-24:
> 17:55:04 MQT: Attempting connection...
> 17:55:05 DNS: Query done. MQTT services found 0
> 17:55:05 MQT: Connect failed to :1883, rc -2. Retry in 10 sec
> 

These are normal if you have not yet configured a MQTT server/broker

andrethomas, 2018-05-24:
> it worked and I set up my wifi (even though I had edited the user config file previously).

For that to work from compile you would have needed to specify that you wanted to also flash the settings. Normally once this was done you would only flash the sketch as you do not want to reconfigure the device each time

[image]

Slowsteve, 2018-05-25:
I'm really gratefull for all the help and encouragement you've given. I now have rf bridge connecting to wifi (superfast) and it is seeing button presses from my rf remote controls. I've got homeassistant installed on the Pi in a virtual python environment and have installed mosquitto.
 I am now going to try to configure mqtt to connect everything together. 
As an absolute begginer in this field, (retired builder), everything takes such a long time as many new skills and tools have to be researched and implemented, (my dog doesn't recognise me anymore!). Due to you and many others in this 'community' it is possible to make exciting progress.

Slowsteve, 2018-06-10:
Thankyou for all the help.

closed 2018-06-10 by Slowsteve