arendst/Tasmota #2798: Username & Password Set up

luisfpinto, 2018-05-21:
Hello,
 I want to improve the integration of Tasmota for the Yeti app. 

Some users reported that they use username and password to protect their device.

I discovered where I can set up a password to protect Tasmota but I wanted to know if I the username can be different from 'admin' (the default value)

Thanks

arendst, 2018-05-21:
Hi Luis. Currently the username is hardwired set to admin.

I thought that it would be handy to have it fixed and rely on the password for "some kind of" security. I have to draw the line somewhere ;-)

luisfpinto, 2018-05-22:
Great! This will do things easier for me to code in Yeti 😂

arendst, 2018-05-23:
As an addition...

The admin username CAN be changed in user_config(_override).h so I guess you still will have to take that into account.

Sorry....

luisfpinto, 2018-05-24:
Done! Next version will handle username and password 👍  Thank you! 😄

shayshtern, 2018-05-26:
Hi! what if I flashed but didnt set user and password - BUT it ask me for password?
I cant login... :(

luisfpinto, 2018-06-14:
The new update adds functionality to introduce username and password!