arendst/Tasmota #2801: PIR on a Timer

Lawrencezarb, 2018-05-22:
If I connect a PIR sensor to one of the GPIO ports to control the realy, is it possible to configure the Sonoff (maybe using rules?) to only detect the movement between certain times of the day. 

Please note that the sonoff is not connected to any Home Automation system and is free standing.

andrethomas, 2018-05-22:
I don't believe that will be possible.

I think the question is what you would do with the data that is created when the PIR sensor is triggered... normally one would push this via MQTT to something like mosquitto and perhaps node-red? I think the simplest would be to implement something to ignore triggers during specific times of the day on the automation system side.

Perhaps another solution (albeit a little more technical) could be to power the PIR sensor from one of the GPIO pins which is time controlled by Sonoff-Tasmota - that would give a less than perfect solution but still a solution - question is though... what will the PIR be triggering and how would you process the triggering of the PIR sensor anyway leading to the suggestion that without an automation system of sort (such as home assistant or node-red) there would not be much point.

Unless, of course, it is possible that I misunderstand your question entirely.

ascillato, 2018-05-22:
> If I connect a PIR sensor to one of the GPIO ports to control the realy, is it possible to configure the Sonoff (maybe using rules?) to only detect the movement between certain times of the day.

Yes, it is totally possible but only with the last version of tasmota (development branch) v5.14.0a

Configure the pin where the PIR is, as switch1 (for example). You can use switchmode1 3 (to detect only if PIR sends a 1 to the Sonoff)

Set a timer to be triggered when you want to control the relay with PIR
Set another timer to be triggered when you don't want to control the relay with PIR

Set a rule to detect the fist timer and set a variable to 1
Set another rule to detect the other timer and set the same variable to 0

Set a rule to detect the state change of the switch and that makes a power with the value of the variable
(if the variable is 1 will power on, if the variable is 0 will power off <- if the relay was off nothing will happen)

**Everything together. Config**

* Set pin as switch1
* set timer1
* set timer2
* on console: `backlog rule 1; rule 4; switchmode1 3`
* set rules, on console: `rule on Clock#Timer=1 do var1 1 endon on Clock#Timer=2 do var1 0 endon on switch#state do power1 %var1% endon`

(var1 in this example will act as an enabled/disabled option)

This is an example, may be that some tunning is needed. See commands and rules on the wiki.
Please check it and if it works you can add it to the wiki as an example, because your idea is very usefull. :+1: 

As extra can be added that on boot it did not check the pir for some warming time. Also to send a mqtt message to an automation system, etc.

andrethomas, 2018-05-22:
Perhaps easier...

Connect PIR as normal to voltage and send output of PIR through a relay to an input pin.

Turn relay on/off at specific times so that when the relay is off no input will be received on input pin.

ascillato, 2018-05-22:
Yes, it is easier, but you need 2 relays. One for controlling the PIR and another for the light. If it is your case you can do it that way if you like it.

Now automations can be solved in different ways. All are valid options. Thanks for share. :+1:

Lawrencezarb, 2018-05-23:
Hi All,

Thanks for all your useful and helpful comments. I am wating for my PIR sensor so cannt test it yet. 

The real world senerio for me, is that the Sonoff will be use to control my sprinkers, which I will use the timers.

I have a pond in my garden that a Heron likes to visit. The idea is to use the PIR to trigger the sprinklers, to hopfully scare the Heron away, but only operate in the early mornings, say from 5am to 8am

reloxx13, 2018-05-23:
@ascillato using `switchmode1 3` is require to use pulsetime as timer right?

currently im using a pir without rules, so it turns on 24h if someone is moving and im using `switchmode1 1`
and have the jumper on the outside (pulses as long someone is moving)

[image]

so it will use the configured on time i configured by the potentiometer on the pir.

if i change this to switchmode1 3 now, the device wont turn off, since its only listening on the ON state.
so i need the pulsetime to get it off again, since i cannot use "if" in the rules like 
if motion==1 and var1==1 set state = 1
if motion== 0 set state 0

if i use switchmode1 1, the rule is failing cuz it acts like this:
if motion(1 or 0) set state = 1 (or always 0 by time)
results in always on if var1 is 1.

HC-SR501 infos (german only, but good imagess): http://www.netzmafia.de/skripten/hardware/RasPi/Projekt-PIR/index.html

another usefull hints for PIR: #640

my pir box:
[image]

reloxx13, 2018-05-24:
Feedback:
Rules, Timer, Pulsetimer disabled.
As soon as i change it to Switchmode1 3, the Sensor is delayed to over 10sec before turning on anything.
Tried both Jumper Positions.
Highest sensity set on the PIR.

closed 2018-06-02 by Lawrencezarb