arendst/Tasmota #2803: proposal for Modbus devices integration

gtortone, 2018-05-22:
Various Modbus (https://en.wikipedia.org/wiki/Modbus) devices (sensors, relays board, ...)
are available on the market at affordable price. These kind of devices can be connected
in "daisy chain" mode up to 247 different items using only three pins on ESP (TX/RX/GND)

I developed (and integrated in Sonoff-Tasmota) two power meter (SDM120/SDM630) using
the common "pattern" of pins specification in sonoff_template.h. Users can select two free pins and
assign them to SDM-Tx and SDM-Rx in order to collect on Sonoff power consumption statistics and
publish these metrics on MQTT.

Although this solution works fine it does not exploit all the flexibility of Modbus/RS485 because to use
two different meters user need to "sacrify" two pins for each device making "daisy chain" of sensors
not applicable by design...

With this feature request I'would like to discuss a possible integration of Modbus/RS485 as "bus" 
inside Sonoff-Tasmota (like KNX) with a custom menu' where it is possible to select id and device
type available on the Modbus chain.

let me know what do you think about !

altelch, 2018-05-22:
Hi Gennaro,

I thought of this a while ago but found no time to implement it. My 
initial idea was to do a kind of Modbus-GW not using Modbus TCP but MQTT 
to send the required registers and get the result values back via MQTT. 
This way everything has to be handled by your automation software on the 
other side. The SDM integration is nice but I have some more 
Modbus-Powermeters in use with different register sets and calculations 
to do. When talking about even more sensors it would be rather difficult 
to do more than a gateway solution or you will blow up the code when 
putting all those register-maps and required calculations for many 
different sensors into Tasmota or you need to be able to not only define 
the address and query registers but also the read type (coil, register, 
holding register eg...and this for read and write), how many response 
packets to expect, what type they are and if there are multipliers for 
the read values (eg. Janitza UMG96 and current transformer ratio). Nice 
idea, maybe small change to make modbus a real bus when using multiple 
SDM on one bus but having lots of different Modbus-Devices as pre 
defined senors could be a bit too much for Tasmota. Did you test your 
code with other devices? I can't see start or end-frames or inter 
character spacing or timeout-handling. The SDM seems to be very kind in 
handling the RTU requests but other devices might be more strict in 
protocol implementation 
(http://www.modbus.org/docs/Modbus_over_serial_line_V1.pdf). Depending 
on the RS485-converter you use there's also a need to toggle Send and 
Receive-Mode. So YES, it's a great Idea to support more Modbus-Devices 
but this requires testing every single device to see if it works with 
this simple modbus protocol implementation and if there is not to much 
code size required to support all this devices and how to group them to 
use the tasmota web and mqtt data format.

Greetings and thank you for your SDM-Code,

       Altelch.

gtortone, 2018-05-22:
Hi Altelch,
thanks a lot for your comments and suggestions !

The suggestion of a MQTT message containing the required register to read/write is very interesting, as you write this "gateway" mechanism is very nice because it relieves developers to write new code for each sensor to integrate... a drawback for this solution is that Sonoff-Tasmota is completely "transparent" to values read/write from outside and it is difficult to make available on Sonoff-Tasmota homepage a kind of report about metrics;

regarding Modbus protocol I'm not a Modbus expert but after some read of web pages I learned that a lot of devices implement "Modbus-RTU" protocol (https://www.rtaautomation.com/technologies/modbus-rtu) a subset of "full" Modbus protocol and reading description of protocol I see that is the same SDM devices decode; the relay boards I linked on previous post also implement this kind of protocol;

If it will be possible I would like to follow both ways: the gateway and Modbus-RTU with specialized drivers...

MatzeJoerling, 2018-05-23:
Hi, i like this gateway thing. 
I dont want to have dozens of semi inteligent esp devices. they should be as stupid as possible. 
i want to have one main thing (nodered, hass, home...) that steers everthing.
who cares what is shown on the webpage of esp device. You can do much more with grafana / nodered or all the other tools for visualisation. 

but the idea for specialized drivers is also valid to get a easy start with some mainstream rs485 devices like powermeter, switches. 

thumps up for both.

arendst, 2018-05-23:
Hi Genarro, as expected the user wants it all and that will provide challenges...

I guess the amount of resources Modbus needs, especially if you want to extent the webpages, will make tasmota even more "sluggish" and prone to short of memory exceptions. Currently I try to keep HTML pages as small as possible as there is just not enough memory to provide a plain HTML page for a module configuration for instance. Also rules will bare a heavy load on memory and processing time slots.

By now you probably guess right that I'm more a fan of the MQTT gateway approach providing a stable MQTT-Modbus interface without all kinds of Modbus sensors configuration options.

So I would focus on the gateway option and if that's available possibly think of the all-in-one approach if room permits.

ionciubotaru, 2018-05-23:
Good news @arendst, I think that mqtt to modbus gateway (similar to mqtt to serial gateway) is the best choice. Then the HA system must take care of modbus registers.

gtortone, 2018-05-23:
Hi Theo,
thanks for your warnings and suggestions... message received !

I will investigate on a MQTT-Modbus bridge taking a look at serial bridge available in Sonoff-Tasmota

Regards,
Gennaro

ascillato, 2018-09-06:
For reference: #3133

gtortone, 2018-09-07:
Hi,
regarding #3133 I'm not a RS485 expert but I understand that Modbus is a Master-Slave protocol where a Master (ESP) request a measurement and a Slave (SDM), identified by its id number, answers to that request. In Modbus protocol there is not a Slave-Master communication without a Master request. If I understand correctly a direction pin is only required when other pattern of communication are requested. I suppose also that RS485-TTL adapter provide a correct implementation of RS485 half-duplex communication... my SDM120 is up and running since 5 months without problems...

altelch, 2018-09-07:
Hi, 

normally a direction pin is needed. Most Modbus devices use half duplex so you need to tell your transceiver the communication direction. If you use eg. the max3485 IC you need to toggle the mode. When using RS485-boards it's possible that direction handling is done by on board logic like toggling send direction when RX becomes active. For generic half duplex modbus toggling a gpio before send is required to tell most of the standard transceivers that they are now in send mode but there are more expensive types handling this on their own or using external logic.

Greetings, Altelch.

gtortone, 2018-09-07:
Hi @altech,
currently developments for SDM (RS485) device on Sonoff-Tasmota imply to use a RS485-TTL adapter connected to **two** free pins and (as you suggest) usage of an additional direction pin is not useful... Further developments (if needed) could be provide a solution to use a SDM (RS485) device *without* RS485-TTL adapter, but in that case we will need **three** free pins but these future developments IMHO are not very useful if we will not identify a real benefit.

altelch, 2018-09-07:
Hi Gennaro,

Can you provide me a link with the adapter you use? I did some modbus development a while ago but always based on max485 or max3485 (3,3V version). Also a lot of china boards are just a breakout board based on those chips, so I'm interested in what board you use not requiring direction pin.

Thanks,

   Altelch.

gtortone, 2018-09-07:
Hi @altelch, I'm using this adapter: https://www.hotmcu.com/uart-ttl-to-rs485-twoway-converter-module-p-267.html

but take into account that two ICs soldered on the board have model number erased...

jarcikw, 2018-09-12:
Hi, IF i want to connect SDM120/SDM630. Do i need any converter RS485/Serial (3.3V) to do this or i can connect directly?
Thanks in advance

gtortone, 2018-09-12:
Hi @jarcikw, you need an adapter like this:  https://www.hotmcu.com/uart-ttl-to-rs485-twoway-converter-module-p-267.html

jarcikw, 2018-09-12:
Thanks @gtortone for reply, any additional configuration(like RS485 adress on meter, etc...) counter is needed?

gtortone, 2018-09-12:
Hi @jarcikw, you can use default SDM settings (id = 1, default speed = 9600 baud) and you don't need to change anything in user_config.h (remember only to enable USE_SDM120 or USE_SDM630)

jarcikw, 2018-09-12:
@gtortone , thanks for help.

jarcikw, 2018-09-18:
@gtortone, one more question. How should i connect converter with meter, do i need connect wires between A, B and G?

gtortone, 2018-09-18:
@jarcikw, yes A,B and G of SDM on A,B and G of converter

jarcikw, 2018-09-26:
@gtortone . I have Tasmota 6.2.1.5 with configured SDM120 in user_config:
`  #define USE_SDM120                               // Add support for Eastron SDM120-Modbus energy meter (+1k7 code)
  #define SDM120_SPEED         2400              // SDM120-Modbus RS485 serial speed (default: 2400 baud)`
I'm using Wemos mini board, wiring (Converter and SDM120) A+ <-> A+ , B-<->B- , GND <-> GND 
RX <->  D2 , TX <->D1 (converter and Wemos), when I'm replace D1 with D2 in Tasmota i have got Nan values
My converter:
[image]
My configuration in Tasmota:
[image]

Output in console i have :
`tele/sonoff/SENSOR = {"Time":"2018-09-26T22:51:40","ENERGY":{"Total":0.000,"ActivePower":0,"ApparentPower":0,"ReactivePower":0,"Frequency":0,"Factor":0.00,"Voltage":0,"Current":0.000}}`

What I'm doing wrong?

jarcikw, 2018-10-04:
@gtortone there was a problem with converter, everythings works as expected :)

gtortone, 2018-10-04:
@jarcikw Great ! I'm very glad of this ! What do you think to copy your page about SDM120 configuration in Sonoff-Tasmota wiki ?

closed 2018-10-15 by ascillato2

smartboart, 2019-09-14:
Hi, i would like to read modbus registers from a pv converter / Growatt. Can you explain me, where to change the code or better what to change to read the registers from another device than a SDM120 / 630 ?

meingraham, 2019-09-14:
@smartboart 

Give [Smart Meter Interface](https://github.com/arendst/Sonoff-Tasmota/wiki/smart-meter-interface) a try.

Please address further questions to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

pite81, 2020-04-04:
Hey guys! nice work over there. I having issues with ESP easy and SDM 630 to working together.
Does anybody have a  how to set the device to read the SDM 630 meter, please?
It is working on raspberry pi, but I wish to use it with esp 8266 as I have multiple SDM readers.
Anybody here who has done something like this? Thanks.

[image]

garci66, 2020-05-03:
I have a couple of quite-advanced power analysers / power-meters which have literally hundreds of modbus registers. I really liked the idea of the mqtt-modbus broker if anyone is still interested in it. I dont mind giving it a go but I was wondering if there was a bare-bone device on which to start basing myself (not the modbus part but rather the mqtt command/response logic). I,e, how do I add a new command to the CLI/MQTT ? thanks!