arendst/Tasmota #2805: Sonoff Touch T1 not saving configuration

sleezy100, 2018-05-22:
Hi all,   I've flashed my first Sonoff Touch T1 with Tasmota. The problem im having is:  i connect to it and change the wifi settings, hit save, it then says it has saved new settings and will restart. When it restarts it goes back to wifi manager and never connects to my network.   I've flashed tasmota 5.12.0 with 2.4.0 and again with 2.4.1. It finds my wifi and says it has signal strength of 78%.

ascillato, 2018-05-22:
Hi, may be there is a problem with your password. If it is too long or have special characters like * will not work, and will not connect to your wifi router.

sleezy100, 2018-05-22:
I've just taken it to my router and it connects. As soon as i take it back to the room i want to install it, it will not connect again.  Its wifi manager says that the signal for my network is over 80 %.

ascillato, 2018-05-22:
Your configuration on your router is the default one, or you have changed the channels config? Also remember that a 5G router will not work with ESP8266 chips

Frogmore42, 2018-05-23:
Use 2.3.0, some environments and devices don't work with 2.4.x

jossie67, 2018-05-23:
I have the same issue connecting to Wifi with a Sonoff Touch , I din't change the password of my wifi.

sleezy100, 2018-05-23:
Got it sorted now. Had to put a wifi extender in the same room as the switch.

closed 2018-05-23 by sleezy100