arendst/Tasmota #2806: Sonoff Dual r2 not booting/no network?

Palloquin, 2018-05-23:
Hi there,

I just received 2 new sonoff devices a TH16 and a DUAL R2.

I use the arduino IDE method to upload, this works fine for the TH16. The DUAL uploads just fine too, but never becomes available.

I have set my network setting is the user_config.h to match my wifi and have set:
`#define WIFI_CONFIG_TOOL       WIFI_MANAGER`
This is the exact same setup I'm also using for the TH16.

The sonoff seems to be in some loop where every so often the blue led will blink for a while.
What could I be doing wrong?

Frogmore42, 2018-05-23:
First thing is not providing any of the required information in the template.  

Had you followed the instructions and searched the issues list you would see many issues that look very similar.  Since you got one to work, you might be having the issue where the 2.4.x version of the platform doesn't work for some people/devices.  Had you include the output from status 0, it would be easy to tell.

Had you included the log from the working and the non working ones, it would likely give more of a clue.

Palloquin, 2018-05-23:
Hi Frogmore42,

Thank you for your quick response! I was about to say I did do all the things the checkboxes ask of me , but decided to do them again to make sure I missed nothing.

then I found: (Dont forget to change the CFG_HOLDER => CFG_HOLDER explained )( [here](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#cfg_holder-explained) ) 

that seems to have done the trick.

My bad, thank you!

closed 2018-05-23 by Palloquin