arendst/Tasmota #2807: min PWM frequency

floris-b, 2018-05-23:
Hi all, 
It would be nice for constant current led driver hack I am doing to be able to set the min PWM frequency.
Now I need to set the brightness to about 50% before the leds starts to light up which is not desirable and also has side effects with flickering on transitions.
I looked at core_esp8266_wiring_pwm.c but it does not look easy to me to make the necessary changes.
What would be the best and or quick way to set the min PWM frequency?

arendst, 2018-05-23:
How about command pwmfrequency....

floris-b, 2018-05-23:
Oops I meant the minimum PWM duty cycle

arendst, 2018-05-23:
Yep, that the file to tweak.

floris-b, 2018-05-23:
A pointer would be greatly appreciated
Or could it be as simple as modifying: 
const LCwColor kFixedColdWarm[MAX_FIXED_COLD_WARM] PROGMEM = { 0,0, 255,0, 0,255, 128,128 };

closed 2018-07-10 by ascillato2