arendst/Tasmota #2808: S26 Smart Plug support

nickjedl, 2018-05-23:
Hello

Smart On Off just recently released the S26 Smart Plug. It's a new version of the S20 Smart Plug.
I see support is not yet added for this. So here's me requesting it.

Thank you in advance guys!
Only now gonna start using your software but I have a feeling it will be great! ;)

Jason2866, 2018-05-23:
Why do you think it is not compatible?
Just try S20 as device. It looks like a design upgrade only...

djwillis, 2018-06-02:
Popping the info here as it seems relevant.

The S26 is still an ESP8266 device and as Jason mentions it looks more like a board redesign to create smaller sockets. 

Seems to work with the S20 setup on the same GPIO but more testing is needed. I have more on order to play with as this one got used in another project.

However it is not all good news. At least on the UK style model, the ESP8266 is moved off to a daughter board and the pads (no holes any more) for the serial and 3.3v are much smaller in pitch making soldering more of a pain. 

It may mean these are much less desirable for tinkering than the S20's as flashing them is so easy for a DIY'er.

Attached are a few (bad, as I was in a rush) photos of the board.

[image]
[image]
[image]

stewface, 2018-06-04:
Dam... the S26 is a bit smaller but more importantly its cheaper!

Keep us updated if support comes

arendst, 2018-06-05:
As @djwillis suggested it might just work choosing module S20...

freynder, 2018-06-11:
Hi,
I can confirm it works without issues when choosing module S20.
The pads are tiny and not easily accessible, so loading the firmware was a pain. I connected the power pins to the regulator output pins using test clips and soldered a wire to RX and TX. The button is connected to GPIO0 so can be used to enable the serial bootloader.

fluppie, 2018-06-18:
[image]
[image]
[image]

I soldered some wires to them and installed a BME280 inside :). So when they run above 60 degrees I get a pushmessage that they caught fire :D. Ambient 22-23°C gives 30°C inside the Sonoff S26. 40°C when turned on with a 60W bulb.
[image]

arendst, 2018-06-23:
Great pictures!!

My S26 is on the way. Glad it is supported already. When verified I'll update the readme and wiki.

newzealandpaul, 2018-06-25:
@fluppie are they easy to disassemble? Are the RX and TX ports marked? I can't quite see in the photos. I am keen to purchase one.

fluppie, 2018-06-25:
3 screws, so quite easy yes :). Only the solder pads are tiny, so it shouldn't be the first time that you solder :).
[image]

newzealandpaul, 2018-06-26:
Perfect thanks for the photo!

roguestreak, 2018-06-26:
Sorry to pseudo-hijack, but while we're discussing it can the plug/socket be inverted?
Looking at the S26, for my specific purpose it wouldn't work as it would be "upside down" and not fit;  From these and other photos it SEEMS like both the plug and socket could be rotated 180degrees, can anyone confirm or reject?  OK if it needs longer wires soldered to achieve.

TommySharpNZ, 2018-06-27:
@roguestreak I opened one up and it'd be easy enough to change the orientation.....

arendst, 2018-06-29:
S26 fully supported in latest releases either as S20 or S2X.

closed 2018-06-29 by arendst

PJFonseca, 2018-07-05:
@fluppie thank you. With your images and 4 hours later ( i'm a n00b at soldering ) i did it :)
[image]
[image]

vogler, 2018-08-09:
Might help someone: I had to increase the voltage from 3.3 to 3.4 V (or was it 3.6?) to get it into programming mode. Before that it didn't react. Probably depends on how well you soldered - although mine didn't look too bad 😄

Jason2866, 2018-08-09:
The ESP is a 3.3 Volt device if your power supply for flashing is reliable there is no need
to increase!
I think your power supply does not deliever enough current.
Maybe you use a FTDI device. The 3.3 Volt path are delivering only a few mA!!
Try a extra power supply or use a CH340 Usb serial adapter

vogler, 2018-08-09:
I used a Rigol DP832 for power, the current limit was at 500mA.

Jason2866, 2018-08-09:
Nice Power Supply. Why have you limit current at 500mA?
Just connect the power wires correct and safe and use 3.3V 
No current limit needed. The device takes only the current which it needs!
Before you power on connect GPIO to GND. Urgent keep Gnd from power supply, device and serial adapter always connected. Put serial adapter in usb port. Now power via Rigol with 3.3V.
Start flash process. It will work this way.

vogler, 2018-08-09:
Safety? It didn't check what the max current draw was, but 500mA should be more than enough, no? I think it was 0.3 mA powered down and ~34 mA after I got it into programming mode.

I just wanted to share that I had to increase the voltage. First I thought I fried it somehow. Maybe it was the breadboard/cables.

Kedryn, 2018-08-22:
I'm using an arduino as a programmer (as i did with all sonoff basic i have), but i can't get the S26 in programming mode. Does the led has to stay off? mine is blinkning no matter what.

freynder, 2018-08-24:
[image]
[image]

I hope this helps.

Kedryn, 2018-08-24:
A bit, but i need a closer and more "front" photo of this. If possible, from the side where you see "T1" on the board.

[image]

freynder, 2018-08-24:
Hi,
Ok, here it is under a magnifying glass. Best I can do with my cam. Best of luck.
[image]

Kedryn, 2018-08-24:
Thanks.. i have a bigger filament that i can sold to the other that got severed, but it seems that there are small filaments too that connects to the other pins... this is the condition i'm right now. Better way is if i find a replacement that i can put in place of this, but i think it will be hard. 

[image]

svh1985, 2018-08-25:
Does anybody know if it has 1M or 4M? And is it upgradable?

wojked, 2018-08-30:
I have used a slightly recharged lipo (3.7V) as a power supply, otherwise I was not able to connect so the insufficient current might have been the reason indeed.

fluppie, 2018-08-31:
@svh1985 1MB, mine is running Tasmota 6.1.1 and gives this info:

Flash Size | 1024kB
-- | --
1024kB
536kB
464kB
16kB

freynder, 2018-09-02:
> hi all,
> I'm using a CH340 USB Serial Adapter. I have two Sonoff S26 sockets. I
> managed to flash one by connecting the power to the VCC port but I cannot
> get the second one to enter flash mode no matter what I do. I even used a
> step-down regulator and connected it to the 5v pin to no avail. Connecting
> it to the 3.3v pin does not work and connecting it to the VCC pin makes it
> heat up and still does not work. The same happened with the step-down
> regulator. It just heats up and nothing else happens.
>
> It is still working as I can connect it to a power socket and control it
> via the eWeLink app. I just cannot flash Tasmota. I don't have a
> professional power device so is there anything else that I can use?

Hi,

There is an AMS1117 LDO regulator on the board. I would suggest connecting
the ground and input pins with the 5V output and ground output of your USB
Serial adapter.

You can find the datasheet here:
http://www.advanced-monolithic.com/pdf/ds1117.pdf . Pins 1 and 3 are ground
and Vin respectively.

tam481, 2018-09-02:
@freynder Thank you very much indeed. That worked beautifully. I've managed to flash Tasmota successfully.

ant-thomas, 2018-09-03:
Not sure if this is the right place to ask but I successfully flashed 3 x S26 sockets a few days ago. My only issue is the Blue LED. Usually it flashes, I've managed to set it to permanently on using the "LedPower on" command. Is it possible to turn it off permanently?

ascillato, 2018-09-04:
try commands ledpower and ledstate. please see for more information in the wiki

nate131, 2018-09-05:
I bought 3 of these modules - was able to flash two of them using esptool (platformio was unable to upload the firmware).

Does anyone know the pinout for the esp daughterboard? I ripped one of the traces out accidentally and want to replace it with an esp-01. The mainboard of the S26 has the labels of J1-5 & J10 (6 connection points)

arendst, 2018-09-08:
And in addition set SetOption31 to 1 to get rid of wifi and/or mqtt connection failure blinks

ant-thomas, 2018-09-08:
Great, thanks. Looks like that one I need.

ant-thomas, 2018-09-08:
After buying another 3 S26 sockets I decided to look at an easier way to flash.
I designed and 3D printed a solderless type adapter to flash a bit easier. It's not perfect but does work
[image]
I'll upload the STL soon.
Also found that the 2 more difficult pads out of the 4 to solder are the 3V3 and GND. You can use alternative pads underneath
J1 - 3V3
J2 - GND
[image]

simweb-info, 2018-10-05:
ant-thomas - looks nice, do you have this stl file? would be great :) or maybe to find in thingiverse
br

alexmelSC, 2018-10-18:
Hi,

I'm trying to flash this with SonOTA but the ITEAD wifi won't come up (it came up once and never again). Anyone had any luck?

A

TommySharpNZ, 2018-10-19:
I'm pretty sure the current Sonoff firmware doesn't allow OTA flashing anymore? V1.6 was the last version that worked I think...

alexmelSC, 2018-10-21:
That sounds about right. Anyway happily using them via the EWElink app but just wish I had something to tinker with Home Automation. Will wait.

netro, 2018-10-21:
Hello @alexmelSC @TommySharpNZ and @rzr,
I'm pretty new to this world as I received yesterday my first Sonoff unit ever so I might not understand well what you are talking about..

What I can tell you is that I had some issue with getting the ITEAD WiFi SSID appearing for the first config of the unit but at the fourth try it worked and I was able to configure the socket to connect with my WiFi.

The firmware version showed up in the EWeLink app is 2.6.0.
I didn't have to work with OTA at this time so I might be out of the scope of your discussion ^^

My two cents,

gnomosg, 2018-11-03:
Hi, it took some time, but it works :)
I used pogo pins: 
https://www.aliexpress.com/item/100Pcs-Bag-Spring-Test-Probe-Pogo-Pin-P50-B1-Dia-0-5mm-Length-16-35mm/32845302540.html

[image]

dugite-code, 2018-11-04:
A good write up on the [espurna github](https://github.com/xoseperez/espurna/issues/980#issuecomment-420220446) solved my issues i.e. I wasn't pushing the button hard enough.

That was 3 hours wasted on a sticky button

thomasleitner, 2018-11-10:
Hi Guys, great show here: I was able to flush two S26 with the latest ESPEasy software using the provided pictures above. At times a bit complicated to put it in flash mode but with some fiddling (connect everything except USB, push button and simultaneously connect USB, release button) it works.
Now my problem is: After re-assembling both S26, only one of them works: The blue LED is lit and it connects to the Wifi. For the other one: No blue LED, not Wifi. I re-opened it, resoldered power and reconnected 3.3V and it works great. Only when assembled and connected to the mains, it does NOT work. I've double checked all connections internally, checked soldering etc. but no avail.
Any educated guesses out there what the problem could be?
Thanks // Tom

rzr, 2018-11-10:
@thomasleitner , Can you see any difference between the two, can you at least try to revert to original firmware ?
I have one S26 that was just paired once.

thomasleitner, 2018-11-10:
Thx.  However I don’t think that it’s a software problem. As I’ve explained, when reconnecting to an external 3,3v supply, it works great. Only when reassembled and connected to 230v AC power, it does not power up. Hmm ...

dugite-code, 2018-11-12:
@thomasleitner could it be the button it getting pushed by the casing? The LED might not be on because it's gone into flash mode.

I know the button extender on mine will sometimes get caught in the case because it's not a flat surface, kind of a design flaw in the S26 IMHO

thomasleitner, 2018-11-12:
@dugite-code yes I was already thinking about that, but the button feels as if it would click when the case is assembled. I’ll double check, though ... thanks ...

pwaldon, 2018-12-13:
Hey,

is it safe to reflash S26 with 2.6.0 firmware, or it doesn't matter? (Some time ago I had 1.8.1 but the new one is 2.6.0, dunno why).

dugite-code, 2018-12-14:
@rzr if you're using esptool like I do https://github.com/arendst/Sonoff-Tasmota/wiki/Esptool#optional-backup-firmware

neirbomn, 2019-02-23:
I have tried flashing these after following this thread and Jon Oxer SuperHouse Episode 31. I think I have a power related issue as flashing results in the "A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header" error. To rule out issues with the MacBook I used an ESP-01 and a cheap USB adapter for flashing these. All good. I conclude that it's a power issue therefore as I know the USB Serial adapter is good but it produces ~3.3V (possibly slightly less as my cheap multimeter reads it as 3.25V). Can I use a LiPo battery to "up the voltage" a bit as I'm pretty sure there's not enough juice for the flashing procedure? I'm thinking I could do VCC and GND from battery and just use the serial connection for Tx and Rx. Would this work? I don't have a bench supply but I may be able to get access to one. The other idea I had would be to extract the ESP breakout board and then flash this before resoldering it. In this case, how would I tie GPIO 0 to GND? Is there a contact for this on the ESP board?

rzr, 2019-02-23:
I managed to flash mine , but I have to say that getting to flashing mode is not straightforward.

Also I'd be curious to compare what is differing in original images backuped from different devices...

neirbomn, 2019-02-23:
> I managed to flash mine 

Do you mind if I ask you for details of what your process was to successfully flash them? I've soldered onto pads for all without success and also using the Jx points under the board. I'm using the same Freetronics USB Serial adapter as Jon Oxer and the latest esptool. Scoured the troubleshooting section looking for a solution but without success yet. I would really like them to work. I've temporarily junked one of them by pulling off the solder pad inadvertently so that's a job for another day. Starting to wish I'd bought the S20s instead.

rzr, 2019-02-23:
I used exptool from git at 460800 baudrate and address 0x000000

In my notes:

```
    $esptool \
        --port ${port} \
        write_flash \
        $address \
        $@

    Wrote 454864 bytes (313424 compressed) at 0x00000000 in 27.6 seconds (effective 131.8 kbit/s)...
```

And for the USB/UART adapter I think i git it along nodemcu like ESP8266, 
I can get the reference if needed

dugite-code, 2019-02-24:
Hey, I got that error when I hadn't pushed the button firmly enough, it was a bit sticky on mine. Is the led flashing?

neirbomn, 2019-02-24:
Thanks for that but no flashing on LEDs on S26.

rzr, 2019-02-24:
have you managed to backup the stock fw ? I can try to diff your image with mine to see if there is any hw version or others hints ?

neirbomn, 2019-02-24:
> have you managed to backup the stock fw ? I can try to diff your image with mine to see if there is any hw version or others hints ?

Thank you for the offer and thanks for helping me out, problem is I get the same error when attempting to backup stock fw. It's definitely not getting into the bootloader properly or it's crashing early once there and I can only conclude that it's a power issue. I used a different adapter to flash an ESP-01 and this worked.

I have a few things left to try and then I'm out of ideas. **Really** wish I'd gone for the S20s now!

1. Try and flash with the ESP-01 adapter. Same as [this one](https://www.ebay.com.au/itm/ESP-01-ESP8266-Wi-Fi-Transceiver-Module-USB-to-ESP-01-Adatper-WS/192287007218?hash=item2cc532e1f2:g:Si0AAOSw44BYT16X:rk:1:pf:1&frcectupt=true). I know this works with ESP8266, just not with S26.
2. Try with a bench supply... (if I can get hold of one)

neirbomn, 2019-02-25:
> And for the USB/UART adapter I think i git it along nodemcu like ESP8266,
> I can get the reference if needed

@rzr Could you find a reference for this too? My USB to Serial device uses the ATmega16u2 MCU. I have flashed the ESP-01 and decided to use the esptool and settings with an Adafruit Huzzah to load the sonoff.bin file. It didn't work at the low baudrate. I omitted the baudrate and all was ok. The only device I haven't checked is the USB to Serial device I have. If yours works then I may as well try this too. I have another device but even though it has a switch for 3.3V the Vout remains at 5V. I may use the Vout from the ATmega and the Tx and Rx from the other device...

MatzeMuc86, 2019-03-11:
Dear community,
I started using Tasmota a few weeks ago successfully for my NodeMCU + DHT 22 and love it. THank you so far for the great software and development!
Now I tried to flash my Sonoff S26. Flashing was successful (at least what I assume) with version 6.4.1.19 and Visual Studio Code. I can access the Tasmota web interface via browser, changed the module to Sonoff S2X (8) and can toggle the switch. I even can hear the relay working but nothing happens. I tried a second Sonoff S26 which I used before with original firmware and it behaves the same way. Same if I change to generic device and manually assign GPO12 to relay1 or relay1i.
So I am out of ideas why it seems to work properly but it is not working as expected.
Any hints?
Thank you and best
MatzeMuc86

AndyP2, 2020-02-06:
I might have found a simpler method to avoid the "fatal error occurred: Timed out waiting for packet header" error.
I was using [tasmotizer](https://github.com/tasmota/tasmotizer) - power on with the button pressed, release button, start the flashing process. It backed up the original firmware then timed out and gave me the error. However, if I held the button down for the whole of the upload process it flashed successfully; hopefully this will help others who have the same problem.

[11 more replies]